redrathnure/armbian-mkspi #2: Not able to compile

Tina-Builds, 2023-01-06:
I am sorry, I guess I do something terrible wrong:
When I clone the repo with 'git clone https://github.com/redrathnure/armbian-mkspi.git' and then do a './compile.sh' I get errors:
"...
[ o.k. ] Create example config file using template [ config-default.conf ]
[ o.k. ] Using config file [ /root/armbian-mkspi/userpatches/config-example.conf ]
[ o.k. ] Command line: setting BOARD to [ mkspi ]
[ o.k. ] Command line: setting BRANCH to [ current ]
[ o.k. ] Command line: setting RELEASE to [ jammy ]
[ o.k. ] Command line: setting BUILD_MINIMAL to [ no ]
[ o.k. ] Command line: setting BUILD_DESKTOP to [ no ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ no ]
[ o.k. ] Command line: setting COMPRESS_OUTPUTIMAGE to [ sha,gpg,img ]
/root/armbian-mkspi/lib/functions/main/config-prepare.sh: line 85: /root/armbian-mkspi/config/boards/mkspi.: No such file or directory
[ error ] ERROR in function prepare_and_config_main_build_single [ functions/cli/cli-entrypoint.sh:106 -> functions/main/config-prepare.sh:88 -> functions/logging/traps.sh:0 ]
[ error ] Board configuration does not define valid kernel config
[ o.k. ] Process terminated
"
Any ideas?

redrathnure, 2023-01-08:
`master` branch does not have any customizations, please checkout `custom/mkspi_23.02.0-trunk.0135` branch (e.g. `git checkout custom/mkspi_23.02.0-trunk.0135` after `git clone ...` command).

redrathnure, 2023-01-14:
@Tina-Builds does it work? Should we close an issue?

redrathnure, 2023-01-26:
please try following commands:
```
git clone https://github.com/redrathnure/armbian-mkspi.git
cd armbian-mkspi
git checkout custom/mkspi_23.02.0-trunk.0171
```

williangomesa, 2023-01-26:
Thanks for the repply, I do received this message:

fatal: not a git repository (or any of the parent directories): .git
root@willian-VirtualBox:/home/willian/armbian-mkspi#

williangomesa, 2023-01-26:
It did work now, thank you very much for the help.

williangomesa, 2023-01-30:
HI guys, sorry for posting in the wrong place, I wasn't able to put in the correct place;  i'm trying to use an adxl345 module for the input shaper, first, i've tried to use an arduino nano as secondary mcu over usb to make it work (the module is working, I've tested in another printer), but as soon as I've declared the secondary MCU the board doesn't come after reset, keeps showing "The klippy host software is attempting to connect.  Please
retry in a few moments"; then I tried to use the guide to work with the "raspberry" as a seconary MCU, the same error happens, I'm using the mkspi_23.02.0-trunk.0171, compiled as ./compile.sh BOARD=mkspi BRANCH=current RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img

williangomesa, 2023-01-30:
I do receive the message"ln: failed to create symbolic link '/etc/init.d/klipper_mcu': File exists
" when configuring the mks pi to use adxl345

redrathnure, 2023-01-30:
@williangomesa perhaps you do not [need ~~RPi~~ MKS PI as a secondary MCU](https://www.klipper3d.org/RPi_microcontroller.html) for your configuration. 

I assume you should see new serial device after connect Arduino(with adxl345 ) to your ~~RPi~~ MKS PI board. E.g. check `ls /dev/serial/by-id/*` output, most likely you will see two devices here (on is 3D printer board and second is Ardiono one). 
In this case you should play with `[mcu rpi]` and `[adxl345]` sections. `serial` should point to one of `/dev/serial/by-id/*` file (but not `/tmp/klipper_host_mcu`)

williangomesa, 2023-02-20:
Hi, did you @redrathnure have tried to use a usb flash drive in the board ? i'm asking because i'm able to use in another brand of board, in order to print, but in this case it seems that doesn't work.

redrathnure, 2023-02-20:
@williangomesa I am not sure I got your question right. Do you mean using USB stick to store some files? Or do you want to boot OS from USB stick or external USB drive?

williangomesa, 2023-02-20:
I mean, using a USB stick in order to print.

redrathnure, 2023-02-20:
I have done backups to/from USB stick. So I can confirm that USB ports work, even USB3 mode. 
Most likely you would find a way how to mount it automatically (AutoFS, USB mount or some scripts to automatize this tasks).

redrathnure, 2023-05-09:
Guess original issue was solved. Please feel free to reopen issue in case of compilation problems.