redrathnure/armbian-mkspi #5: Wired Network and LCD Screen Stop Working

emericklaw, 2023-03-12:
### What happened?

The wired networking stops working along with the LCD screen.

This might be linked to doing an apt upgrade...

### How to reproduce?

After initially flashing this firmware everything works as expected, wired, wireless and LCD are perfect.

After an 'apt upgrade' and a reboot the wired networking stops working (no lights on the port) and the LCD. If I connect a screen to the HDMI port and configure the wifi using armbian-config I can get connected again.

I am still not 100% sure if the apt update breaks it or just rebooting, the first time this happened I reflashed the card and started again.

I have a few errors in the syslog:

> Mar 12 15:35:45 mkspi systemd-udevd[447]: Network interface NamePolicy= disabled on kernel command line, ignoring.
> Mar 12 15:35:45 mkspi systemd-udevd[450]: Using default interface naming scheme 'v249'.
> Mar 12 15:35:45 mkspi systemd-udevd[450]: eth0: Failed to query device driver: Device or resource busy
> Mar 12 15:35:45 mkspi fake-hwclock[410]: Sun Mar 12 03:35:35 PM UTC 2023
> Mar 12 15:35:45 mkspi systemd-udevd[450]: eth0: Could not set offload features, ignoring: Device or resource busy
> ...
> ...
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.5142] policy: auto-activating connection 'Wired connection 1' (f89a75a1-7fe7-381e-bb75-1043d639d08d)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.5207] device (eth0): Activation: starting connection 'Wired connection 1' (f89a75a1-7fe7-381e-bb75-1043d639d08d)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.5230] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.5313] device (eth0): set-hw-addr: set-cloned MAC address to 0E:B7:23:E8:64:FD (0E:B7:23:E8:64:FD)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.5316] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:10 mkspi kernel: [  326.034152] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [Generic PHY] (irq=POLL)
> Mar 12 15:41:10 mkspi kernel: [  326.046993] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
> Mar 12 15:41:10 mkspi kernel: [  326.254229] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
> Mar 12 15:41:10 mkspi kernel: [  326.254272] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
> Mar 12 15:41:10 mkspi kernel: [  326.254290] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_open: Hw setup failed
> Mar 12 15:41:10 mkspi NetworkManager[847]: <warn>  [1678635670.7595] platform-linux: do-change-link[2]: failure changing link: failure 110 (Connection timed out)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7598] device (eth0): state change: config -> failed (reason 'config-failed', sys-iface-state: 'managed')
> Mar 12 15:41:10 mkspi NetworkManager[847]: <warn>  [1678635670.7650] device (eth0): Activation: failed for connection 'Wired connection 1'
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7677] device (eth0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7729] device (eth0): set-hw-addr: reset MAC address to 0E:B7:23:E8:64:DD (deactivate)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7772] policy: auto-activating connection 'Wired connection 1' (f89a75a1-7fe7-381e-bb75-1043d639d08d)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7817] device (eth0): Activation: starting connection 'Wired connection 1' (f89a75a1-7fe7-381e-bb75-1043d639d08d)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7829] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7900] device (eth0): set-hw-addr: set-cloned MAC address to 0E:B7:23:E8:64:FD (0E:B7:23:E8:64:FD)
> Mar 12 15:41:10 mkspi NetworkManager[847]: <info>  [1678635670.7903] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:10 mkspi kernel: [  326.292325] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [Generic PHY] (irq=POLL)
> Mar 12 15:41:10 mkspi kernel: [  326.294643] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
> Mar 12 15:41:11 mkspi kernel: [  326.502850] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
> Mar 12 15:41:11 mkspi kernel: [  326.502899] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
> Mar 12 15:41:11 mkspi kernel: [  326.502924] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_open: Hw setup failed
> Mar 12 15:41:11 mkspi NetworkManager[847]: <warn>  [1678635671.0091] platform-linux: do-change-link[2]: failure changing link: failure 110 (Connection timed out)
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0094] device (eth0): state change: config -> failed (reason 'config-failed', sys-iface-state: 'managed')
> Mar 12 15:41:11 mkspi NetworkManager[847]: <warn>  [1678635671.0153] device (eth0): Activation: failed for connection 'Wired connection 1'
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0186] device (eth0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0246] device (eth0): set-hw-addr: reset MAC address to 0E:B7:23:E8:64:DD (deactivate)
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0300] policy: auto-activating connection 'Wired connection 1' (f89a75a1-7fe7-381e-bb75-1043d639d08d)
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0357] device (eth0): Activation: starting connection 'Wired connection 1' (f89a75a1-7fe7-381e-bb75-1043d639d08d)
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0375] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0458] device (eth0): set-hw-addr: set-cloned MAC address to 0E:B7:23:E8:64:FD (0E:B7:23:E8:64:FD)
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.0462] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:11 mkspi kernel: [  326.548740] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [Generic PHY] (irq=POLL)
> Mar 12 15:41:11 mkspi kernel: [  326.551156] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
> Mar 12 15:41:11 mkspi NetworkManager[847]: <warn>  [1678635671.2624] platform-linux: do-change-link[2]: failure changing link: failure 110 (Connection timed out)
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.2626] device (eth0): state change: config -> failed (reason 'config-failed', sys-iface-state: 'managed')
> Mar 12 15:41:11 mkspi kernel: [  326.758846] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
> Mar 12 15:41:11 mkspi kernel: [  326.758879] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
> Mar 12 15:41:11 mkspi kernel: [  326.758889] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_open: Hw setup failed
> Mar 12 15:41:11 mkspi NetworkManager[847]: <warn>  [1678635671.2672] device (eth0): Activation: failed for connection 'Wired connection 1'
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.2694] device (eth0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
> Mar 12 15:41:11 mkspi NetworkManager[847]: <info>  [1678635671.2736] device (eth0): set-hw-addr: reset MAC address to 0E:B7:23:E8:64:DD (deactivate)

And the armbian-config tool when I try and activate the wired connection:

[image]

I can try more troubleshooting steps but I am at a loss of where to go next.

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

emericklaw, 2023-03-12:
Now it's dark it turns out the LEDs on the network port are actually lit with no cable plugged  on. They are very very dim.

[image]

redrathnure, 2023-03-12:
@emericklaw could you double check that you have *not* updates kernel and boot packages? 

 Usually I use Jammy image and and do not have any LCD issues so far.

emericklaw, 2023-03-12:
I suspect I have if they are included in an **apt update/upgrade** which I ran without thinking after the initial setup.

I was also using Jammy image which I will reflash shortly and try again.

redrathnure, 2023-03-12:
Just got the same problem after `apt update`. Kernel package was updated and original `/boot/dtb/rockchip/rk3328-roc-cc.dtb` was lost (which explains problems with LCD, ethernet and perhaps other SPI ports). Easiest way how to check:
```
sudo dtc -I dtb -O dts -o rk3328-roc-cc.dts /boot/dtb/rockchip/rk3328-roc-cc.dtb
less rk3328-roc-cc.dts | grep model 
```
If you do not see ` model = "Makerbase mkspi";` then kernel was overwritten. 

As solution I may propose to run `sudo armbian-config` on the fresh new image (before apt update), and disable kernel updates (System -> Freeze - Disable Armbian kernel updates)

Jarewa, 2023-03-14:
I was just about to write that kernel updates are not blocked

redrathnure, 2023-03-14:
Any ideas how to freeze them during image build time?

redrathnure, 2023-03-16:
FYI I kernel packages were freezed in [0.1.7 release](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.7_23.02.0-trunk.0236). You may try it.

redrathnure, 2023-05-09:
Kernel packages were freezed in the new releases.

closed 2023-05-09 by redrathnure