redrathnure/armbian-mkspi #7: Tutorial - Installing Klipper in MKS PI

onolox, 2023-03-19:
### Which feature would you like to have?

Process of install this Armbian:

- Download the image you like
- Install in the SD Card of eMMC via [BalenaEtcher](https://www.balena.io/etcher)
- Check the size of the system partition, maybe you need to rezise it, (leave 500MB free at the end).
- Turn on, get the IP from the LCD (cable or wireless)
- Log in as root, pass 1234, folow steps in the console
- Log in as user created
- apt update, apt upgrade
- Turn on the internal PINs for ADXL345: https://www.klipper3d.org/RPi_microcontroller.html?h=gpio
- Turn on the PINS via  armbian-config > System > Hardware toggle on the option there, confim and reboot
- Install Klipper via: https://github.com/th33xitus/kiauh

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

onolox, 2023-03-19:
To use the internal PINs of the MKS PI:

[mcu mkspi]  
serial: /tmp/klipper_host_mcu    

[output_pin BEEPER_pin]
pin: mkspi: gpiochip1/gpio24

[image]

madejackson, 2023-08-16:
thanks for the guide. Unfortunately i do not have an option gpio. I have an option of 1w-gpio, but activating this one just makes my system unbootable. Using jammy current 6.1.37

RimmeVG, 2023-08-25:
Hello,
When i flash the image to sd card, i have a working MKS Skipr.
When i flash the image to eMMC, then it does not boot.
Is this something that needs to be worked on or should it already have to work?
Thank you.

onolox, 2023-08-25:
I use eMMC, no problems.

PimpstaPee, 2023-12-26:
Touch screen dont work mks ts35 v1.0 on [0.3.0-24.2.0-trunk](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.0-24.2.0-trunk)
Do u know how fix it?

redrathnure, 2023-12-26:
@PimpstaPee  which image (please answer in the first bug) and please do not copy-paste you very important question in all open tickets.  Ideally I would open new bug for this problem.

PimpstaPee, 2023-12-26:
@redrathnure [Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.6.7.img.xz](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F0.3.0-24.2.0-trunk/Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.6.7.img.xz)

yakun110, 2024-11-06:
谢谢你。朋友。我尝试下，官方的固件，不支持我的usb wifi。

yakun110, 2024-11-08:
> 感谢您的指导。不幸的是我没有gpio选项。我有 1w-gpio 选项，但激活此选项只会使我的系统无法启动。使用 jammy current 6.1.37

me too