redrathnure/armbian-mkspi #11: I would like to ...

denis245707, 2023-05-27:
### Which feature would you like to have?

Good, how do you like the idea of using the TS35 interface (FPC) to connect the camera?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2023-05-28:
Sounds interesting, however I have no idea how to do this:)

ChristineBoersen, 2024-02-05:
I found some notes on klipper using one of the interfaces that might help you, as I presume this is along the lines of what youare trying to do?
https://www.reddit.com/r/klippers/comments/10g9ir7/klipper_raspberry_pi_camera/

SnakeKVC, 2024-11-16:
Hello!
Is it possible to make an i2c bus on reserved pins to connect eddy coil?

redrathnure, 2024-11-16:
@SnakeKVC MKSPI has reserved pins for I2C bus (see next to EMMC connector). However I have not tested it neither with original nor with custom images.
[image]

SnakeKVC, 2024-11-16:
@redrathnure Thank you so much! I've become completely blind.)))

SnakeKVC, 2024-11-16:
I added overlays=rk3328-i2c0 but I get some nonsense
i2cdetect -l
i2c-0 i2c rk3x-i2c I2C adapter
i2c-1 i2c rk3x-i2c I2C adapter
i2c-4 i2c DesignWare HDMI I2C adapter
root@CoreXY-BIG:/boot# i2cdetect -y 0
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: 08 09 0a 0b 0c -- 0e 0f
10: 10 11 12 -- 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a -- 2c 2d 2e 2f
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: - - - - - - - - - - - - - - - - - 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d -- 6f 
70: 70 71 72 73 74 75 76 77

redrathnure, 2024-11-16:
I have 0 experience working with I2C devices. Need time to find something to connect and experiment with....

SnakeKVC, 2024-11-16:
[image]
I would be very grateful to you.
I wanted to connect OLED Display

redrathnure, 2024-11-17:
@SnakeKVC  I'm afraid it isn't possible on MKSPI. I2C pins are kinda shared with Ethernet status LEDs.
[image]
Actually two bottom resistors are absent on PCB (at least on my board). So most likely "some nonsense" which you received is noise or something from ethernet/RJ45 status, but not from device hooked to I2C pins. Most likely you should desolder resistors from LED lines and solder them to I2S lines... but we are talking about 0402 SMD components.

BTW, SKIPR board do not have these pins at all. Related CPU pins connected only to RJ45 LEDs.
[image]

And the second rk3328 hardware I2C port is already utilized for RK805 power controller communication. IMO ether you try to utilize some unused pins to emulate I2C bus (software I2C, if it's possible with this chip) or find some other approach to connect the display (HDMI screen, I2C from MCU/tool head and so on).

SnakeKVC, 2024-11-18:
@redrathnure Thanks for the clarification. It looks like nothing will come of it.

SnakeKVC, 2024-11-18:
@redrathnure I don't have the resistors that go to the LEDs soldered.

[image]