redrathnure/armbian-mkspi #12: need help with configuration of modules for mks klipad50

fryc88, 2023-06-20:
### Which feature would you like to have?

Hi,
I'm struggling to install system on klipad50, that's MKS device, no commonly used yet. It's like combo of IPS50 touchscreen, MKS PI and RTL8723BS (if i'm reading it right) wifi card.
I tried stock MKS PI system, but it won't run WiFi drivers, except of that everything else works, but there's no ethernet connection on it, so can't do it any other way.
I also tried manually installing drivers at least for USB WiFi card that i've got, but it's asking for headers for 5.16.20 trunk version, that's doesn't seem to be downloadable anywhere, at least not via google, and because of above - no connection - I can't use apt.
So I would like to try your system, but I can't find info how do I tell it to use these modules while compiling it? On your GIF in github there's one more step at end, to choose kernel settings, but that doesn't show in my case.
Thanks in advance for any help.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

fryc88, 2023-06-21:
Just a quick update, yesterday I've flashed your precompiled firmware for MKS PI board and that's what are my findings, so I'm assuming something's missing or not turned on. I've compared original config file and yours, added some bits and bobs to yours according to differences, that added me touchscreen, but:
- no video via HDMI-A-0 output, as mentioned this screen appears to be connected (without hdmi connector, just via traces) to HDMI, using HDMI to LVDS converter chip, pretty much the same as original IPS50 screen, but just all on one board
- can't still get built in wifi to work, RTL8723BS (i think) chip seems to be on in config, supposed to be built in kernel, but it does not come up, as far as I've found it works on SDIO bus, that's not detecting anything, so I'm using USB wifi card for now
- touchscreen is working fine after moving configs from original old armbian to your build, events can be seen in event monitor
- also after playing with config files I've got screens backlight on, but still no video.

Any ideas with above how to hopefully get it sorted? I know prebuilt MKS image works with screen, but it does not work with RTL8723BS wifi, and also i couldnt find headers to compile&add USB wifi driver...

redrathnure, 2023-06-24:
Hi @fryc88
First of all, please note that this repo is fork of [armbian](https://github.com/armbian/build) but *not* a [mks repo](https://github.com/makerbase-mks/MKS-PI). Technically this means you may use tools, docs and tutorial for armbian but should not expect that that something from MKS repo(s) will work here. Why? Short answer is they (MKS team) do not share real sources ofr their configs, modules etc. And even if their buils names itself as "armbian" and some wifi card works here, it does not mean that similar build and official kernel module will work (they love to do closed sorce customizations and name them using officially open source names). Which also means that (available) schematic, bit reverse engineering and lot of tries-and-fails as the best friends in your task.

Hints for a few points from your questions:
1. Perhaps you should try to build this image from the sources. It will solve a question with kernel sources/headers, kernel configuration (including embeded and dynamic modules) and possiblity to adjust DTS and patch some sources. Please note that officil kernel sources (e.g. from kernel org) is not the same what is used for this build. There are lot of patches applied depending of specific board and build.
1.2When you build image from the sources, you may not specify board name in cli, then you will see GUI to choose board, build type and OS type. It may be useful if you need to reconfigure kernel modules instead using default one.

About `no video via HDMI-A-0 output, as mentioned` 
2.1 Please not that the build is use default HDMI drivers. If you need special modes, modules or any other kernel adjustment, you have to add them to kernel patches and build your custome image. And pay attention to DTS. Hint: this board [pretends](https://github.com/redrathnure/armbian-mkspi/blob/main/config/boards/mkspi.conf) by [Rocksimp Renegade board](https://github.com/redrathnure/armbian-mkspi/blob/main/config/boards/renegade.conf). So you may compare theirs schematic to e.g. verify pinouts and adjust DTS.
2.2 The both images (MKS and "my" custom one) has a "only one screen" behavior. Means when system detected SPI display connected, then HDMI is disabled. And viceverce, when system was booted with HDMI monitor on, then SPI display and touchscreen are not working. Seems detection happened during boot time. And perhaps this is default configuration of xserver, but not hardware limitation (I do not belive there are any hardware solution which switches between SPI and HDMI videos)
2.3. My image use standart kernel drivers for HDMI, [ilitek,st7796 ][https://github.com/redrathnure/armbian-mkspi/blob/644cb389bca66dc2229ac09df4881a9b34959777/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L176] SPI screens and custom [ti,tsc2046 module](https://github.com/redrathnure/armbian-mkspi/blob/644cb389bca66dc2229ac09df4881a9b34959777/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L193) for touch screen. If you have different LCD or/and touchscreen or/and HDMI ports, you should find right drivers/modules and adjust [DTS for the board](https://github.com/redrathnure/armbian-mkspi/blob/main/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch). Sometimes insperation may came from boards with similar hardware but with true open sourced OS :)
2.4. `pretty much the same as original IPS50 screen` please note "original" is MKSPI-TS32 (LCD works fine with ilitek,st7796 module, but tsc2046-like touch need custome kernel driver). Also pay attention to schematic and [pins](https://github.com/redrathnure/armbian-mkspi/blob/1efbbd23b1df5e585de78fcb817512569305bb5d/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L140-L155)

3. `also after playing with config files I've got screens backlight on, but still no video.` I would recommend to double check pins and [DTS for the board](https://github.com/redrathnure/armbian-mkspi/blob/main/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch). And double check that LED chips are the same. Otherwise try to use [different kernel module ][https://github.com/redrathnure/armbian-mkspi/blob/644cb389bca66dc2229ac09df4881a9b34959777/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L176].

4. `touchscreen is working fine after moving configs from original old armbian to your build, events can be seen in event monitor` You may [decode](https://github.com/redrathnure/armbian-mkspi) DTS files from original and custom images and compare them. Pay attention to [pins](https://github.com/redrathnure/armbian-mkspi/blob/1efbbd23b1df5e585de78fcb817512569305bb5d/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L140-L155) and [related configuration](https://github.com/redrathnure/armbian-mkspi/blob/1efbbd23b1df5e585de78fcb817512569305bb5d/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L174-L213)

5. `can't still get built in wifi to work, RTL8723BS (i think)` I would recomend to check schematic, DTS and loaded kernels in original repo. However it's always bit challenging to make relatively fresh WIFI chips works propertly. Also you may look to better supported bords if they have similar WIFI.

borgaspower, 2024-01-05:
Display rotation is not working on klipad50 :(

torte71, 2024-03-10:
First: Thanks for your work Maxim!

I've just got internal wifi working with your images on the klipad50 by replacing the devicetree file with the version, that came with Sovol's original 5.16 kernel. 
After applying a patch for the naming of spi-dev to the dtb, it now has working wlan0.

About the patch:
Your image creates /dev/spi0.2 instead of /dev/spi0.0 on the klipad50. I assume (or better: guess), that this blocked the detection of the wifi device, so I started playing around with the spi section. I still don't fully understand that stuff, but it seems to work for whatever reason. :D

The change (and original dtb source) can be found here: https://netztorte.de/3d/doku.php?id=rk3328-roc-cc_dts#wifipatch

There is a compiled version of the dtb at the bottom of this page: https://netztorte.de/3d/doku.php?id=rebuilding#internal_wifi
(It also shows the required steps for setting up klipper on the klipad50 based on your image, with screen-rotation and enabled beeps).

I've tested the new .dtb successfully with following versions:
0.3.2-24.2.0-trunk: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.1.76.img.xz
0.3.4-24.2.0-trunk: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.img.xz 
0.3.4-24.2.0-trunk: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.7.5.img.xz

MrTondy, 2024-11-26:
> First: Thanks for your work Maxim!
> 
> I've got internal wifi working with your images on the klipad50 by replacing the devicetree file with the version, that came with Sovol's original 5.16 kernel. It will lack the spi-dev, but that can be re-enabled with a patch.
> 
> The change (and original dtb source) can be found here: https://netztorte.de/3d/doku.php?id=rk3328-roc-cc_dts#wifipatch
> 
> There is a compiled version of the dtb at the bottom of this page: https://netztorte.de/3d/doku.php?id=rebuilding#internal_wifi (It also shows the required steps for setting up klipper on the klipad50 based on your image, with screen-rotation and enabled beeps).
> 
> I've tested the new .dtb successfully with following versions: 0.3.2-24.2.0-trunk: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.1.76.img.xz 0.3.4-24.2.0-trunk: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.img.xz 0.3.4-24.2.0-trunk: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.7.5.img.xz

Hi did you get it fully working in the end? I'm struggling with the Sovol Klipper Screen to get it up to date and gaining full control of it, and I was thinking of installing a clean OS but I am not skilled enough to compile, debug, etc... so I was wondering if you have a complete and fully working build to share

redrathnure, 2024-11-26:
@MrTondy I am not sure you need to compile something here. It's easier than looks like. 
Basically you would need following steps:
1. install a fresh OS using image from [the message bellow](https://github.com/redrathnure/armbian-mkspi/issues/12#issuecomment-1987135330) (or from [latest release page](https://github.com/redrathnure/armbian-mkspi/releases))
2. then change your DTS file. see instruction on the start of [the page](https://netztorte.de/3d/doku.php?id=rk3328-roc-cc_dts#wifipatch) and line to change [at the end](https://netztorte.de/3d/doku.php?id=rk3328-roc-cc_dts#wifipatch). You may ignore huge part on the middle. 
3. Reboot the system and check how it works. I would recommend to reread https://netztorte.de/3d/doku.php?id=rebuilding#internal_wifi page, which looks like step-by-step instruction how to reinstall Klipper from scratch.