redrathnure/armbian-mkspi #14: Wifi Adapter

rafaelrabaco, 2023-09-19:
### What happened?

I don't know if this is a bug or if it really doesn't have the drivers, but, how do I use a usb wifi?
My adapter is RTL8188GU(RTL8710B)/(VID:PID = 0x0BDA:0xB711)
Appears on the system when I use the `lsusb` command, but I can't connect or scan wifi networks..
I tried installing the drivers but it asks for the kernel header..

### How to reproduce?

`lsusb` - to see connected USB devices
`ifconfig`

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2023-09-25:
I am not sure RTL8188GU is supported by Linux Kernel out of the box. If you need kernel headers you may build an image for your platform. In the output directory you will find deb packages for kernel, headers etc.

rafaelrabaco, 2023-09-26:
Okay, I'll try that!

Do you happen to use the MKS PI + TS35 & ADXL345?
I ask this because when I use it, the screen goes white using the official MKS Pi image, Could this be just a hardware problem?

redrathnure, 2023-09-27:
Yes, I use both and no problems at all.

closed 2023-12-31 by redrathnure