redrathnure/armbian-mkspi #15: LCD TS-35 does not work after updating (September 21, 2023)

PrintMan133, 2023-09-21:
### What happened?

LCD TS-35 does not work after updating

### How to reproduce?

LCD TS-35 does not work after updating

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2023-09-25:
My TS-35 works fine, tested it yesterday with [0.2.1-23.08.0-trunk](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.2.1-23.08.0-trunk) release. I would recommend to check your hardware , especially contacts and try new release images. 
And please double check you have frozen kernel packages. If you update kernel to latest from official Armbian repo you will loose screen and couple of other hardware components.

NexGen-3D-Printing, 2023-10-07:
I have this issue with all of the OS images after Kernal v5, I'm running a generic HDMI "Model-B" display, fires up and works out the box in Kernal v5 but will not work with any of the newer images running Kernal 6 or upwards.

May 15 release, Bullseye Current works, all after this, I get not picture.

PimpstaPee, 2023-12-26:
Touch screen dont work  mks ts35 v1.0 on [https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.0-24.2.0-trunk](url)

redrathnure, 2023-12-26:
@PimpstaPee which image?

PimpstaPee, 2023-12-27:
@redrathnure @redrathnure [Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.6.7.img.xz](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F0.3.0-24.2.0-trunk/Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.6.7.img.xz)
Если Вам будет удобно, можно на английском

redrathnure, 2023-12-31:
@PimpstaPee moved to https://github.com/redrathnure/armbian-mkspi/issues/20

redrathnure, 2023-12-31:
LCD TS-35 works fine

closed 2023-12-31 by redrathnure