redrathnure/armbian-mkspi #16: CAN support in kernel

opiswahn, 2023-09-21:
### Which feature would you like to have?

I tried to use CAN with the current debian image, but it seems it is not compiled into the kernel. Can this be added?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2023-09-22:
Hi [opiswahn](https://github.com/opiswahn), I will look at weekends. However I have no idea how to work with CAN bus support. I would be appreciated for any hints.

opiswahn, 2023-09-23:
Hi, i think it actually is ok after all, I tried the newest bleeding version (both Debian and Ubuntu)and it works fine as a module. I'm not sure why it didn't work with the stable version, maybe I did something wrong. So all good I guess : ☺️

opiswahn, 2023-09-23:
BWT, thx for your work on those images, whey work really great 👍

closed 2023-09-23 by opiswahn

fbeltrao-br, 2024-01-20:
I've tested CAN BUS on MKS Skipr three months ago, and unfortunately all redrathnure images using kernels 5.15 and 6.1 (I didn't tested 6.6) have issues, something causes disconnections on very early print stages. The same configuration on the garbage EOL 5.16 kernel Armbian compiled by MKS, works without issues.

yammaster, 2024-01-23:
Wich image includes the CAN module? I tried the 6.1 curent but I don't see can interface.

redrathnure, 2024-01-23:
OK, after a some reading and trails I would say the both images has CAN support. However you should understand how to configure CAN interface, how to flash/configure the both parts and be aware about current limitation: SKIPR CAN will work only when "main board" and "controller part" are connected via USB interface (will not work when you use default UART connection).

I managed to configure MKSPI <--> U2C <--> BTT EBB36 plate and it works on the speeds up to 500 000 bps (Not 1Mibps as stated in documentation). However it requires to flash the BTT EBB36 plate with right parameters and having proper data cables (they must be twisted, even with short cables). 
I used https://www.klipper3d.org/CANBUS.html + BTT EBB36 docs as entry point. I believe SKIPR configuration should be similar, except you have SKIPR Controller part instead U2C stick.

yammaster, 2024-01-24:
As I'd afraid - the mcu flashing went wrong. Now I have a can0 interface - yeah!

yammaster, 2024-01-25:
I use it in the moment only for my test environment so I haven't issues with non stabel working .... not yet ;-)

fbeltrao-br, 2024-01-27:
> OK, after a some reading and trails I would say the both images has CAN support. However you should understand how to configure CAN interface, how to flash/configure the both parts and be aware [...]

I certainly know how to setup a CAN interface. Did you tried to make a real use of the CAN interface? Here it fails randomly, mostly during first home procedure, it's pretty much unreliable.

I have both MKS Skipr with built-in CAN and MKS Pi with BTT EBB36 and U2C, though I tested only with the Skipr.

redrathnure, 2024-03-16:
@fbeltrao-br, @yammaster  I know it's pretty old discussion, however finally I reach a stage when I  can answer "Yes, I use it in real prints, and yes, it works correctly". 
However interface was semy broken when I had followed official Klipper documentation. So you should double check your configs and adapt them to the used distro. Please check out https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus for more details, if you are  still have interest in CAN bus configuration.

yammaster, 2024-03-17:
> Yeah, just wanted to respond you that no any modprobe magic is required, just a proper Klipper configuration of mcu firmware.
> 
> BTW, is it working stable in your case? There is a message that a print interrupts for an unknown reason.

Hi Alex,
I had a issue with timeout while homing, as you asked me but I solved it with this:

https://klipper.discourse.group/t/communication-timeout-during-homing-probe/3559

regards
René

yammaster, 2024-03-17:
Hi Alex,

I use it with CAN bus for real prints without any issues, accept the timeout while homing (as I discribed in your repository).

 

Regards

René

 
 

Gesendet: Samstag, 16. März 2024 um 16:14 Uhr

SnakeKVC, 2024-11-07:
Path /etc/network/interfaces.d/ not found latest release 
Armbian-unofficial_24.11.0-trunk_Mkspi_bookworm_current_6.6.59.img.xz 
How configure can interface?
device can0 present.

SnakeKVC, 2024-11-07:
> @SnakeKVC you should check how networks is configured in your case. Maybe you should edit `/etc/network/interfaces` or you distro uses NetPlan (/etc/netplan/*) for network management.
Дистрибутив скорее Ваш чем мой :-)
Сеть настроена через NetworkManager.
В предыдущих дистрибутивах я CAN прописывал в /etc/network/interfaces.d/can0
теперь папка /etc/network/interfaces.d отсутствует.
Предлагают поместить конфиг с настройками интерфейса в /etc/systemd/network/ и использовать systemd-networkd
но совместное использование systemd-networkd и NetworkManager вызывает проблемы.

redrathnure, 2024-11-07:
@SnakeKVC Wrong claim in the wrong place.

Armbian project builds a right Linux kernels for some SBCs (Raspberry Pi, Orange Pi etc). The project is not responsible neither for the Linux distributive (Ubuntu Jammy, Debian Bookworm etc) nor for 3. party manuals (Klipper, CANbus, KIAUH etc). 

This project has an even narrower goal - to assemble a kernel just for a few specific MKS boards. Handling distributive related changes or any other Klipper related manuals is our of the scope, sorry.

So, a few hints for you:
1.`Path /etc/network/interfaces.d/ not found` means something was changes in the latest version of Linux distro. Debian related forums would explain more.
2. CANBus configuration is kinda Klipper related topic. You may try to ask in their Discord or Killer forum.
3. The Google Translator does not always work correctly, so English is best choice to ask questions.

SnakeKVC, 2024-11-07:
@redrathnure 
Thank you very much.
I have already figured out the problem.
I disabled NetworkManager and used systemd-network

redrathnure, 2024-11-07:
@SnakeKVC How it works with KlipperScreen? I mean is it OK to use KlipperScreen without NetworkManager?

SnakeKVC, 2024-11-07:
@redrathnure Yes, klipperscreen work is fine