redrathnure/armbian-mkspi #17: Help with serial port for custom skipr board from Elegoo

SHexplorer, 2023-10-23:
### Which feature would you like to have?

Hi,

maybe you can help me. Im currently trying to run your image for my Elegoo Neptune 4, which uses a custom mainboard based on the skipr. The default image is working just fine but i'm missing one serial port to access the printers touchscreen. Here is the dmesg output of the original image:

`[   0.000000] Kernel command line: root=UUID=5dcae443-0796-4029-ae6c-bf5bd2a37067 console=uart8250,mmio32,0xff130000 console=tty0 rootflags=data=writeback rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 bootsplash.bootfile=bootsplash.armbian`
`[    0.002596] printk: console [tty0] enabled`
`[    1.580954] ff110000.serial: ttyS0 at MMIO 0xff110000 (irq = 19, base_baud = 1500000) is a 16550A`
`[    1.583232] ff120000.serial: ttyS1 at MMIO 0xff120000 (irq = 20, base_baud = 1500000) is a 16550A`
`[    1.585297] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 21, base_baud = 1500000) is a 16550A`
`[    1.679961] printk: console [ttyS2] enabled`

and this is the output of your image:

`[    0.000000] Kernel command line: root=UUID=3de17528-b19a-4b06-896f-307ffb69b32e rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=ccac1dbb-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1`
`[    0.001098] printk: console [tty1] enabled`
`[    0.927506] ff110000.serial: ttyS0 at MMIO 0xff110000 (irq = 21, base_baud = 1500000) is a 16550A`
`[    0.930697] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 22, base_baud = 1500000) is a 16550A`
`[    0.930843] printk: console [ttyS2] enabled`
`[    7.361638] systemd[1]: Created slice system-getty.slice - Slice /system/getty.`
`[    7.369790] systemd[1]: Created slice system-serial\x2dgetty.slice - Slice /system/serial-getty.`
`[    9.052335] systemd[1]: Found device dev-ttyS2.device - /dev/ttyS2.`

Any idea how I can make the ttyS1 port working?

Thanks in advance!

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2023-10-23:
Hi, sorry I have no experience with your board. In general "screen connection" consist of 2 parts:
1. LCD screen driver. TS35 v1.0 uses st7796 controller,  E.g. see [custom kernel module](https://github.com/redrathnure/armbian-mkspi/blob/main/userpatches/kernel/rockchip64-current/zz-mkspi-kernel-st7796-module.patch) and [related DTS declaration](https://github.com/redrathnure/armbian-mkspi/blob/dde54bc70c80f85335fe851735717ec3c4a95b50/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L169-L171)
2. Touchscreen driver. TS35 v1.0 uses TSC2046 module, please see [related DTS declaration](https://github.com/redrathnure/armbian-mkspi/blob/dde54bc70c80f85335fe851735717ec3c4a95b50/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch#L160-L167)
Of course you should know pins and other schematic details for you board. If you have working image, you may try to look at related DTS file and guess pinout and related configuration (please refer[this part](https://github.com/redrathnure/armbian-mkspi#how-to-rotate-screen) how to convert dtb file to human readable dts one) .

SHexplorer, 2023-10-23:
The touchscreen doesn't matter. It is just a serial connection (its a nextion touchscreen so it only communicates with serial commands) and I'm writing my own code for it. I just need it available as normal serial device. Maybe I should test in with the original skipr image from makerbase and see, if it's working there.

redrathnure, 2023-10-24:
Please try to unpack DTS files and compare related sections.

SHexplorer, 2023-10-24:
Hey, thank you for your help. I managed to get the serial port working. For now I just replaced your rk3328-roc-cc.dtb with the original in the boot partition and started the board. There are quite some changes Elegoo did with their custom board.

Anyway, thank you for your help and tips, that let me understand a bit more of armbian and linux kernel design!

One last question: When I want to build my own image, how would I commit these changes? Do I have to create a .patch file for these changes it the DTS file like [here](https://github.com/redrathnure/armbian-mkspi/blob/dde54bc70c80f85335fe851735717ec3c4a95b50/userpatches/kernel/rockchip64-edge/zz-mkspi-kernel-rockchip64-edge.patch)?

redrathnure, 2023-10-24:
Perfect, this means that kernel and all required modules are already here. 

First of all good editor with good diff tool are your best friends on this journey. You need to understand what is different in DTS declaration, e.g. comparing mkspi and your working files. Seems chineese engineers are not "too creative" and usually do not introduce significant changes to reference design. Most likely you would have a few minor changes comparing e.g. with MKSPI or original Renegate boards.

About your custom changes:
1. I done it in **very wrong** way. Means new config is placed into userpatches dir and basically I override Renegate board by MKSPI declaration. If you will contribute these changes to official Armbian repo, you must introduce new board type with separate configuration, patches etc. Otherwise you, you may put it into userpatches and override MKSPI changes :)
2. Please pay attention to `current` and `edge` suffixes. At the end it affects result kernel version. Not sure what is the diffrence between them, but now the both works fine.
3. Looks like target distribution is just matter of taste. If you have problem with kernel or/and DTS config you will see it in both Debian and Ubuntu images. So just pick whatever you are familiar with.
4. [How to Build](https://github.com/redrathnure/armbian-mkspi/blob/dde54bc70c80f85335fe851735717ec3c4a95b50/README.md#how-to-build) section has a few useful commands.

When you realize necessary diffs/changes and if you will have interest on this image after few iteration, we may think about proper structure of this repo. I means keeping patches in proper places and support two different boards on the same time without destroying each other.

closed 2024-11-04 by redrathnure