redrathnure/armbian-mkspi #19: Any tips on setting a pin state?

moswald14, 2023-11-23:
### What happened?

I just want to set a pin to activate a relay. I have tried many things, shell scripts, python, 99% suggestions from chat-gpt but nothing works. I'm trying to use 1_C7 and 1_C3 on gpiochip1, I found that there is a folder named gpiochip1 under /sys/class/gpio/gpiochip32/device but I have no idea what I'm doing. I'm trying to set openHAB up to turn on a relay that runs a pump for 5 minutes and I will need some pins to run my moisture sensors off of. I feel like I'm going to need to buy a better supported pi but any help would be appreciated. I have plenty of time to troubleshoot, so just tell me what I should do.

### How to reproduce?

I've tried pyGPIO, RPi.GPIO, pigpiod with varying unsuccess. either libraries are just not supported or python just wouldn't work with the libraries. maybe we could work with [pyGPIO and make a mapping](https://github.com/chwe17/pyGPIO/tree/master/pyGPIO/gpio/mapping) for this board? I'm not sure, I'm a pi noob.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FabioM3, 2023-11-24:
You have activated the io pins?

moswald14, 2023-11-24:
can you give me a link? im finding absolutely nothing about this board. the documentation is absolute crap

onolox, 2023-11-24:
Here in issues I've made a tutorial.

moswald14, 2023-11-25:
awesome, I will give it a shot, though I'm not trying to use the pins through klipper

moswald14, 2023-11-25:
awesome, thanks so much for putting me in the right direction. here is the magic command after enabling gpio:
`sudo gpioset gpiochip1 23=1
`
Turns gpio pin 23 on gpiochip1 on, then switch to 0 to turn off, neat

closed 2024-11-06 by redrathnure