redrathnure/armbian-mkspi #21: Unable to install Kernel Headers

AdamTuraj, 2024-02-02:
### What happened?

I am trying to install Kernal headers to download a driver to a USB Wifi stick (https://github.com/McMCCRU/rtl8188gu). The issue is there are no headers for the current Bookworm Edge 6.6.9 version. When running the make file for the driver, it points to a folder whose parent exists, but the build folder does not. If there's any way to get these kernel headers built that would be much appreciated.

### How to reproduce?

git clone https://github.com/McMCCRU/rtl8188gu.git
cd rtl8188gu
make

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-02-02:
IMO the easiest way is to build your custom image. There will be few additional files in the output directory including kernel headers package.
Option 2, which **might** work is to find a right package on the official Armbian dep repo. You would need package for Renegade board and exactly the same kernel version. However I have no idea how realistic is to found this combination and would it work at all.

Bit more details:
this is custom build with customized kernel. Most likely you need a kernel headers which was used exactly for this build/kernel and most likely for the same distro. So I would just install VirtualBox and build a needed image with all additional deb packages.

AdamTuraj, 2024-02-03:
I'm not the most savvy when it comes to building OS's, etc. Just brought it up to see if this is something you can easily fix. Maybe someone will come around with a fix at one point although that's not me

Phil1988, 2024-02-03:
+1 for this.

I also have no experience in building an OS, nor am I a programmer, so I hope someone with more knowledge can fix this :)

ChristineBoersen, 2024-02-03:
This is the whole reason I have to build my own to begin with (that odd version of the Realtek driver that is "sorta" upstream kernel) and no one seems to set their default outputs to put source and header on them (multiple people's builds, not singling this repository out there).

redrathnure, 2024-02-04:
@ChristineBoersen which artifact do we need from build? I may rebuild latest release and upload needed packages.

redrathnure, 2024-02-04:
@AdamTuraj , @Phil1988 could you please check https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.2-24.2.0-trunk images. Please write back results:
1. have you managed to build custom driver/module?
2. Did you use `linux-headers-xxx.deb` package? Should I publish them for the next releases?

AdamTuraj, 2024-02-04:
@redrathnure I only have one EMMC and would rather not reinstall everything. I was able to get it working although I had to defreeze the kernel. Everything seems stable although not the most ideal. If @Phil1988 has any issues I can pull it out, back everything up and try it out. Although I just got my printer working with clean firmware and need to catch up on some prints.

redrathnure, 2024-02-04:
You may try a new image on SD card (without erasing already worked system). In any case it would be interested to see your step-by-step instructions to build https://github.com/McMCCRU/rtl8188gu.git module. Guess it would be interested for other peoples.

AdamTuraj, 2024-02-04:
I have a Qidi printer and they only pull the OS from the onboard EMMC. Unfortunately cannot use an SD card

redrathnure, 2024-02-04:
Hm... now I'm bit confused.... which board in it? Is it the similar to MKSPI?

AdamTuraj, 2024-02-04:
The board is this:
https://qidi3d.com/en-ca/products/x-max-3-x-plus-3-x-smart-3-motherboard?variant=41682606588044

From what I understand Makerbase made it for them or maybe they sell something similar. Not really sure

ChristineBoersen, 2024-02-05:
> Hm... now I'm bit confused.... which board in it? Is it the similar to MKSPI?

Sorry I wasn't trying to be confusing, just giving a normal justification for including headers/sources is drivers that you must build before you have network (without playing games) to get people to first network connection more successfully. I'll be posting the (mostly copy/paste) instructions about the wirelesss to a different post soon.

redrathnure, 2024-02-05:
It's OK. I just have not expected that this image is able to work on something except MKSPI board.

Phil1988, 2024-02-05:
> Hm... now I'm bit confused.... which board in it? Is it the similar to MKSPI?

Its asort of MKS SKIPR board which is also similar to the MKSPI.
But they didnt include the SD card.

I am currently very busy, so I cant test this for the next ~2-3 weeks.

BTW: Blame me for others to report issues here on the MKS based 3d boards :)
At least I was the first who wrote some instructions on how to get your OS image working on these boards :D

redrathnure, 2024-02-05:
Actually it's perfect that this image is useful for more peoples. Perhaps we should collect list of boards which works (or might work) with this build and related known issues. So would be nice to understand name of board and needed steps to make it work. For example I assume you have slightly modified DTS files and probably some extra configuration steps.

Feel free to publish this info here (I will add it to the readme and release notice) or open PR with related changes.

Phil1988, 2024-02-05:
There was little to no modifications needed :)

You can have a closer read to my tutorial here:
https://openqidi.com/books/build-entire-operating-system-and-klipper-from-scratch/page/freeqidi

I expect these boards to work directly with no changes:
MKS PI
MKS SKIPR
QIDI X-4 and X-6 Mainboards (made by MKS for X-Plus 3 and X-Max 3)

redrathnure, 2024-02-05:
See https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.2-24.2.0-trunk, including pre-installed `usb-modeswitch` and `build-essential` packages.

ohyeah8000, 2024-02-18:
Hello Maxim,

first of all thank you very much for this really nice repo and your work!
It is a very useful piece of software for an increasing number of people I know.

I'm not a programmer but user of one of these Qidi X-Max 3 machines with an MKS PI X-4 board.
Last year I startet to search for alternative ways and your work was mentioned many times. 

However currently I still struggle to implement RTL8188GU support and tried many things.
Good you have build essentials and mode switch included but it's still not possible for me to activate wireless connectivity.
 
iwconfig shows no wireless extensions and this means the driver is a missing part for the kernel. I tried current and also edge kernels now.

What is the right way to enable wifi then?
DKMS and some preparation? Or is something else missing?

Thanks a lot for good advice!

PS: What is the reason you don't have the edge kernel in current releases present anymore?

redrathnure, 2024-02-18:
Hi @ohyeah8000, thanks for the feedback. 

I does not have access to your board and this WIFI dongle, so may only guessing:
1. Looks like kernel 6.4 and later already support RTL8188GU devices. Using `rtl8xxxu` module. If you use Edge images (which has kernel v6.6. or v6.7) you may try follwoing:
1.1. Check that your device is connected, e.g. by `lsusb`
1.2. Ensure that it is not in CD/storage mode. Otherwise use `usb_modeswitch` to switch to model mode
1.3. Try to load `rtl8xxxu` module. `sudo lsmod | grep rtl`, `sudo modprobe rtl8xxxu` and so on
1.4. If no result, check dmesg. E.g. by `sudo dmesg | grep rtl8xxxu` or `dmesg | grep 8188`
1.5. Perhaps it worth to check https://github.com/McMCCRU/rtl8188gu/issues/66#issuecomment-1765821197 discussion

2. If you use kernel <= 6.4 (e.g. 6.1 from current image), then seems you need to build custom module. E.g. https://github.com/McMCCRU/rtl8188gu.git or https://github.com/lwfinger/rtl8188gu. Or one of backports: https://github.com/lwfinger/rtl8xxxu/tree/main, https://github.com/wandercn/RTL8188GU. Workflow very similar to previous section:
2.1. Check that device is visible and is not in CD/storage mode
2.2. Build module and install it. E.g. `mame && sudo make install`, probably reboot
2.3. Check module status and/or try to load it. `sudo lsmod | grep 8188` and `sudo modprobe -r 8188gu`
2.4. If does not work, check dmesg and/or ask people who already done this task :)

PS: please ignore 0.3.2 release, It's better to look at the latest https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.3-24.2.0-trunk one, which has the both current and edge images. Please note edge image has weird issue with TS35 touchscreen.

ohyeah8000, 2024-02-18:
Thanks man! Oh, shoot, installed 2 different 0.3.2 versions today including Klipper and everything. I have a lot of practice now and I guess a new approach will be even faster... ;)

Of course I tried almost everything you mentioned above but didn't succeed in the end. I really think the Kernel version is important. I'll let you know how it turns out and give fedback after the next attempt.

redrathnure, 2024-02-19:
OK, just FYI, new release is out including current with 6.6 kernel. please check https://github.com/redrathnure/armbian-mkspi/releases page.

lucymorn, 2024-03-04:
A bit off topic, but any idea how to get headers for `5.16.20-rockchip64`? I am trying to get `aic8800` working, but when trying to build the driver -
```make[1]: *** /lib/modules/5.16.20-rockchip64/build: No such file or directory.  Stop.```

redrathnure, 2024-03-05:
Perhaps it's a question to official MKS repo.

Guilouz, 2024-03-17:
@Phil1988 Your link is dead

https://openqidi.com/books/build-entire-operating-system-and-klipper-from-scratch/page/freeqidi

Phil1988, 2024-03-19:
@Guilouz Yes. It gave quite some confusion about the creator so I published it at my git:
https://github.com/Phil1988/FreeQIDI

Have fun, give it a star and stuff :D

frap129, 2024-04-02:
2 things to report. First, images built from this repo work on the new X-7 mainboard that comes in the QIDI Q1 Pro. Second thing to report is that WiFi works on it after building the aic8800 driver from [here](https://github.com/lynxlikenation/aic8800). The initial image has to built with `INSTALL_HEADERS=yes BUILD_KSRC=yes INSTALL_KSRC=yes`, and  `eject` must be installed. I have added the steps to build and install the driver at build time in `userpatches/customize-image.sh` in my fork, which you can find [here](https://github.com/frap129/armbian_qidi-q1-pro/blob/9f08ed2023b0949025b11ebdba212250aa1202d9/userpatches/customize-image.sh#L32).