redrathnure/armbian-mkspi #22: Addition CAN-BUS MCP2515 SPI

Jarewa, 2024-02-03:
### Which feature would you like to have?

Is it easy to add the can module after spi in armbians?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2024-02-03:
Personally I have no idea what this and how make it worked. May be somebody has a better understanding:)

ChristineBoersen, 2024-02-05:
I need to look through my old notes,  but it should be identical from your POV, by using a DTS overlay. 
Read how to do it to a "Renegade" board, and it will get you mostly there.
I have to look back through my notes, as I've done it once with one of the hat type boards on a Renegade, but, from my memory it was along the lines of 
- 5.x kernel wasn't happy with MCP2515 + rk2238 + children
    - This was all solved in 6.1.x+ when legacy was merged to current for rk3328
- Renegade config worked in 6.1.x+
    - I can't remember if I had to use a custom overlay or if I used CanBoot at that point