redrathnure/armbian-mkspi #23: For the default (not minimal) install to include enough to build the OS or drivers to get to first internet faster

ChristineBoersen, 2024-02-05:
### Which feature would you like to have?

For the default (not minimal) install to include enough to build the OS or drivers to get to first internet faster

Goals
- Have build-essential included in non-minimal install
- Have usb-modeswitch included in non-minimal cli (it is part of desktop already due to xmod*)

#21 Continuation of justification for installing kernel headers + source for non-minimal install

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

ChristineBoersen, 2024-02-05:
Also, nearly, if not all, of this, is needed to build kiauh which is a common way to install klipper and friends, hence, it is much quicker to get the compiled on your host vs on the mkspi cpu  (fast build machine).

redrathnure, 2024-02-05:
Done, https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.2-24.2.0-trunk

closed 2024-02-05 by redrathnure