redrathnure/armbian-mkspi #25: The build was too slow

HasibTheNeWb, 2024-02-25:
### Which feature would you like to have?

Can anyone give the already made output\images for Debian? Actually i waited almost 3 hours to complete the build. And i cancelled it. If anyone already made the img file for the Debian please share with me. 😥

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2024-02-25:
Hi, have you checked a [release page](https://github.com/redrathnure/armbian-mkspi/releases)? E.g. [0.3.4-24.2.0-trunk](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.4-24.2.0-trunk) has images for Bookworm and Bullseye distros.

HasibTheNeWb, 2024-02-25:
> Bullseye

Oh, sorry id didn't notice. Is it possible to install open media vault in this version?

redrathnure, 2024-02-25:
Should be no problem. It's usual Debian or Ubuntu, whatever you choose.

HasibTheNeWb, 2024-02-25:
Bro, can you tell me whats is the login username and password, i just installed, but can't login with ssh- username: mks pass: makerbase

HasibTheNeWb, 2024-02-25:
NVM.
Thank you, man.

HasibTheNeWb, 2024-02-25:
Getting this error when trying to install open media vault in Mks pi 

-bash: /usr/bin/sudo: Input/output error
-bash: /usr/bin/sudo: Input/output error

redrathnure, 2024-02-25:
have no idea what this. Perhaps you should try an Ubuntu image, they are tested better.

closed 2024-11-04 by redrathnure