redrathnure/armbian-mkspi #27: mcu 'rpi': Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu'

wnto, 2024-05-19:
### What happened?

I installed 

Armbian-unofficial_24.2.0-trunk_Mkspi_jammy_current_6.6.17.img.xz

on a new skipr motherboard and followed the 

https://github.com/NexGen-3D-Printing/MKS-SKIPR-SETUP

kiauh works, I was able to flash the MCU (mks_skipr.bin file in sd card renamed to .CUR), Mainsail and klipper screen (pi-ts35) work, wifi works). 

but I get the error 

mcu 'rpi': Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu' 

in klippy.log.

running 

systemctl status klipper-mcu

I get:
Unit klipper-mcu.service could not be found.

service klipper-mcu start

and 
a8@mkspi:~$ service klipper-mcu start
Failed to start klipper-mcu.service: Unit klipper-mcu.service not found.
a8@mkspi:~$ Fa

finally,

a8@mkspi:~$ sudo sysctl -w kernel.sched_rt_runtime_us=-1
kernel.sched_rt_runtime_us = -1
a8@mkspi:~$ systemctl status klipper-mcu
Unit klipper-mcu.service could not be found.

I read nexgen suggesting using the bullseye version, but that was in 2023. I will try that, anyway and see if it helps, but wanted to report this here for now. 
Pointers are welcome.

### How to reproduce?

Command you gave to the framework!running 

systemctl status klipper-mcu

I get:
Unit klipper-mcu.service could not be found.

service klipper-mcu start

and 
a8@mkspi:~$ service klipper-mcu start
Failed to start klipper-mcu.service: Unit klipper-mcu.service not found.
a8@mkspi:~$ Fa

finally,

a8@mkspi:~$ sudo sysctl -w kernel.sched_rt_runtime_us=-1
kernel.sched_rt_runtime_us = -1
a8@mkspi:~$ systemctl status klipper-mcu
Unit klipper-mcu.service could not be found.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://pastebin.com/yvDQStwj

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wnto, 2024-05-19:
For further reference, I found the following, from which I took the command used above, but this method failed me.

https://klipper.discourse.group/t/klipper-mcu-service-fails-to-start/12219

Same method also metioned more briefly in 
https://klipper.discourse.group/t/armbian-kernel-klipper-host-mcu-got-error-1-in-sched-setschedule/1193

redrathnure, 2024-05-19:
Hi, klipper_host_mcu is a klipper service which works on MKS PI/RPI. Usually it's used to control CPU temp or map extra pins, buttons or sensors which connected directly to MKS PI/RPI. For example you may check https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#adxl345spi-usage or https://www.klipper3d.org/RPi_microcontroller.html articles

wnto, 2024-05-20:
ah, this explains a bit. I did find these references before and didn't know what to do about them, as I am not trying to use the pi side of the skipr board as a secondary mcu. Thank you for your prompt reply. 
I wonder what might be triggering this error, though? 
I am just trying to connect the host to the main MCU, but without any success so far, and the error that I have is this mcu 'rpi': Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu'.
Any ideas on how to suppress it?

closed 2024-05-20 by wnto