redrathnure/armbian-mkspi #28: AP6212 WIFI not working

icode, 2024-06-03:
### What happened?

I use kingroon kp3s pro v2 printer, it is [kingroon-kp-cheetah-v2.2](https://kingroon.com/products/kingroon-kp-cheetah-v2-0-mainboard-for-klp1-3d-printer?variant=45178992296154) mainboard.

Now, i not found wifi interface, and i don't know how to fixed.

i use this img
[Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.img.xz](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F0.3.4-24.2.0-trunk/Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.img.xz)

### How to reproduce?

[image]

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-06-03:
Hi, it's hard to say. I do not have this board and cannot test an image. Perhaps somebody here can help you.

Lebensgefahr, 2024-12-08:
> ### What happened?
> 
> I use kingroon kp3s pro v2 printer, it is [kingroon-kp-cheetah-v2.2](https://kingroon.com/products/kingroon-kp-cheetah-v2-0-mainboard-for-klp1-3d-printer?variant=45178992296154) mainboard.
> 
> Now, i not found wifi interface, and i don't know how to fixed.
> 
> i use this img [Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.img.xz](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F0.3.4-24.2.0-trunk/Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.img.xz)
> ### How to reproduce?
> <img alt="image" width="239" src="https://private-user-images.githubusercontent.com/972959/335966728-c9e0da8c-8394-40a8-b1c8-6a55b4429e92.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3MDExMTAsIm5iZiI6MTczMzcwMDgxMCwicGF0aCI6Ii85NzI5NTkvMzM1OTY2NzI4LWM5ZTBkYThjLTgzOTQtNDBhOC1iMWM4LTZhNTViNDQyOWU5Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOFQyMzMzMzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDZjN2FlMzdmYmZlOWE2MTE1NDMwNzlmZDI1ZGIwYTNlZTFhM2I2YTNhMzQ3YTRlZDM1ZTYwYmVhZWNiNWVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.kNghQPvUZXsp5mgZuZB9hEmAAcPim4ZCut_eVxnwmno">
> ### Branch
> 
> main (main development branch)
> ### On which host OS are you observing this problem?
> 
> Other
> ### Relevant log URL
> 
> _No response_
> ### Code of Conduct
> 
>     * [x]  I agree to follow this project's Code of Conduct

Kingroon modified [rk3328-roc-cc.dtb](https://github.com/Lebensgefahr/kingroon_kp3s_pro_v2/blob/main/files/userpatches/overlay/rk3328-roc-cc.dtb) file on the place. You need to turn it back.  Just copy it into it's place and restart your system.

karabas2011, 2024-12-18:
ap6212 works if replaced with kingroon's  rk3328-roc-cc.dtb renamed to rk3328-mkspi.dtb

karabas2011, 2024-12-19:
Does your touchscreen work?

Lebensgefahr, 2024-12-19:
> Does your touchscreen work?

There is no touchscreen on this printer.