redrathnure/armbian-mkspi #29: I would like to update the kernel on an older install

dancat, 2024-06-11:
### Which feature would you like to have?

Hi, not really a request, but a question.
I have an older installation working but based on the older kernel (the one that was freezed on the initial installation 'Welcome to Armbian 23.02.2 Jammy with bleeding edge Linux 6.1.10-rockchip64').
I updated regularly all the packages and there were no problems until now.
I now would like to update the kernel so I can also move to a CAN-bus implementation. Is there a way to update the kernels on my current installation to the last release without reflashing the oimage and reinstalling/configuring everything?

All the best,
Daniel

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2024-06-11:
HI @dancat , I am not 100% sure... in theory you may take the latest kernel packages and try to install them.... however I would recommend 'backup - reinstall - restore from backup' approach. I personally use just another one microSD to do test installation and heve possibility to rolling back in case of fails (just insert old SD).

BTW, are you sure old kernel does not have proper CAN-bus support?

dancat, 2024-06-13:
Thanks for feedback!

I am not sure about the CAN-bus support, I will check. I think I have one of the very early releases :)

Best Regards,
Daniel

closed 2024-11-04 by redrathnure