redrathnure/armbian-mkspi #31: Disable kernel console debug messages for ttyS2

Phil1988, 2024-07-14:
### What happened?

I cant get my serial port ttyS2 to work as I want to.
I am using this OS:
Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.6.7.img.xz

Its seems that even if all services being shut down/disabled/stopped, there is something using it.

How do I know?
My oscilloscope shows something.

<details>
  <summary>TL;DR - nothing is actively running on the ttyS2</summary>
  
No process is using the ttyS2:
```
mks@mkspi:~/X3seriesLCD$ sudo lsof | grep ttyS*
systemd-l  694                      root    7r      REG               0,24      4096       5312 /sys/devices/virtual/tty/tty0/active
systemd-l  694                      root   17u      CHR                4,6       0t0         29 /dev/tty6
python     904                       mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     904                       mks   18uW     CHR              166,0       0t0        690 /dev/ttyACM0
python     904 1116 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     904 1116 python           mks   18uW     CHR              166,0       0t0        690 /dev/ttyACM0
python     904 1587 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     904 1587 python           mks   18uW     CHR              166,0       0t0        690 /dev/ttyACM0
python     904 1588 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     904 1588 python           mks   18uW     CHR              166,0       0t0        690 /dev/ttyACM0
python     904 1617 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     904 1617 python           mks   18uW     CHR              166,0       0t0        690 /dev/ttyACM0
python     904 1618 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     904 1618 python           mks   18uW     CHR              166,0       0t0        690 /dev/ttyACM0
Xorg       952                      root   12u      CHR                4,2       0t0         25 /dev/tty2
Xorg       952 1229 InputThre       root   12u      CHR                4,2       0t0         25 /dev/tty2
agetty    1028                      root  cwd       DIR              179,2      4096          2 /
agetty    1028                      root  rtd       DIR              179,2      4096          2 /
agetty    1028                      root  txt       REG              179,2    134480      35604 /usr/sbin/agetty
agetty    1028                      root  mem       REG              179,2   3407104       9498 /usr/lib/locale/locale-archive
agetty    1028                      root  mem       REG              179,2    353616       3899 /usr/lib/locale/C.utf8/LC_CTYPE
agetty    1028                      root  mem       REG              179,2   1651472      31068 /usr/lib/aarch64-linux-gnu/libc.so.6
agetty    1028                      root  mem       REG              179,2    202912      30789 /usr/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1
agetty    1028                      root  mem       REG              179,2     27028      59819 /usr/lib/aarch64-linux-gnu/gconv/gconv-modules.cache
agetty    1028                      root    0u      CHR                4,1       0t0         24 /dev/tty1
agetty    1028                      root    1u      CHR                4,1       0t0         24 /dev/tty1
agetty    1028                      root    2u      CHR                4,1       0t0         24 /dev/tty1
agetty    1028                      root    4r  a_inode               0,15         0       2074 inotify
sudo      5264                      root    8u      CHR                5,0       0t0         15 /dev/tty
sudo      5266                      root    8u      CHR                5,0       0t0         15 /dev/tty
```

no systemd-services running;
```
mks@mkspi:~/X3seriesLCD$ systemctl list-units | grep serial
  sys-devices-platform-ff110000.serial-tty-ttyS0.device                                                                  loaded active plugged   /sys/devices/platform/ff110000.serial/tty/ttyS0
  sys-devices-platform-ff120000.serial-tty-ttyS1.device                                                                  loaded active plugged   /sys/devices/platform/ff120000.serial/tty/ttyS1
  sys-devices-platform-ff130000.serial-tty-ttyS2.device                                                                  loaded active plugged   /sys/devices/platform/ff130000.serial/tty/ttyS2
  sys-devices-platform-serial8250-tty-ttyS3.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS3
  sys-devices-platform-serial8250-tty-ttyS4.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS4
  sys-devices-platform-serial8250-tty-ttyS5.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS5
  sys-devices-platform-serial8250-tty-ttyS6.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS6
  sys-devices-platform-serial8250-tty-ttyS7.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS7
  system-serial\x2dgetty.slice                                                                                           loaded active active    Slice /system/serial-getty
```
  
</details>

Permissions look good:
```
mks@mkspi:~/X3seriesLCD$ ls -l /dev/ttyS2
crw-rw---- 1 root dialout 4, 66 14. Jul 15:04 /dev/ttyS2
```

but after every reboot they change to
`crw-rw---- 1 root tty 4, 66 13. Jul 18:26 /dev/ttyS2`

But the kernel messages seems to be configured to use ttyS2 as console:
```
mks@mkspi:~/X3seriesLCD$ dmesg | grep ttyS2
[    0.000000] Kernel command line: root=UUID=6e0fe519-a43a-40fe-a5c3-aa853b368360 rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=2ceabfd6-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[    1.038417] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 23, base_baud = 1500000) is a 16550A
[    1.038623] printk: console [ttyS2] enabled
[    7.471293] systemd[1]: Expecting device dev-ttyS2.device - /dev/ttyS2...
[    8.510699] systemd[1]: Found device dev-ttyS2.device - /dev/ttyS2.
```

How can I shut ther kernel down so its not using `console=ttyS2,1500000`?
I simply want to use this serial port for serial communication :)

Not related:
And it would be great of you can change the links from:
https://openqidi.com/books/build-entire-operating-system-and-klipper-from-scratch/page/freeqidi
to 
https://github.com/Phil1988/FreeQIDI

See here:
https://github.com/redrathnure/armbian-mkspi/releases
[image]

and there
https://github.com/redrathnure/armbian-mkspi
[image]

### How to reproduce?

-

### Branch

other

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-07-14:
Most likely the `/dev/ttyS2` is an UART2 port, which is connected to CH340C IC (USB Type C debug port). And most likely it's configured somehow via `console`/`consoleargs` boot arguments. UnfortunatelyI am not sure how to change it in Armbian. Perhaps something should be added into `/boot/armbianEnv.txt` files, and probably you should check official Armbian forum for instructions how to do this in right way.

BTW I found a file with following content:
```
if test "${console}" = "display" || test "${console}" = "both"; then setenv consoleargs "console=tty1"; fi
if test "${console}" = "serial" || test "${console}" = "both"; then setenv consoleargs "console=ttyS2,1500000 ${consoleargs}"; fi
```
Perhaps you may avoid second line condition explicitly declaring `setenv console "display"`...

Phil1988, 2024-07-17:
@redrathnure 
Thanks for your fast reply.
I was not successfull in working this out.

Im running the MKS OEM maiboard for Qidi named X-6 (similar to X-4).
It has serial pins connected to a "serial display". 
The stock OS works and I quessed it was ttyS2 and tested.
There are high chances that acuallty ttyS1 is for this port and what I am measuring with the oscilloscope are not signals from the kernel messages but anything else on ttyS1.

The armbianEnv.txt cotains `overlays=rk3328-i2c0 rk3328-uart1` which may be right.
I also tested `overlays=mkspi-uart1` in the /boot/armbianEnv.txt that frap129 suggested here: https://github.com/redrathnure/armbian-mkspi/pull/26
But it was not successfull.

@frap129
I only want to use the onboard serial pins. 
In the meantime I believe that they are actually ttyS1, but neither with ‘rk3328-uart1’ nor with ‘overlays=mkspi-uart1’ does ttyS1 work. 
At best I get this error:
```
raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'Input/output error')
```

I can measure signals on the pins, even if nothing should actually access the serial ports. 
I can not establish any serial connection with my script, but I know that the script does work properly as it was tested with an USB TTL adapter and on other hardware, where I exactly know the serial port and how to enable it :)

I hope you can help.

frap129, 2024-07-17:
I made `mkspi-uart1` while working on images for the QIDI X-7 (Q1 Pro mainboard), which uses ttyS1 for the screen, so I would say it's likely that the X-6 uses the same. For reference, I am using a patched version of the [nextion](https://github.com/frap129/nextion) python library for interacting with the display.

Could you share more details on how your script connects, like baud rate? Also, have you checked that your user has permissions on /dev/ttyS1?

Phil1988, 2024-07-17:
I do also think that the X-6 uses the same as the X-7.
My script connect like KlipperLCD on ttyS1/ttyS2 (testing both) with a baud rate of 115200.

Permissions look good for both:
```
mks@mkspi: ls -l /dev/ttyS*
crw-rw---- 1 root dialout 4, 64 17. Jul 22:59 /dev/ttyS0
crw-rw---- 1 root dialout 4, 65 17. Jul 19:17 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 17. Jul 19:17 /dev/ttyS2
crw-rw---- 1 root dialout 4, 67 17. Jul 19:17 /dev/ttyS3
crw-rw---- 1 root dialout 4, 68 17. Jul 19:17 /dev/ttyS4
crw-rw---- 1 root dialout 4, 69 17. Jul 19:17 /dev/ttyS5
crw-rw---- 1 root dialout 4, 70 17. Jul 19:17 /dev/ttyS6
crw-rw---- 1 root dialout 4, 71 17. Jul 19:17 /dev/ttyS7
```

nothing uses them:
```
mks@mkspi:~$ systemctl list-units | grep serial

 sys-devices-platform-ff110000.serial-tty-ttyS0.device                                                                  loaded active plugged   /sys/devices/platform/ff110000.serial/tty/ttyS0
  sys-devices-platform-ff130000.serial-tty-ttyS2.device                                                                  loaded active plugged   /sys/devices/platform/ff130000.serial/tty/ttyS2
  sys-devices-platform-serial8250-tty-ttyS1.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS1
  sys-devices-platform-serial8250-tty-ttyS3.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS3
  sys-devices-platform-serial8250-tty-ttyS4.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS4
  sys-devices-platform-serial8250-tty-ttyS5.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS5
  sys-devices-platform-serial8250-tty-ttyS6.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS6
  sys-devices-platform-serial8250-tty-ttyS7.device                                                                       loaded active plugged   /sys/devices/platform/serial8250/tty/ttyS7
```

```
mks@mkspi:~$ sudo lsof | grep ttyS*
[sudo] Passwort für mks:
systemd-l  715                      root    7r      REG               0,24      4096       5260 /sys/devices/virtual/tty/tty0/active
systemd-l  715                      root   17u      CHR                4,6       0t0         29 /dev/tty6
python     916                       mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     916                       mks   18uW     CHR              166,0       0t0        661 /dev/ttyACM0
python     916 1107 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     916 1107 python           mks   18uW     CHR              166,0       0t0        661 /dev/ttyACM0
python     916 1613 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     916 1613 python           mks   18uW     CHR              166,0       0t0        661 /dev/ttyACM0
python     916 1614 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     916 1614 python           mks   18uW     CHR              166,0       0t0        661 /dev/ttyACM0
python     916 1642 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     916 1642 python           mks   18uW     CHR              166,0       0t0        661 /dev/ttyACM0
python     916 1643 python           mks   11uW     CHR               4,64       0t0        134 /dev/ttyS0
python     916 1643 python           mks   18uW     CHR              166,0       0t0        661 /dev/ttyACM0
Xorg       959                      root   12u      CHR                4,2       0t0         25 /dev/tty2
Xorg       959 1234 InputThre       root   12u      CHR                4,2       0t0         25 /dev/tty2
agetty    1029                      root  cwd       DIR              179,2      4096          2 /
agetty    1029                      root  rtd       DIR              179,2      4096          2 /
agetty    1029                      root  txt       REG              179,2    134480      35604 /usr/sbin/agetty
agetty    1029                      root  mem       REG              179,2   3407104       9498 /usr/lib/locale/locale-archive
agetty    1029                      root  mem       REG              179,2    353616       3899 /usr/lib/locale/C.utf8/LC_CTYPE
agetty    1029                      root  mem       REG              179,2   1651472      31068 /usr/lib/aarch64-linux-gnu/libc.so.6
agetty    1029                      root  mem       REG              179,2    202912      30789 /usr/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1
agetty    1029                      root  mem       REG              179,2     27028      59819 /usr/lib/aarch64-linux-gnu/gconv/gconv-modules.cache
agetty    1029                      root    0u      CHR                4,1       0t0         24 /dev/tty1
agetty    1029                      root    1u      CHR                4,1       0t0         24 /dev/tty1
agetty    1029                      root    2u      CHR                4,1       0t0         24 /dev/tty1
agetty    1029                      root    4r  a_inode               0,15         0       2074 inotify
[... cut ...]              root    8u      CHR                5,0       0t0         15 /dev/tty
sudo      2112                      root    8u      CHR                5,0       0t0         15 /dev/tty
```

and user mks is part of the "dialout" group:
```
mks@mkspi:~$ groups mks
mks : mks tty disk dialout sudo audio video plugdev games users systemd-journal input netdev
```

PS: I would appreciate if you could accept my DM in discord - maybe we solve it there better (tomorrow) and I will post the findings here

redrathnure, 2024-07-19:
Sorry, I was bit busy and lost the meaning of this thread:)
@Phil1988 m @frap129, questions:
1. Do we need the https://github.com/redrathnure/armbian-mkspi/pull/26 PR with "QIDI X-7 (Q1 Pro mainboard)," remark?
2. Do we need to add https://github.com/redrathnure/armbian-mkspi/issues/31#issuecomment-2233345983 info to readme?
3. Have you guys managed to make it working on "Qidi named X-6 (similar to X-4)"  board?

Phil1988, 2024-07-20:
Hey @redrathnure

1. Yes please include it. It will make the onboard serial (ttyS1) usable for the boards: X-4, X-6 and X-7
2. I thing we don't need it. It's a very rare case and my initial intention was to make the serial working. I was unsure of the kernel messages interfere with the serial and though that the onboard might be ttyS2. In short: leave it out.
3. Yes! It's working on X-4 and X-6 bopards with the dtbo of @frap129 . It would be great somehow the "mkspi-uart1" could be selected in the armbian-config hardware section (to add overlays=mkspi-uart1 to the /boot/armbianEnv.txt. This would make it work easier for ever X-4 to X-7 user.

frap129, 2024-07-22:
>     1. Do we need the [Add DTBO to allow enabling UART1 #26](https://github.com/redrathnure/armbian-mkspi/pull/26) PR with "QIDI X-7 (Q1 Pro mainboard)," remark?

I don't think the remark about the specific board is needed. UART1 is a valid port on all MKS PI based boards, per the schematic in [makerbase-mks/MKS-PI](https://github.com/makerbase-mks/MKS-PI/), not just the ones QIDI uses

>     2. Do we need to add [Disable kernel console debug messages for ttyS2 #31 (comment)](https://github.com/redrathnure/armbian-mkspi/issues/31#issuecomment-2233345983) info to readme?

I think thats a good idea, it would be useful for any needing ttyS2

>     3. Have you guys managed to make it working on "Qidi named X-6 (similar to X-4)"  board?
Yes!

Phil1988, 2024-07-23:
I think I misread the question of @redrathnure.
frap129 is right. Its not a specific remark. I just wanted to let you know that the DTBO PR is a good think and should be added in my opinion.

redrathnure, 2024-07-27:
Done. Please see https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#disable-debug-console-uart2--or-freeup-uart1-interface and https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.3.5-24.2.0-trunk for more details.

closed 2024-07-27 by redrathnure