redrathnure/armbian-mkspi #33: [Bug]: No HDMI signal with SKIPR and any 6.x kernels

attilabody, 2024-11-28:
### What happened?

I haven't been able to get a screen on HDMI with the latest (0.4.0-24.11.0-trunk) current (or edge), so I bisected. Results:
The most recent release which works is 0.4.0-24.11.0-trunk with 'current' (5.15.111) kernel. None of the releases show any image on HDMI using the 'edge' kernel (which was already 6.x-based in the first release)

### How to reproduce?

Download any release with 6.x kernel (I tested with jammy because that is the most supported, but it should not matter), connect only an HDMI monitor to an MKS SKIPR board and boot.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-11-28:
The screenshot was taken from `Armbian-unofficial_24.11.0-trunk_Mkspi_jammy_edge_6.11.6` image. And I pretty sure the 6.6.59 kernel build would give similar result.

So questions:
1. Have you hooked screen before boot of the board? Please note the images do not detects HDMI monitors which attached after system boot.
2. What is connected to the other end of HDMI cable? Is it regular monitor with standard screen resolutions (and fully supported EDI features), or is it a some "DIY" screen like MKS IPS50? Some devices are use non standard screen resolutions and this may give a problems. 
3. Have you tried to connect it to the PC/Laptop under "normal" Linux/Ubuntu OS with 6.x kernel?

[image]

attilabody, 2024-11-28:
Hi, first of all, thanks for caring!

1: they were hooked before powering on
2: I tried it with a Phillips 24" HDMI monitor and a 7" generic HDMI touchscreen (AliExpress style), both failed
3: The Monitor is usually connected to my desktop setup running on the latest released kernel (6.11.6-arch1 ATM)

Additional info: The screen(s) get activated when the board powers on, then once more during the boot process, but they display no image and go to power saving very soon.

I have another skipr unopened in its box I can try on that to minimize the chance of hardware malfunction, also several of the 7" touchscreen, but the fact that it works flawlessly with 5.x kernels gives quite a low probability.

Also, the same display works fine on my Sovol SV08 (BTT-CB1 based electronics)  with the image build from main branch of this repo (`BOARD=bigtreetech-cb1 BRANCH=current RELEASE=noble`)

I cannot decide what logs or other info might be relevant, please specify and I will post them.

Slightly related: The SPI-based touchscreen works fine on 6.x

redrathnure, 2024-11-28:
It's weird, let's try to realize what's going on here. 

1. (let's start from basics) could you share hardware version of your board(s). The label on the button side. Just to ensure we have the same boards:)
2. I assume that MKS-PI TS35 shows nothing when HDMI monitor connected, right?
3. "SPI-based touchscreen" - what do you mean? Is it about MKS-PI TS35 touchscreen?
4. Could you share output of following command from SKIPR + Armbian + connected HDMI monitor. Better for 6.x not working and 5.x working versions:
4.1. `fbset -i`
4.2. `ls /sys/class/drm/`
4.3. `dmesg | grep HDMI`
4.5. `cat /sys/class/drm/card0-HDMI-A-1/modes`
4.6. `cat /sys/class/drm/card1-HDMI-A-1/edid | parse-edid` or copy of `/sys/class/drm/card1-HDMI-A-1/edid` file.
5. Could you share output from any Linux machive/board when the same monitor works:
5.1. `cat /sys/class/drm/card0-HDMI-A-1/modes`
5.2. `cat /sys/class/drm/card1-HDMI-A-1/edid | parse-edid` or copy of `/sys/class/drm/card1-HDMI-A-1/edid` file.

Basically I would try to understand how system see this HDMI device.

BTW, I do believe running cutting edge image with 6.x kernel is **not** absolutely necessary (for 3D printer). My machine still works on pretty old image and I have never experienced any problem. And in worst case we may try to build an image with latest Armbian but 5.x kernel.

attilabody, 2024-11-28:
1: I have 2 boards, both 1.0
2: I did not connect them at the same time. Should I?
3: Yes

I tested with my second board and another 7" touch and the results seemed to be the same until I removed the power from the screen (powered on USB) then connected it again. The console appeared in a few seconds. Then I tested (after reboot and a blank screen) if disconnecting/reconnecting the HDMI cable does the trick, but unfortunately, it does not.

Requested info from 6.6.59-current-rockchip64, immediately after boot, screen is  blank
```
root@mkspi:~# fbset -i

mode "1024x600"
    geometry 1024 600 1024 600 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,0/0
endmode

Frame buffer device information:
    Name        : rockchipdrmfb
    Address     : 0
    Size        : 2457600
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 4096
    Accelerator : No

```

```
root@mkspi:~# ls /sys/class/drm/
card0  card0-HDMI-A-1  card1  renderD128  version
```

```
[    0.074295] platform ff370000.vop: Fixed dependency cycle(s) with /hdmi@ff3c0000
[    0.074441] platform ff3c0000.hdmi: Fixed dependency cycle(s) with /vop@ff370000
[    3.063268] dwhdmi-rockchip ff3c0000.hdmi: supply avdd-0v9 not found, using dummy regulator
[    3.063604] dwhdmi-rockchip ff3c0000.hdmi: supply avdd-1v8 not found, using dummy regulator
[    3.064048] dwhdmi-rockchip ff3c0000.hdmi: Detected HDMI TX controller v2.11a with HDCP (inno_dw_hdmi_phy2)
[    3.065769] dwhdmi-rockchip ff3c0000.hdmi: registered DesignWare HDMI I2C bus driver
[    3.066416] rockchip-drm display-subsystem: bound ff3c0000.hdmi (ops 0xffff8000811db2c0)
[   27.361699] platform hdmi-sound: deferred probe pending

```

```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/modes
1024x600
800x600
800x600
800x600
800x600
720x576
720x576
720x576
720x480
720x480
720x480
720x480
640x480
640x480
640x480
640x480
720x400
```

```
root@mkspi:~# cat /sys/class/drm/card1-HDMI-A-1/edid | parse-edid
cat: /sys/class/drm/card1-HDMI-A-1/edid: No such file or directory
Partial Read... Try again
```

but

```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/edid | parse-edid
Checksum Correct

Section "Monitor"
	Identifier "JRP7007"
	ModelName "JRP7007"
	VendorName "JRP"
	# Monitor Manufactured week 1 of 2021
	# EDID version 1.3
	# Digital Display
	DisplaySize 470 260
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-83
	VertRefresh 56-75
	# Maximum pixel clock is 210MHz

	#Extension block found. Parsing...
	Modeline 	"Mode 15" 27.00 720 732 796 864 576 581 586 625 -hsync -vsync 
	Modeline 	"Mode 0" 50.25 1024 1068 1156 1344 600 603 609 625 +hsync -vsync 
	Modeline 	"Mode 1" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 2" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 3" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 4" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 5" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 6" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 7" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 8" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 9" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 10" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 11" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 12" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 13" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 14" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 16" 74.25 1920 2008 2052 2200 540 542 547 562 +hsync +vsync interlace
	Modeline 	"Mode 17" 74.25 1920 2448 2492 2640 540 542 547 562 +hsync +vsync interlace
	Modeline 	"Mode 18" 74.25 1280 1720 1760 1980 720 725 730 750 +hsync +vsync 
	Option "PreferredMode" "Mode 15"
EndSection
```
... to be continued...

attilabody, 2024-11-28:
After power-cycling the touchscreen: (what isn't mentioned, didn't change)

```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/modes
800x600
800x600
848x480
640x480
```

```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/edid | parse-edid
Partial Read... Try again
```

```
root@mkspi:~# ls -l /sys/class/drm/card0-HDMI-A-1/edid 
-r--r--r-- 1 root root 0 Nov 28 22:43 /sys/class/drm/card0-HDMI-A-1/edid
```
... to be continued ...

redrathnure, 2024-11-28:
Sounds strange. Logs shows that the board recognizes monitor and able to read EDID info. 

May be issue in non standard "1024x600" resolution.... 

or this may point to luck or power (if display powered by USB). Could you try to connect it via USB3 port with all other USB devices disconnected (especially if you have WIFI dongle or WEB cam here)? Or try to power it from external source.

attilabody, 2024-11-28:
On Armbian_23.05.0-trunk_Mkspi_jammy_current_5.15.111:

```
root@mkspi:~# fbset -i

mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

Frame buffer device information:
    Name        : rockchipdrmfb
    Address     : 0
    Size        : 8294400
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 7680
    Accelerator : No
```

```
root@mkspi:~# ls /sys/class/drm/
card0  card0-HDMI-A-1  card1  renderD128  version
```

```
root@mkspi:~# dmesg | grep HDMI
[    3.797223] dwhdmi-rockchip ff3c0000.hdmi: Detected HDMI TX controller v2.11a with HDCP (inno_dw_hdmi_phy2)
[    3.798552] dwhdmi-rockchip ff3c0000.hdmi: registered DesignWare HDMI I2C bus driver
```

```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/modes
1920x1080
1920x1080i
1920x1080i
1920x1080
1920x1080
1920x1080i
1920x1080i
1280x720
1280x720
1280x720
1024x768
800x600
800x600
720x576
720x576
720x576
720x480
720x480
```

```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/edid | parse-edid
Checksum Correct

Section "Monitor"
	Identifier "JRP7007"
	ModelName "JRP7007"
	VendorName "JRP"
	# Monitor Manufactured week 1 of 2021
	# EDID version 1.3
	# Digital Display
	DisplaySize 470 260
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-83
	VertRefresh 56-75
	# Maximum pixel clock is 210MHz

	#Extension block found. Parsing...
	Modeline 	"Mode 15" 27.00 720 732 796 864 576 581 586 625 -hsync -vsync 
	Modeline 	"Mode 0" 50.25 1024 1068 1156 1344 600 603 609 625 +hsync -vsync 
	Modeline 	"Mode 1" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 2" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 3" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 4" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 5" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 6" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 7" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 8" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 9" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 10" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 11" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 12" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 13" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 14" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 16" 74.25 1920 2008 2052 2200 540 542 547 562 +hsync +vsync interlace
	Modeline 	"Mode 17" 74.25 1920 2448 2492 2640 540 542 547 562 +hsync +vsync interlace
	Modeline 	"Mode 18" 74.25 1280 1720 1760 1980 720 725 730 750 +hsync +vsync 
	Option "PreferredMode" "Mode 15"
EndSection
```

attilabody, 2024-11-28:
On my laptop:

```
[root@p52s ~]# cat /sys/class/drm/card1-HDMI-A-2/modes 
1024x600
1920x1080
1920x1080
1920x1080i
1920x1080i
1920x1080i
1920x1080
1920x1080
1920x1080i
1920x1080i
1280x1024
1280x720
1280x720
1280x720
1280x720
1024x768
1024x768
1024x768
832x624
800x600
800x600
800x600
800x600
720x576
720x576
720x576
720x576i
720x576i
720x480
720x480
720x480
720x480
720x480i
720x480i
720x480i
720x480i
640x480
640x480
640x480
640x480
720x400
```

```
[root@p52s ~]# cat /sys/class/drm/card1-HDMI-A-2/edid | parse-edid
Checksum Correct

Section "Monitor"
	Identifier "JRP7007"
	ModelName "JRP7007"
	VendorName "JRP"
	# Monitor Manufactured week 1 of 2021
	# EDID version 1.3
	# Digital Display
	DisplaySize 470 260
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-83
	VertRefresh 56-75
	# Maximum pixel clock is 210MHz

	#Extension block found. Parsing...
	Modeline 	"Mode 15" -hsync -vsync 
	Modeline 	"Mode 0" +hsync -vsync 
	Modeline 	"Mode 1" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 2" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 3" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 4" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 5" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 6" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 7" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 8" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 9" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 10" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 11" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 12" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 13" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 14" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 16" +hsync +vsync interlace
	Modeline 	"Mode 17" +hsync +vsync interlace
	Modeline 	"Mode 18" +hsync +vsync 
	Option "PreferredMode" "Mode 15"
EndSection
```

attilabody, 2024-12-01:
Might be a bit more relevant: Armbian on RPi4 with the same display

```
root@armbian-pi64:~# fbset -i

mode "1024x600"
    geometry 1024 600 1024 600 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/0
endmode

Frame buffer device information:
    Name        : vc4drmfb
    Address     : 0
    Size        : 1228800
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 2048
    Accelerator : No
```

```
root@armbian-pi64:~# ls /sys/class/drm/
card0  card1  card1-HDMI-A-1  card1-HDMI-A-2  card1-Writeback-1  renderD128  version
```

```
root@armbian-pi64:~# dmesg | grep HDMI.
[    4.488426] input: vc4-hdmi-0 HDMI Jack as /devices/platform/soc/fef00700.hdmi/sound/card1/input9
[    4.506583] input: vc4-hdmi-1 HDMI Jack as /devices/platform/soc/fef05700.hdmi/sound/card2/input11
```

```
1024x600
800x600
800x600
800x600
800x600
720x576
720x576
720x576
720x576i
720x576i
720x480
720x480
720x480
720x480
720x480i
720x480i
720x480i
720x480i
640x480
640x480
640x480
640x480
720x400
```

```
root@armbian-pi64:~# cat /sys/class/drm/card1-HDMI-A-2/edid | parse-edid
Checksum Correct

Section "Monitor"
	Identifier "JRP7007"
	ModelName "JRP7007"
	VendorName "JRP"
	# Monitor Manufactured week 1 of 2021
	# EDID version 1.3
	# Digital Display
	DisplaySize 470 260
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-83
	VertRefresh 56-75
	# Maximum pixel clock is 210MHz

	#Extension block found. Parsing...
	Modeline 	"Mode 15" -hsync -vsync 
	Modeline 	"Mode 0" +hsync -vsync 
	Modeline 	"Mode 1" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 2" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 3" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 4" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 5" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 6" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 7" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 8" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 9" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 10" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 11" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 12" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 13" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 14" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 16" +hsync +vsync interlace
	Modeline 	"Mode 17" +hsync +vsync interlace
	Modeline 	"Mode 18" +hsync +vsync 
	Option "PreferredMode" "Mode 15"
EndSection
```

redrathnure, 2024-12-01:
> Sounds strange. Logs shows that the board recognizes monitor and able to read EDID info.
> 
> May be issue in non standard "1024x600" resolution....

What it also strange, that Laptop and Armbian with 5.x kernel see 1920x1080 resolution and looks like uses it as default. I am not sure why 6.x kernel does not works in this case, and RPi4 does not use this resolution too.

Could you try to use  `fbset` command to force 1920x1080 resolution?

BTW, there are new release with the new kernels. And if it does not works, you always may use latest release with 5.x one

attilabody, 2024-12-04:
'''
root@mkspi:~# fbset -g 1920 1080 1920 1080 32
ioctl FBIOPUT_VSCREENINFO: Invalid argument
'''

redrathnure, 2024-12-06:
OK, a short update: I found a screen where I can reproduce it. I have a display module with 1024x600 resolution which does not works property on the latest images. Now I am trying to realize why and how to fix it...

redrathnure, 2024-12-11:
@attilabody could you try [0.4.4-25.2.0-trunk](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.4.4-25.2.0-trunk) release (one of current images)? There are some HDMI related adjustments.

attilabody, 2024-12-17:
> @attilabody could you try [0.4.4-25.2.0-trunk](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.4.4-25.2.0-trunk) release (one of current images)? There are some HDMI related adjustments.

Sorry, I did not notice. I will do it this evening.

redrathnure, 2024-12-17:
feel free to use latest 1.x images, they should have the same patches

attilabody, 2024-12-17:
Tested with the jammy image. There is picture, initially in 1920x1080 resolution (which this display "emulates", so the console is kind of readable, but image quality is not the best.
```
root@mkspi:~# fbset -i

mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,0/0
endmode

Frame buffer device information:
    Name        : rockchipdrmfb
    Address     : 0
    Size        : 8294400
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 1
    YPanStep    : 1
    YWrapStep   : 0
    LineLength  : 7680
    Accelerator : No
```
After power-cycling or reconnecting, the signal seems to switch to the the native (1024x600) resolution (as far as I can see by naked eye), but the console still being drawn onto a 1920x1080 "virtual" screen. Output of "fbset -i" does not change.

After boot:
```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/modes 
1920x1080
1920x1080
1920x1080i
1920x1080i
1920x1080i
1920x1080
1920x1080
1920x1080i
1920x1080i
1280x1024
1280x720
1280x720
1280x720
1280x720
1024x768
1024x768
1024x768
800x600
800x600
800x600
800x600
720x576
720x576
720x576
720x480
720x480
640x480
640x480
640x480
```
After reconnect:
```
root@mkspi:~# cat /sys/class/drm/card0-HDMI-A-1/modes 
1024x768
800x600
800x600
848x480
640x480
```

attilabody, 2024-12-21:
Isn't it possible that the root cause is quite simple, like a missing modeline from the device tree?

redrathnure, 2024-12-21:
Kind of... I ~~did~~ copied [a fix for MKS IPS50 screen](https://github.com/redrathnure/armbian-mkspi/blob/main/patch/kernel/archive/rockchip64-6.6/general-add-hdmi-mks-ips50-resolutions.patch), which is basically a declaration of new modeline. However I am not sure it's a right way to go, we cannot introduce custom modelines for endless list of noname lcd screens (IMO).

BTW, this is fragmen from kernel debug logs, which shows that EDID data was decoded but kernel module just rejected custom modes with "NOCLOCK" resolution:
```
[    3.201460] rockchip-drm display-subsystem: [drm:update_display_info.part.0] [CONNECTOR:43:HDMI-A-1] HDMI: DVI dual 0, max TMDS clock 0 kHz
[    3.201502] rockchip-drm display-subsystem: [drm:update_display_info.part.0] [CONNECTOR:43:HDMI-A-1] ELD monitor HDMI
[    3.201534] rockchip-drm display-subsystem: [drm:update_display_info.part.0] [CONNECTOR:43:HDMI-A-1] HDMI: latency present 0 0, video latency 0 0, audio latency 0 0
[    3.201570] rockchip-drm display-subsystem: [drm:update_display_info.part.0] [CONNECTOR:43:HDMI-A-1] ELD size 28, SAD count 1
[    3.203289] rockchip-drm display-subsystem: [drm] User-defined mode not supported: "1024x600": 60 50250 1024 1068 1156 1344 600 603 609 625 0x68 0x9
[    3.204541] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "1024x600": 60 50250 1024 1068 1156 1344 600 603 609 625 0x68 0x9 (NOCLOCK)
[    3.204596] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x400": 70 28320 720 738 846 900 400 412 414 449 0x40 0x6 (NOCLOCK)
[    3.204637] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "832x624": 75 57284 832 864 928 1152 624 625 628 667 0x40 0xa (NOCLOCK)
[    3.204678] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x480i": 60 13500 720 739 801 858 480 488 494 525 0x40 0x101a (BAD)
[    3.204719] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x480i": 60 13500 720 739 801 858 480 488 494 525 0x40 0x101a (BAD)
[    3.204760] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x576i": 50 13500 720 732 795 864 576 580 586 625 0x40 0x101a (BAD)
[    3.204799] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x576i": 50 13500 720 732 795 864 576 580 586 625 0x40 0x101a (BAD)
[    3.204841] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "640x480": 60 25200 640 656 752 800 480 490 492 525 0x40 0xa (NOCLOCK)
[    3.204879] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x480i": 60 13514 720 739 801 858 480 488 494 525 0x40 0x101a (BAD)
[    3.204919] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x480i": 60 13514 720 739 801 858 480 488 494 525 0x40 0x101a (BAD)
[    3.204958] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x480": 60 27027 720 736 798 858 480 489 495 525 0x40 0xa (NOCLOCK)
[    3.204998] rockchip-drm display-subsystem: [drm:drm_mode_prune_invalid] Rejected mode: "720x480": 60 27027 720 736 798 858 480 489 495 525 0x40 0xa (NOCLOCK)
[    3.205061] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:43:HDMI-A-1] probed modes:
[    3.205098] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x40 0x5
[    3.205140] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080": 60 148352 1920 2008 2052 2200 1080 1084 1089 1125 0x40 0x5
[    3.205181] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080i": 60 74250 1920 2008 2052 2200 1080 1084 1094 1125 0x40 0x15
[    3.205222] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080i": 60 74250 1920 2008 2052 2200 1080 1084 1094 1125 0x40 0x15
[    3.205263] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080i": 60 74176 1920 2008 2052 2200 1080 1084 1094 1125 0x40 0x15
[    3.205305] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080": 50 148500 1920 2448 2492 2640 1080 1084 1089 1125 0x40 0x5
[    3.205346] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080": 50 148500 1920 2448 2492 2640 1080 1084 1089 1125 0x40 0x5
[    3.205390] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080i": 50 74250 1920 2448 2492 2640 1080 1084 1094 1125 0x40 0x15
[    3.205431] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1920x1080i": 50 74250 1920 2448 2492 2640 1080 1084 1094 1125 0x40 0x15
[    3.205472] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1280x1024": 75 135000 1280 1296 1440 1688 1024 1025 1028 1066 0x40 0x5
[    3.205513] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1280x720": 60 74250 1280 1390 1430 1650 720 725 730 750 0x40 0x5
[    3.205554] rockchip-drm display-subsystem: [drm:drm_helper_probe_single_connector_modes] Probed mode: "1280x720": 60 74176 1280 1390 1430 1650 720 725 730 750 0x40 0x5
[... cut ...]
[    3.206465] rockchip-drm display-subsystem: [drm:drm_client_modeset_probe] [CONNECTOR:43:HDMI-A-1] Found mode 1920x1080
[    3.206492] rockchip-drm display-subsystem: [drm:drm_client_modeset_probe] picking CRTCs for 4096x4096 config
[    3.206527] rockchip-drm display-subsystem: [drm:drm_client_modeset_probe] [CRTC:38:crtc-0] desired mode 1920x1080 set (0,0)
[    3.206565] rockchip-drm display-subsystem: [drm:__drm_fb_helper_initial_config_and_unlock] test CRTC 0 primary plane
[    3.206609] rockchip-drm display-subsystem: [drm:drm_fbdev_dma_helper_fb_probe] surface width(1920), height(1080) and bpp(32)
```

yammaster, 2025-04-10:
Hi @all,
maybe I had a similar problem with my Elcrow 7" which works in low resolution mode fine but not in 1200x600. First I tried to modify  the extrargs with 1200x600 - screen doesn't work at all. So I tried to find out, wat's in the edid settings:

`cat /sys/class/drm/card1-HDMI-A-2/edid | parse-edid
`

and I couldnt find my 1200x600 resolution, only something like that;

```
Modeline 	"Mode 9" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
Modeline 	"Mode 10" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync 
```
So I took the 1280x720 for the following config line in armbianEnv.txt:

`extraargs=drm_kms_helper.edid_firmware=HDMI-A-1:edid/1280x720.bin video=HDMI-A-1:1280x720@60`

and it works for me. At last I edited the KlipperScreen.conf and I run in the higher resolution mode - but I don't know if it make sense or is it necessary for the KlipperScreen ;-)

redrathnure, 2025-04-10:
Interesting, I should check it with my screen....

@yammaster is image OK after this change? Especially image boundary and fonts...

yammaster, 2025-04-11:
yes, I think so

[image]