redrathnure/armbian-mkspi #34: [Bug]:   Unknown key name 'TransmitQueueLength' in canbus

FoxWind-coder, 2024-12-03:
### What happened?

when trying to raise the can bus 0, the systemd-networkd status displays

Dec 03 11:53:19 mkspi systemd-networkd[266]: /etc/systemd/network/80-can0.network:11: Unknown key name 'TransmitQueueLength' in section 'Link', ignoring.

version Armbian-unofficial_25.2.0-trunk_Mkspi_jammy_current_6.6.63

### How to reproduce?

fill in the config according to the template in readme.md

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FoxWind-coder, 2024-12-03:
[image]
in addition, there is a failed service

FoxWind-coder, 2024-12-03:
Yes, I tried to do it, but nothing changed. clipper sees the cards connected to the can, but when trying to do any action, it falls into timer too close. I tried increasing the timeout in mcu.py , as well as tracking packages, but nothing helped. I have now installed Armbian-unofficial_24.2.1_Mkspi_jammy_current_6.6.17, the printer has been printing for about an hour without errors. for debugging, I can provide another mks skipr and thr42

FoxWind-coder, 2024-12-03:
Sorry, I use a translator. There are mistakes:)

redrathnure, 2024-12-03:
We should find out how to specify TransmitQueueLength  for systemd-networkd case.

redrathnure, 2024-12-03:
Could you try 2 steps from the https://github.com/maz0r/klipper_canbus/blob/main/extras/systemd-networkd.md#creating-a-can-network-for-systemd-networkd page?

FoxWind-coder, 2024-12-03:
Yes, I'll try it, but tomorrow. I clarify that systemd creates a bus without a `TransmitQueueLength`, but it seems that there is not enough bandwidth. 
also, I want to clarify, which language is more convenient for you to communicate in? I don't know English very well.

redrathnure, 2024-12-03:
Yes, you are right. My setup needs at least ` txqueuelen 1024`. Otherwise I getting an error even during homing. 

I would say even auto translated english is better some other peoples may/will use these issue for troubleshooting :)

redrathnure, 2024-12-15:
I updated https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus section. It was tested on my printer with the new image

closed 2024-12-15 by redrathnure