redrathnure/armbian-mkspi #35: [Bug]: MCU Shutdown on 0.4.3-25.2.0-trunk

DK4YA, 2024-12-09:
### What happened?

[image]

### How to reproduce?

install 0.4.3-25.2.0-trunk

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DK4YA, 2024-12-09:
I Went back to Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.7.5. Working nice.

redrathnure, 2024-12-09:
@DK4YA CAN bus? If yes, how you configured it?

DK4YA, 2024-12-10:
Yes, CAN Bus was enabled like my config before. I´d copied my config on the new installtion.
But i need to create /interfaces.d/can0 by my self. I need to enable CAN BUS with "sudo ip link set can0 up type can bitrate 1000000". What config details you would like to see?

redrathnure, 2024-12-10:
Seems Ubuntu and Debian distros recently switched to NetworkManager so "/etc/interfaces.d/can0" does not work anymore. Now we should use systemd-network to create can interface. Please see [How to Configure CAN Bus section](https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus), [systemd-networkd instructions](https://github.com/maz0r/klipper_canbus/blob/main/extras/systemd-networkd.md#creating-a-can-network-for-systemd-networkd) and [[Bug]: Unknown key name 'TransmitQueueLength' in canbus ](https://github.com/redrathnure/armbian-mkspi/issues/34#issue-2714847108) for more details.

redrathnure, 2024-12-16:
I updated https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus part, please double check it and `sudo ip link set can0 ` **after reboot**.

DK4YA, 2024-12-16:
But one question I have, The bootup process is much more longer. On the display i can reading something about the systemmd and its waiting of something, that ok?

redrathnure, 2024-12-16:
Normally it should not wait too long. Usually it may be something network or disk related. You may try to check boot log.

DK4YA, 2024-12-17:
Sure you can check it.
[boot.log](https://github.com/user-attachments/files/18162046/boot.log)

DK4YA, 2024-12-17:
> Normally it should not wait too long. Usually it may be something network or disk related. You may try to check boot log.

Hey Maxim,

I maked a photo from this waitingtime.i need to wait 2 minutes 
[image]

redrathnure, 2024-12-17:
It's defensively about network connection and looks like something very local specific. I do not see this error neither for my test setup nor for for a 3D printer. However please note I use and test only Ubuntu images, maybe you have debian specific problem.

DK4YA, 2024-12-17:
Okay, I actually just flashed the distro onto the EMMC and installed all the necessary instances with Kiauh. Configured the CAN bus according to your instructions. Activated the network with ambian-config. I didn't do anything else. This is before Klipper starts. But it's not a big deal, it's waiting for something. But it seems to work well. However, the CPU load is a bit higher than with the old distro. My PWM fan also runs faster without load, which means the temperature is slightly higher. Nevertheless, it's up-to-date software for a rather poorly designed board. And that makes it really great! BTW. The BTT PI2 also has a Rockchip and very old distros. Is there any way to do something with that?

redrathnure, 2024-12-17:
About MKSPI/SKIPR boards. I have noted slightly higher CPU load too.

1. In short, MKS decided to overlook CPU for their images (set higher power voltage, higher CPU frequency and more aggressive power profile). I do not see huge benefit out of it except heating your room or printer enclose :) In my latest images I use "recommended" values for Renegate board (I believe it's kinda parent/reference design for MKSPI and related boards). There are still more then enough power for all tasks (including non Klipper related, like e.g. backups via restic/resticprofile) and I see lower crystal temp which not is is 65-67°C under heavy load and without any cooling. In any case, feel free to diff a new and old DTS files and compare operating-points and rk805/regulators sections.

2. I would not recommend to use edge images for real work load. Kernel v6.6 is LTS and it's stable enough, what I cannot say about latest v6.12 line (each build brings new surprises. Plus, maybe it;s time to try Ubuntu :) At the end 2 minutes delay per each start it's kinda annoying issue and IMO it's worth to spend a time to find proper solution.

3. `Nevertheless, it's up-to-date software for a rather poorly designed board. ` . `up-to-date software` yes, defensibly, especially non-EOL Debian :) However I would not be so skeptical about hardware. They copied pretty good design and I still do not see better alternative for this prices (Display + eMMC combination). 

About BTT PI2: I do not have this board and no plans to buy it. A building images without hardware (and free time for troubleshooting) is not the best idea.  However you may check main [Armbian repo](https://github.com/armbian/build/tree/main/config/boards) and their forum e.g. [this topic](https://forum.armbian.com/topic/38873-bigtreetech-cb2-and-pi-2-support/#comment-190238)). Perhaps somebody already doing the adaptions.

redrathnure, 2024-12-18:
> Now it doesn't work anymore.
Could you try to forget the WIFI network and connect it again? And I am bit confused, I would expect `systemd-networkd` manages only can0 interface and it must not touch eth/wlan and other real connections. Ethernel and WLAN should be managed via NetworkManager

redrathnure, 2024-12-18:
Just as an experiment, could you try microsd with ubuntu image? Is it general problem or distributive specific...

DK4YA, 2024-12-18:
> > Now it doesn't work anymore.
> 
> Could you try to forget the WIFI network and connect it again? And I am bit confused, I would expect `systemd-networkd` manages only can0 interface and it must not touch eth/wlan and other real connections. Ethernel and WLAN should be managed via NetworkManager

I really don't know who is managing what. I just tried deleting the WiFi in armbian-config and then re-establishing the connection.

DK4YA, 2024-12-18:
> Just as an experiment, could you try microsd with ubuntu image? Is it general problem or distributive specific...

Tell me more precisely what you want, I'm sure I can try something out on my experimental board. But why SD card? Yes, but it's definitely possible.

redrathnure, 2024-12-18:
SD/emmc it isn't  meter. I am curious what cause the issue.
Questions:
1. Can you reproduce the same issue in Ubuntu?
2. Can you step by step disable all network connections and reboot after each step? E.g. remove can0 and reboot. Note is there are any delay or not. Then forget wifi connection, reboot and so on. Perhaps this would gibe an idea which interface/network connection lead to this waiting.

DK4YA, 2024-12-18:
We can test that,
but my command knowledge is not sufficient. If you can help me, I can flash it onto an SD card. But you have to tell me which Ubuntu you mean. Maybe a link or something.
Maybe we should do that outside of this thread, since it's probably no longer relevant.

redrathnure, 2024-12-18:
For example [1.0.0 noble current](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F1.0.0-25.2.0-trunk/Armbian-unofficial_25.02.0-trunk_Mkspi_noble_current_6.6.65.img.xz). Debian and Ubunut have similar structure and packet manager. So usually you connects to wifi/lan, then install klipper and related soft via kiauh, then configure can0 and finish printer configuration (from web interface).
Something like 
```
ip addr show
# Connect to wlan
sudo armbina-config

# kiauh
cd ~
git clone https://github.com/dw-0/kiauh.git
./kiauh/kiauh.sh

#CAN Bus
    cat <<-'EOF' | sudo tee /etc/systemd/network/10-can.link
[Match]
Type=can

[Link]
TransmitQueueLength=1024
# see also https://www.freedesktop.org/software/systemd/man/latest/systemd.network.html#%5BLink%5D%20Section%20Options
EOF

   cat <<-'EOF' | sudo tee /etc/systemd/network/25-can0.network
[Match]
Name=can0

[CAN]
BitRate=1M
RestartSec=200ms
# see also https://www.freedesktop.org/software/systemd/man/latest/systemd.network.html#[CAN]%20Section%20Options
EOF 
sudo systemctl enable systemd-networkd
sudo systemctl start systemd-networkd
sudo systemctl status systemd-networkd

sudo reboot 
```

DK4YA, 2024-12-18:
> For example [1.0.0 noble current](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F1.0.0-25.2.0-trunk/Armbian-unofficial_25.02.0-trunk_Mkspi_noble_current_6.6.65.img.xz). [...]

Okay, i prepair a SD with this Ubuntu. This takes a while.

DK4YA, 2024-12-18:
> SD/emmc it isn't meter. I am curious what cause the issue. Questions:
> 
> 1. Can you reproduce the same issue in Ubuntu?
Yes, i´ve got waiting time with onlay Wlan on.
[image]

> 2. Can you step by step disable all network connections and reboot after each step? E.g. remove can0 and reboot. Note is there are any delay or not. Then forget wifi connection, reboot and so on. Perhaps this would gibe an idea which interface/network 
connection lead to this waiting.

The forgot funktion for Wlan is changed to reset all, i think. 
[image]

This is actually the connectivity of this SKIPR. Now I restart

No, i´ve got the waiting time. Am I doing something wrong?

redrathnure, 2024-12-18:
> in this bootsection it will stuck it too around more then a minute.

This should not be a case. It may stuck on bootstage if you gave a problem with microsd or emmc.

And after all I still have no idea why you have this delay... it's weird

redrathnure, 2024-12-18:
Damn, now I have the same on my printer &#128533; At least I may reproduce this issue locally....

redrathnure, 2024-12-18:
Okay, after several failed experiments and [reading this endless discussion](https://bugs.launchpad.net/netplan/+bug/2060311) I gave up. I just disabled this "nice" service (`sudo systemctl disable systemd-networkd-wait-online`) and got rid of the 2 minute delay. 
Please keep in mind it's fast and dirty workaround, I have no idea what this service was introduced for.

DK4YA, 2024-12-19:
> Okay, after several failed experiments and [reading this endless discussion](https://bugs.launchpad.net/netplan/+bug/2060311) I gave up. I just disabled this "nice" service (`sudo systemctl disable [...]

But one thing is a bit strange now. I can no longer access the Bookworm distro via WiFi. Even unplugging and re-plugging the dongle does not lead to a connection. A LAN connection works immediately. I then switched the WiFi connection back on using armbian-config and restarted after a test. WiFi is off again. I then deactivated the service "systemctl disable systemd-networkd-wait-online". Reconnected WiFi using armbian-config. Now the WiFi connection works again when starting up and the waiting time is gone. The printer is ready to use in a few seconds. And do I have to ask the question, why is systemd-networkd-wait-online needed?

redrathnure, 2024-12-19:
I have feeling this mess happen because systemd-networkd and NetworkManage try to handle the same wifi connection.

If I got it right, all Armbian distros use NetworkManager by default. When you install KlipperScreen it also asks about NetworkManager. Actually you would see wifi section in the KlipperScreen only if this interface is handled by NetworkManager.

But NetworkManager cannot handle CAN interface. This is why we enabled systemd-networkd. Main idea is the systemd-networkd is for CAN only, the rest of interfaces must be handled by NetworkManager.

DK4YA, 2024-12-19:
Hello Maxim,

I assume that the idea is to use both managers. Who managed /interfaces.d/can0 before? I thought the network manager would regret it.

I actually finished a print and got this one:

Hello Maxim,

I assume that the idea is to use both managers. Who managed /interfaces.d/can0 before? I thought the network manager would regret it.

redrathnure, 2024-12-19:
[Looks like it's for ifupdown package](https://manpages.ubuntu.com/manpages/xenial/man5/interfaces.5.html)

DK4YA, 2024-12-25:
> [Looks like it's for ifupdown package](https://manpages.ubuntu.com/manpages/xenial/man5/interfaces.5.html)

Hohoho or rather ha ha ha... Merry Christmas Maxim.
I had some problems with two ebb36 because of a resistance problem for the Max chip. I spent my last ebb36 but I designed a new bracket for ebb36. Okay, I started the printer again now and found funny news. Like before I could only connect via eth0, wifi doesn't start on boot. Reassigning wlan0 at armbian-config is fine so far and seems to work too, but wifi connection only works when eth0 is connected. Now you. You have such crap, not with a crappy Win10 machine. Maaaan.

redrathnure, 2024-12-25:
Merry Christmas, 

I'm not sure I understood the last message, sorry.
just trying to guess and comment on it.

> I had some problems with two ebb36 because of a resistance problem for the Max chip. I spent my last ebb36 but I designed a new bracket for ebb36. 

I have newer understand real benefits out of PT100. Does it really have sense to spend extra $$ and have additional related "fun"?

> I spent my last ebb36 but I designed a new bracket for ebb36.

have not got this part at all :) 

> Like before I could only connect via eth0, wifi doesn't start on boot. Reassigning wlan0 at armbian-config is fine so far and seems to work too, but wifi connection only works when eth0 is connected.

Something defensively wrong with network configuration. Ethernet and WLAN must not interfere each other. CAN too. 
From my experience:
1. Test stand (MKSPI  + WIFI, mostly no ethernet connected, Ubuntu Noble current and edge). A few days total uptime (testing releases, multiple experiment etc), no any issues.
2. Main printer (SKIPR + WIFI only, no ethernet + EBB36 via CAN, Ubuntu Noble current). 12+ hours printing and a few days uptime. No any problems. My setup sequence was: install image, connect wot WLAN via armbian-config, install Klipper/KlipperScreen, configure CAN using snippet from readme. I see normal boot time, stable connection and pretty good performance of the both CAN and WLAN networks.

> Now you. You have such crap, not with a crappy Win10 machine. Maaaan.

perhaps my English is not so good, but I am totally lost here :)

> ^^^^> ithat's something I don't understand.
> I've never understood how to properly configure a network. This text in particular doesn't make sense.

Yeah, I share your pain :) Sometimes all this complexity give just frustration. So how I understand it:
1. For many many years Linux used if-up/if-down scripts to manage network settings. Especially Debian based distros, especially server releases. It's kinda scripts which are executed when einterface connection and showdown. E.g. you may look at "old" "/etc/network/interfaces.d/can0" based instructions. Usually you would see `ifconfig`, `if-up`, `if-down` and `ip` command in different combinations. 
2. There are NetworkManager tool which are doing similar work but use totally different configuration files and way of working. If I got it right, it's more useful for environments with X Window (e.g. Ubuntu Desktop and similar). Usually you would see `nmcli` command or UI based configuration here.
3. A few years ago people introduced Netplan, I am not sure why and what for, but looks like it's just wrapper over different systems to configure network (ifupifdown, NetworkManager etc).  Probably you would see `netpaln applay` or similar commands in the guides. 
4. Initially netplan use if-up/if-down scripts to do it's job. This is why we was able to use old-good "/etc/network/interfaces.d/can0" based instructions (almost) without any changes.
5. A few month ago Armbian drop support of if-up/if-down scripts and hard switch to NetworkManager (probably because Debian did this step first). This is the exactly time when "/etc/network/interfaces.d/can0" based instructions stop working. and...
6. ... we found that NetworkManager is unable to manage CAN interfaces. So we need to use  systemd-networkd (because it can manage CAN bus :/
7. systemd-networkd has the similar idea as  if-up/if-down scripts, but it's do it in more declarative way. Instead editing `/etc/network/interfaces.d/*` files now we have to edit lot of weird `*.service` and `*.link` configuration files. IMO it less intuitive and more messy approach, but we have what we have.... 
8. and yes, another one painful point:  NetworkManager  and systemd-networkd should not works together. They have totally different ideas, implementation and way how they integrated into system. So in theory:
  * we should never use them both together
  * however if we need, we may do this but having lot of side effects, e.g. surprises such your system brings
 
So, current status:
1. Ethernet and WLAN connections must be configured via NetworkManager. Means using `nmcli` command OR via armbian-config OR via KlipperScreen. 
2. CAN should be configured via systemd-networkd (combination of `*.service` and `*.link` files).
3. systemd-networkd MUST NOT take care about ethernet or WLAN interfaces. systemd-networkd MUST NOT be configured to test internet/network connection in any way. And yeah, looks like this works a bit different from distro to distro and probably even from release to release.
4. All this ~~fun~~ hell is distributive specific logic. I am 99% sure that neither Kernel nor DTS descriptors do not affect network configuration. And I am pretty sure that this is not even related to Armbian (it's manages only kernel and few system tools, the rest is taken from Ubunutu/Debian official distributive.

And probably I would address all these network/CAN/WiFI/boot delay issues to the related forum. It is not Kernel or Armbian related issues.

fbeltrao-br, 2025-01-04:
[redrathnure](https://github.com/redrathnure) builds have been working great for the past months with both Skipr and EBB36 in CAN mode, until yesterday when I started to have MCU timeouts when homing.

After checking the GitHub page and this thread, nothing in software would seem out of place, specially because I have been running the same setup for the past 7 months without changes in both software and hardware, until I decided to check the connections:

[image]

Notice the discoloring in the two left MX3.0 terminals, they are 18 AWG silicon insulated wires for the 24V and GND lines, the other two on the left are 26 AWG for the CAN H and L (the insulation are mangled because it's not very easy to remove the terminals without destroying the housing).

I guess the EBB36 didn't liked the 80W heater cartridge I was using for the past months, I'll limit the power to something around 65% to make it approximately 50W. There was also some discoloring in the EBB36 connector.

I'm too lazy to crimp the terminals again so until now I don't know if something fried or if it was just a bad connection.

If you are having connection issues, make sure to check the connections closely.

redrathnure, 2025-01-04:
@fbeltrao-br  yeah, the bug report became a thread about canbus pain:) anyway… 

I've experienced similar problems. First of all we all love to use cheap random solutions instead of industrial/automotive certified cables and connectors. Just because “they” want to squeeze money out of pure DIYers and a few proper cables may cost a half of entry level 3D printer, right? However we should keep in mind a few points:
1. 3D printer means vibration, vibration means movement, including movement in most unexpected places. A few days ago I had a problem with a self -pulled AC C13 connector (and it was a bit unexpected finding)
2. What we usually use in self build and low cost printers is not really designed to work in CnC machines. Because load, heat, vibration and bending (sometimes with extreme radius). For example ~~cheap~~ budget wires with pure contacts/crimping areas in 60-120W power lines usually fail with melted connectors (sometimes with sparks and other special effects). BTW this is not only about CANbus lines, I've had fun even with “usual” 24W and 40W heater elements.
3. ...and widely used “genius” ideas like "let's use Micro Molex for 100W power delivery" or "Micro/TypeC USB to power RPi" is kind of a bad joke.

So, about CANbus, from my experience I would say:

1. If network link randomly fails (without significant software/OS/kernel updates) OR if the print fails in some specific bed area then in 90% cases it;s hardware issues. Mostly contacts related. Monitoring the amount of tr/rx error packages may give you early warning. But it's better to think about proper line and connectors before print fails.
2. Cables must be good quality, especial attention for clamping areas and connectors. It would be a good idea to select power delivery wires with a cross-section 1.5-2 times larger than that required. And yes, it’s pretty hard to do for 2 pins of Molex Micro-Fit.
3. All connectors must be physically fixed. Spring or screw connection is better, or at least with thermo glue. Extreme solutions like “just solder it” or "zip tail it to nearest case element" work well too :)  And all cables must be physically secured. Connectors/sockets must not have any bending or pulling forces in all head positions. Molex /XH2.54/SM2.5/USB and other budget solutions are not designed to be dynamically loaded.
4. Even budget crimping tools with bit patient and practice give much more better results that random crap from Aliexpress (even if was sent as part of Kit or ready machine).

DK4YA, 2025-01-06:
Hello Fbeltrao,

first of all I want to thank you for your flash manual. I shared it in the Klipper discourse and it worked well.
Only the second MCU and how to install it was missing for me.

> I guess the EBB36 didn't liked the 80W heater cartridge I was using for the past months, I'll limit the power to something around 65% to make it approximately 50W. There was also some discoloring in the EBB36 connector.

I´ve got a 70W Heatcartridge. When it heats up, the fan slows down significantly. Which is obviously not the point. That's why I would like to reduce the heating power. Can you please tell me how you did that?

best regards Sebastian

fbeltrao-br, 2025-01-06:
> I´ve got a 70W Heatcartridge. When it heats up, the fan slows down significantly. Which is obviously not the point. That's why I would like to reduce the heating power. Can you please tell me how you did that?

max_power parameter https://www.klipper3d.org/Config_Reference.html#extruder

redrathnure, 2025-01-06:
Probably a better solution would be considering new cables for power line. There is FLRY cable type which pretty inexpensive and widely used in auto repair shops (should be easy to find online). 0.5-0.7 mm^2 should be enough for 70-80W heater. 
[image]

DK4YA, 2025-02-06:
Hello Maxim,

Do you have any idea why the screenshot shows a LAN connection even though I only have a WiFi connection active?

[image]

redrathnure, 2025-02-06:
You have real (internal) IP address assigned for the ethernet interface, means cable is connected and your DHCP server/router gives a lease. Perhaps you have a multiple network managers (e.g. systemd-networkd and NetworkManager) and lan was disabled in one of them.

redrathnure, 2025-02-06:
OK, it is not controlled by NetworkManager. Then, `ip addr show`? Do you see your interface here? Do you see any related records in `/etc/netplan/*` files (e.g. ` sudo less /etc/netplan/*`)?