redrathnure/armbian-mkspi #36: MSK Pi - Cannot make Screen Working

doomgrave, 2024-12-11:
### What happened?

I have an X4 Pro that mount a custom MSK Pi. Everything works with your rom but i cannot manage to make the touchscreen display to work.

Could there be compatibily issue or i may fix it?
I hope you can provide me some knowledge!

### How to reproduce?

The Display it's connected here, seems to be HUART1
[image]

I leave here some messages i get:
`sudo dmesg | grep -i drm
`
```
[    3.098496] rockchip-drm display-subsystem: bound ff370000.vop (ops 0xffff800                      0811d7080)
[    3.101735] rockchip-drm display-subsystem: bound ff3c0000.hdmi (ops 0xffff80                      00811db2c0)
[    3.103035] [drm] Initialized rockchip 1.0.0 20140818 for display-subsystem o                      n minor 0
[    3.103177] rockchip-drm display-subsystem: [drm] Cannot find any crtc or siz                      es
[    3.107894] rockchip-drm display-subsystem: [drm] Cannot find any crtc or siz                      es
[    4.144373] [drm] Initialized lima 1.1.0 20191231 for ff300000.gpu on minor 1
[    6.659171] systemd[1]: Starting modprobe@drm.service - Load Kernel Module dr                      m...
[    6.780325] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    6.781726] systemd[1]: Finished modprobe@drm.service - Load Kernel Module dr                      m.
```
`
sudo dmesg | grep ttyS*
`
```
mks@mkspi:~$ sudo dmesg | grep ttyS*
[    0.000000] Kernel command line: root=UUID=3ce64f4b-facd-405e-9132-715a138f3e1c rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=fcf06030-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory
[    0.001064] printk: console [tty1] enabled
[    0.866727] ff110000.serial: ttyS0 at MMIO 0xff110000 (irq = 21, base_baud = 1500000) is a 16550A
[    0.869171] ff120000.serial: ttyS1 at MMIO 0xff120000 (irq = 22, base_baud = 1500000) is a 16550A
[    0.871632] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 23, base_baud = 1500000) is a 16550A
[    0.871817] printk: console [ttyS2] enabled
[    6.483281] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[    6.490385] systemd[1]: Created slice system-serial\x2dgetty.slice - Slice /system/serial-getty.
[    6.499989] systemd[1]: Expecting device dev-ttyS2.device - /dev/ttyS2...
[    7.980669] systemd[1]: Found device dev-ttyS2.device - /dev/ttyS2.
[    8.946402] cdc_acm 1-1:1.0: ttyACM0: USB ACM device

```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-12-11:
@doomgrave You need to adjust DTS (Device Tree) file for your purposes. For MKSPI it's `/boot/dtb/rockchip/rk3328-roc-cc.dtb`. Please see [an example](https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-rotate-screen) how to unpack and edit it.

I would say you have two options:
1. getting `dtb` file from already worked image.
2. developing a new one based on mkspi version (need to know schematic of your board and may be used already worked dts as reference)

redrathnure, 2024-12-11:
Yes, you can try to swap them. BTW these files are located on boot partition, the FAT formatted one. So you may easy extract it from EMMC (OEM image) and put it to new SD/EMMC partition even before first boot.

doomgrave, 2024-12-12:
> Yes, you can try to swap them. BTW these files are located on boot partition, the FAT formatted one. So you may easy extract it from EMMC (OEM image) and put it to new SD/EMMC partition even before first boot.

First of all ty for your support !
Ive spent a day, reflashed the original firmware and i got the same exact dsmg for the attached display.

So i guess the printer its simply just making klipper communicate with the display in some non standard way.
Before closing the post may you know how i can check where this configuration happens?
In some ways they communicate with the screen as it is an external entity.

doomgrave, 2024-12-14:
> I have no experience with this machine and the board, so I am not sure what you actually asking about :)
> 
> A few ideas:
> 
> 1. If you talking about [KlipperScreen ](https://klipperscreen.readthedocs.io/en/latest/#required-hardware) (see something like [this screen](https://klipperscreen.readthedocs.io/en/latest/img/panels/main_panel.png)), then most likely you have Linux Kernel (dts/dtb files) -> X11 (/etc/X11) -> KlipperScreen (somewere in your home dir, please check documentation) configuration chain.
> 2. If you see kind of classical menu like interface, then it would be mcu -> Klipper configuration chain (see you printer.cfg). Also it possible that LCD connected to Klipper host/MKSPI instead to MCU part. In this case you may found [Klipper MCU service ](https://klipperscreen.readthedocs.io/en/latest/img/panels/main_panel.png)installed and related section in out printer.cfg
> 3. It also possible that you screen is separated "computing module" with own firmware and which use serial connection to just get printer status and  send a commands back (like it was done for a few LCD modules + Marlin). However I have no ide how this kind of configuration works with Klipper.

Im digging through the original image downloaded from:
here:https://drive.google.com/file/d/1LQrxblUMy1nzgDPhWfqrJFE4n6hnQyiI/view
In printer.cfg there is nothing about klipperscreen 

There is klipperscreen service though and this X11. So i guess its the forst option.
Could you point me some guides to do an installation like this?
Copying the specific dtb/x1 configuration klipperscreen could work?
There is the service running but there its no klipperscreen_folder in ~\klipperscreen

redrathnure, 2024-12-14:
You may have a look at https://github.com/Phil1988/FreeQIDI. But in short:
Step 1. Install [kiauh](https://github.com/dw-0/kiauh), e.g. 
``` 
cd ~
git clone https://github.com/th33xitus/kiauh.git
./kiauh/kiauh.sh
``` 
Step 2. Install -> Klipper, Moonraker, Mainsail and KlipperScreen
Step 3. Install klipper_mcu service, if you need access to host temperature or any devices connected to RPi/MKSPI. e.g. 
```
d ~/klipper/

make menuconfig
# In the menu, set "Microcontroller Architecture" to "Linux process," then save and exit.

sudo ln -s $PWD/scripts/klipper-mcu.service /etc/systemd/system/
sudo systemctl daemon-reload
sudo systemctl enable klipper-mcu.service

sudo systemctl stop klipper
make flash

sudo systemctl start klipper-mcu klipper
sudo systemctl status klipper-mcu klipper
```

However I guess that display should be configured before installing KlipperScreen. kiauh -> KlipperScreen step just install required packages (like X11) and use already existed screen. For MKSPI ti's frame buffer device (/dev/fb0).

redrathnure, 2024-12-14:
Could you share your dts file (unpacked version)?

redrathnure, 2024-12-14:
I looked to the https://drive.google.com/file/d/1LQrxblUMy1nzgDPhWfqrJFE4n6hnQyiI/view image. IMO they do not have display in usual sense. Looks like they have some client app which communicates to your screen via serial bus. And I do not see  'KlipperScreen' instance, to the vendor uses some closed source software instead.

doomgrave, 2024-12-15:
> I looked to the https://drive.google.com/file/d/1LQrxblUMy1nzgDPhWfqrJFE4n6hnQyiI/view image. IMO they do not have display in usual sense. Looks like they have some client app which communicates to [...]

Yes indeed, in systemd ive found Klipperscreen service. But no configuration its present.
Thus the screen should be connected to the Uart1.

When i connect the display to the system its reported as  "/device/virtual/vc/vcsa7""
Ive inspected the device tree and could not find any added dsb.

Im losing my head even if im not good with linux im learning a lot.
But i want to discover how to emulate what they have done copying the right services

redrathnure, 2024-12-15:
> Ive inspected the device tree and could not find any added dsb.

a diff between your and MKSPI oem DTS shows:
- `pinctrl` -> `spi0_cs1`, `sdio-pwrseq`, `wireless-wlan` and `spi0_cs0 declaration`
- `wireless-wlan` and `sdio_pwrseq0` declarations

> Im losing my head even if im not good with linux im learning a lot.
> But i want to discover how to emulate what they have done copying the right services

A few hints (without knowing your machine, just looking inside provided image):
1. This hardware and software is defenettly form  Makerbase. So pay attention to everything with prefix `makerbase` or `mks`. And have a look at files inside home director**ies** (`/home/mks` and `/root`)
2. I would recommend to inspect all registered and active services. E.g. have a look to `/etc/systemd/system` dir, especially `/etc/systemd/system/multi-user.target.wants` one.
3. There are some scripts, which are used by services: `/home/mks/Desktop/myfile/ws/start.sh` and `/root/mcu-id.sh`. I have feeling that the first one is a key for understanding how your screen communicates with main board.
4. I would recommend to reach printer seller or/and manufacturer. Perhaps they would help you with technical details. Chances are not high, but not zero :)

Please remember, that MKSPI part (Armbian with Klipper) is only first part. You would need to realize MCU part too. At least if you are going to build/update Klipper MCU firmware. This part would be extremely challenged without knowing schematic.

And a to be honest, a board looks extremely ~~cheap~~ cost optimized. No Ethernet, no HDMI, just a few USB ports. So, if you looking for upgrades in future (CANbus, multiple Z axis, additional sensors etc) the board may be limiting factor. From a time investment perspective, a purchasing a board(s) with better support may be a more cost effective solution:)

doomgrave, 2024-12-15:
> a diff between your and MKSPI oem DTS shows:
>
> * `pinctrl` -> `spi0_cs1`, `sdio-pwrseq`, `wireless-wlan` and `spi0_cs0 declaration`
> * `wireless-wlan` and `sdio_pwrseq0` declarations
>
>
> A few hints [...]

Absolutly, ill dig into that folder and try to discover!
This its my second 3d printer and my forst with klipper.
Anyway thanks to your support and your rom i managed to make iwork 99% of the board with latest Klipper version!
The only thing i miss its the display, i can live without,  or in the worst case ill take an actual mskpi that would be a really easy switch

Thank you for your time, ill let you know and if i reach to do this ill make a guide for the community and point your help!

karabas2011, 2024-12-17:
Kingroon KLP1 with cheetah v2.2 
I installed new image Armbian-unofficial_25.02.0-trunk_Mkspi_bookworm_current_6.6.65.img
installed new klipper moonraker  Klipperscreen
All works but display does not react on touch. 
I replaced  rk3328-mkspi.dtb with  original rk3328-roc-cc.dtb. It shows smaller image but touch does  not work still.
What I need to do?

redrathnure, 2024-12-17:
@karabas2011 First of all I would recommend opennig a new issue. This one is around  X4 Pro, I am not sure you have similar boards.

Then you may try to realize type of your touchscreen and related kernel module. E.g. booting to working (oem?) image and exploring dmesg and lsmod output.

And some kingroon  machine was already mentioned in a [`AP6212 WIFI not working`](https://github.com/redrathnure/armbian-mkspi/issues/28) issue. Perhaps you may reach the author and ask their experience.

karabas2011, 2024-12-18:
Ок. Did it

doomgrave, 2024-12-18:
@redrathnure  an update, maybe usefull also to others exploring this kind of communication:
The screen its connected to the printer voa VT7 (virtual terminal 7)
It is not recognized by xrandr, to be able to do it i will need to set up properly Dbus+LightDm as a window manager.
Im trying to figure out on the original rom how this was setup. Even there they were using a custom gui and not klipperscreen.
After i make the screen avaiable to Xrandr should ne easy to make klipperscreen work!