redrathnure/armbian-mkspi #37: [Bug]:MKS IPS50 display does not react to touch

karabas2011, 2024-12-18:
### What happened?

I have some kingroon KLP1  printers with motherboard Cheetah-V2.2  ( rp3328 + MCU GD32F303 + THR rp2050) with
MKS PI-TS35 display.  and AP6212 wifi module.
It it modified MKSPI software inside ( old debian busty, python 3.7 without ability to upgrade)
I flashed image Armbian-unofficial_25.02.0-trunk_Mkspi_bookworm_current_6.6.65.img.
I installed last klipper+moonraker+KlipperScreen.
Then I flashed both MSUs.
Display shows good image. 

But no touchscreen reactions. If I  connect mouse I be able to press buttons but blindly, no cursor on screen.

Then I replaced rp3328-mkspi.dtb with renamed stock rp3328-roc-cc.dtb
I got smaller image on screen and very seldom and random actions on touch.

BTW if i connect 7" hdmi display - black screen. If i connect big PC hdmi display - it shows image.

### How to reproduce?

Seen above.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-12-18:
Some kingroon machine was already mentioned in a https://github.com/redrathnure/armbian-mkspi/issues/28 issue. Perhaps you may reach the author and ask their experience.

You may try the file from https://github.com/redrathnure/armbian-mkspi/issues/28#issuecomment-2526495845 (please do not foreget to rename it to rk3328-mkspi.dtb if you use fresh images)

redrathnure, 2024-12-18:
@karabas2011 reading all you comment and open issues, I would assume you took a stock rk3328-roc-cc.dtb, and put it to the [one of latest custom Armbian image](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F1.0.0-25.2.0-trunk). After this you got working wifi, but half working screen (scaled down image). Assuming KlipperScreen was already installed.

If it's correct, let's try the next step to fix your screen:
```
# assuming ~/rk3328-roc-cc.dtb is a stock dtb file
# Unpack it to ~/rk3328-roc-cc_stock.dts
dtc -I dtb -O dts -o ~/rk3328-roc-cc_stock.dts ~/rk3328-roc-cc.dtb
# make a copy for experements
cp ~/rk3328-roc-cc_stock.dts ~/rk3328-roc-cc_stock_test_lcd.dts
# Edit the dts file
nano ~/rk3328-roc-cc_stock_test_lcd.dts
# find a "compatible = "ilitek,ili9341";" line and replace it to "compatible = "ilitek,st7796";"
# basically change ili9341 to st7796
# save and exit
# 
# convert dts -> dtb
dtc -I dts -O dtb -o ~/rk3328-roc-cc_stock_test_lcd.dtb ~/rk3328-roc-cc_stock_test_lcd.dts
# Move result to the boot folder
sudo cp  ~/rk3328-roc-cc_stock_test_lcd.dtb /boot/dtb/rockchip/rk3328-mkspi.dtb
# reboot
sudo reboot
```
Please check how screen works (image and touchscreen)

PS: Not all screens are working with modern 6.6 kernel. Please see [comment and related discussion](https://github.com/redrathnure/armbian-mkspi/issues/33#issuecomment-2549609371) for more details. Or starting from [this comment as possible solution for( X session only)](https://github.com/redrathnure/armbian-mkspi/issues/13#issuecomment-1875940052). From the other side, you may try some image with 5.x kernel, if external HDMI screen is so critical for your setup.

karabas2011, 2024-12-19:
Thanx!  
# find a "compatible = "ilitek,ili9341";" line and replace it to "compatible = "ilitek,st7796";"
yes, now resolution is correct.
Touchscreen works very wierd.  It seems no calibrations or different resolution, I cannot determine ...

redrathnure, 2024-12-19:
OK means you have st7796 based and screen and it works. 
Is touchscreen react to touch at all? Can you use some tools to catch events from it? And maybe use some stylus or any kind of plastic pointer, resistive touch is not so responsive, especially after experience with modern smartphones...

karabas2011, 2024-12-19:
Touchscreen works, but unpredictably for me. Some times  it seems randomly, sometime it feels it thinks that display is much bigger than it is.

redrathnure, 2024-12-19:
Is it works with OEM image? Could you check with kernel module is used here?

karabas2011, 2024-12-19:
> Is it works with OEM image? Could you check with kernel module is used here?

Yes. I have 3 printers to experiments and able to compare everything. I have several emmc cards to swap. One stock and one new image inside

redrathnure, 2024-12-19:
I am not sure can help more without hardware on my table. Or at least schematic. Only one last idea from my side:
you may take DTS from OEM MKSPI image and try to copy "spi_for_touch@1" part. They should have minimal diff so simple copy-paste may work... at least as an experiment....

Otherwise I would recommend to find the help on the relevant forums or chat group. Perhaps somebody (who has access to printer) already did this work.

karabas2011, 2024-12-20:
new image lsmod shows:
fb_st7796 
stock image lsmod shows 
fb_ili9341

How can I ( for experiment) force to load last one insteed of st7796?
i cannot rmmod it till it in use

karabas2011, 2024-12-20:
>  > . Only one last idea from my side:
> you may take DTS from OEM MKSPI image and try to copy "spi_for_touch@1" part. 

unfortunately they are absolutely identical.

redrathnure, 2024-12-20:
I would recommend to double check the latest messages and **OEM** DTS file**s**. [ads7846](https://github.com/redrathnure/armbian-mkspi/blob/c2ecf163de0900270c1b00e447a4ae62433828e6/config/boards/mkspi.csc#L16C10-L16C17) is not the same (the first is about input, the second is a driver for LCD controller). And DTS from OEM MKSPI and your board are not identical (or we look at different images)

karabas2011, 2024-12-20:
Are there any image for RK3328 that works with MKS Pi-TS35 ? except old busty?

redrathnure, 2024-12-20:
Custom images work MKS Pi-TS35, but this is for MKSPI or SKIPR boards. MKS images are stored [here](https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e), however yes, they are full of old software.

karabas2011, 2024-12-24:
I think new image uses some resources shared with touchscreen pins or something like this. 
Should I disable UART1 ( I looked in readme dtb overlays )

```
### Overlay details:
### mkspi-disable-lcd-spi
DTBO to disable spi_for_{lcd,touch} when enabling uart1
```

redrathnure, 2024-12-24:
This overlay came out of [this discussion](https://github.com/redrathnure/armbian-mkspi/issues/31#issuecomment-2240971371), please check it, I do not remember all details (it was renamed `mkspi-uart1` -> `mkspi-disable-lcd-spi`).

karabas2011, 2024-12-24:
temporary solution is
1 take a spare emmc and install Armbian_23.02.0-trunk_Mkspi_bullseye_edge_6.1.20.img
2 install kiauh and etc
3 copy and edit kingroon's rk3328-roc-cc.dtb with  ilitek,ili9341  to ilitek,st7796
4 copy all configs to ~/printer_data/config

Now I got touchscreen working. ethernet and wifi work too.
But cannot  connect to MCU_THR - timeout.
 I think to reflash it

redrathnure, 2024-12-24:
Basically it's https://github.com/redrathnure/armbian-mkspi/issues/37#issuecomment-2552207343 version + kiauh installation :)

> But cannot connect to MCU_THR - timeout. 

How you connect to your MCU? CANbus?

karabas2011, 2024-12-24:
V2.2 is simple serial 
ttyS0 in config
Hmm I replaced emmc with bookworm (with nonworking touch) and it connects THR and shows temperatures...
Wierd.

karabas2011, 2024-12-24:
tail of dmesg:
  
```
   12.843597] systemd[1]: Starting Journal Service...
[   13.140496] systemd[1]: Started Journal Service.
[   13.209987] systemd-journald[671]: Received client request to flush runtime journal.
[   13.213192] systemd-journald[671]: File /var/log/journal/39b9a8c486c04b4dafd42b21c7248936/system.journal corrupted or uncleanly shut down, renaming and replacing.
[   14.198475] systemd-journald[671]: Received client request to relinquish /var/log/journal/39b9a8c486c04b4dafd42b21c7248936 access.
[   16.126397] rk_gmac-dwmac ff550000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   16.200465] rk_gmac-dwmac ff550000.ethernet eth0: PHY [stmmac-1:00] driver [Rockchip integrated EPHY] (irq=POLL)
[   16.210072] rk_gmac-dwmac ff550000.ethernet eth0: No Safety Features support found
[   16.210105] rk_gmac-dwmac ff550000.ethernet eth0: PTP not supported by HW
[   16.214550] rk_gmac-dwmac ff550000.ethernet eth0: configuring for phy/rmii link mode
[   17.770061] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   18.923150] platform hdmi-sound: deferred probe pending
[   53.848694] irq 37: nobody cared (try booting with the "irqpoll" option)
[   53.848725] CPU: 0 PID: 77 Comm: irq/37-rk805 Tainted: G         C         6.1.20-rockchip64 #trunk
[   53.848746] Hardware name: Firefly roc-rk3328-cc (DT)
[   53.848757] Call trace:
[   53.848766]  dump_backtrace+0x140/0x150
[   53.848791]  show_stack+0x18/0x28
[   53.848806]  dump_stack_lvl+0x68/0x84
[   53.848829]  dump_stack+0x18/0x34
[   53.848849]  __report_bad_irq+0x4c/0xdc
[   53.848869]  note_interrupt+0x164/0x378
[   53.848890]  handle_irq_event_percpu+0x4c/0x60
[   53.848908]  handle_irq_event+0x48/0x98
[   53.848925]  handle_level_irq+0xd0/0x160
[   53.848945]  generic_handle_domain_irq+0x2c/0x48
[   53.848962]  rockchip_irq_demux+0x9c/0x220
[   53.848985]  generic_handle_domain_irq+0x2c/0x48
[   53.849003]  gic_handle_irq+0x98/0xd0
[   53.849018]  call_on_irq_stack+0x2c/0x54
[   53.849037]  do_interrupt_handler+0xd8/0xf0
[   53.849058]  el1_interrupt+0x38/0x70
[   53.849080]  el1h_64_irq_handler+0x18/0x28
[   53.849095]  el1h_64_irq+0x64/0x68
[   53.849110]  _raw_spin_unlock_irq+0x1c/0x48
[   53.849132]  irq_finalize_oneshot.part.56+0x68/0x108
[   53.849151]  irq_thread_fn+0x60/0xa8
[   53.849168]  irq_thread+0x11c/0x220
[   53.849185]  kthread+0xf0/0x108
[   53.849206]  ret_from_fork+0x10/0x20
[   53.849225] handlers:
[   53.849232] [<00000000f46b426f>] irq_default_primary_handler threaded [<00000000c62932c9>] regmap_irq_thread
[   53.849272] Disabling IRQ #37
user@bulleye:~$

```

karabas2011, 2024-12-24:
```
[    6.741321] systemd[1]: Finished Load Kernel Module configfs.
[    6.754257] systemd[1]: modprobe@drm.service: Succeeded.
[    6.756050] SPI driver ads7846 has no spi_device_id for ti,tsc2046
[    6.756649] SPI driver ads7846 has no spi_device_id for ti,ads7843
[    6.757219] SPI driver ads7846 has no spi_device_id for ti,ads7845
[    6.757786] SPI driver ads7846 has no spi_device_id for ti,ads7873
[    6.758383] systemd[1]: Finished Load Kernel Module drm.
```

karabas2011, 2024-12-24:
BTH from what version  extlinux was removed?

redrathnure, 2024-12-24:
> irq 37: nobody cared (try booting with the "irqpoll" option)

it's misconfiguration of IRQ for RK805 in OEM images (seems MKS copy-pastes this fail across  all their RK3328 based products)

> [    6.756050] SPI driver ads7846 has no spi_device_id for ti,tsc2046

should be in your OEM DTS: 
```
               spi_for_touch@1 {
			reg = <0x01>;
			compatible = "ti,tsc2046";
			pinctrl-names = "default";
```

IMO something is not OK with your DTS experiments. I would recommend try to extract and patch OEM DTS files from scratch.

> BTH from what version extlinux was removed?

No idea, perhaps you are asking something about core Debian or/and Armbian configuration.

karabas2011, 2025-01-09:
```
should be in your OEM DTS:

               spi_for_touch@1 {
			reg = <0x01>;
			compatible = "ti,tsc2046";
			pinctrl-names = "default";
```
Exactly so. 

`it's [misconfiguration of interruptions for RK805 in OEM images](https://github.com/redrathnure/armbian-mkspi/commit/2a37468b236bb436d371df9bd2985bb443fcf7ed) (seems MKS copy-pastes this fail across all their RK3328 based products)`

In my dts no marks <&gpio1> but digits. How to know what to change?
` interrupt-parent = <&gpio1>;`

in my DTS
```
pmic@18 {
			#clock-cells = <0x01>;
			#gpio-cells = <0x02>;
			clock-output-names = "xin32k\0rk805-clkout2";
			compatible = "rockchip,rk805";
			gpio-controller;
			interrupt-parent = <0x29>;
			interrupts = <0x18 0x08>; 
```

karabas2011, 2025-01-09:
I tried your last bulleye image today. It behaves like last bookworm I tried: all works except touchscreen.

redrathnure, 2025-01-09:
weird, I am out of ideas, sorry. It's hard to guess without an access to hardware.

karabas2011, 2025-01-10:
What file does contain pin definitions for DTS?
Such as <&gpio1> and etc

redrathnure, 2025-01-10:
You should travel through includes. Some of them on the same DTS folder, some of them are in `${kernel_root}/includes` one

karabas2011, 2025-01-10:
I am looking for where klipper  MCU_THR baudrate is set. It seems that latest firmware works on different baudrate/
Probably that is why in older linux version I cannot connect via serial to my THR ( which is latest firmware)

redrathnure, 2025-01-10:
What is  MCU_THR ? MCU is Kliper related term, most likely it is not about Kernel, DTS or other low level adaptions.

karabas2011, 2025-01-13:
> Cheetah V2.2 is simple serial ttyS0 in config Hmm I replaced emmc with bookworm (with nonworking touch) and it connects THR and shows temperatures... Wierd. All configs are the same It means something with ttyS0

I am able to connect to ttyS0 via minicom and see periodic garbage on screen. (bulleye image)
lsof shows that python tries ttyS0 to handshake.
Klipper tried to connect but failed.

karabas2011, 2025-01-22:
Goof News!
1 I found that correct dtb is located in /boot/dtb/ folder
2 I installed oldest bookworm Armbian_23.08.0-trunk_Mkspi_bookworm_current_6.1.55.img
3 replace DTB and all works except small screen resolution.
4 Using dtc I edited DTB to replace **ilitek,ili9341** to **ilitek,st7796**
System does not boot at all! 
5 Using hex editor I edited once more to be sure the file is the same.
Success! Now all works.

Next
I went to newest bookworm Armbian-unofficial_25.02.0-trunk_Mkspi_bookworm_current_6.12.8.img
Replace and rename dtb from #5
Touch does not work. Ok it partially works but unpredictably and very seldom.

redrathnure, 2025-01-22:
sounds like it's OK with kernel 6.1 but does not fully works with 6.12. Coyld you try a image with 6.6 and 6.9? Maybe something was changed in the latest kernel...

redrathnure, 2025-01-22:
Btw, in generally you may stay with the latest workin (6.1) image. It should solve you problem with eol OS, it isn't necessary to use cutting edge kernel.

karabas2011, 2025-01-22:
Next news
I installed last bookworm 6.1 - it works too  Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.1.78.img
no newer versions exist so
I installed first bookworm 6.6 - touhscreen problem!   Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.im
It means 6.6 kernel  is broken somewhere

karabas2011, 2025-01-22:
> sounds like it's OK with kernel 6.1 but does not fully works with 6.12. Coyld you try a image with 6.6 and 6.9? Maybe something was changed in the latest kernel...

How to get image with 6.9?

redrathnure, 2025-01-22:
> How to get image with 6.9?

Please check [releases ](https://github.com/redrathnure/armbian-mkspi/releases) or [release notes](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F1.0.1-25.2.0-trunk/readme.txt).

> no newer versions exist so
> I installed first bookworm 6.6 - touchscreen problem! Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17.im
>
>It means 6.6 kernel is broken somewhere instead of fixing

Please double check [release notes](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F1.0.1-25.2.0-trunk/readme.txt). I remember some problems with the first versions of 6.6. But then they disappeared with new LTS kernel. 

In generally 

> And if I want last bookworm I need to downgrade kernel...

It is not so easy. You may try to build a fresh image from old tag (which use 6.1 kernel). Or tray to restore 6.1 patches, pin the kernel with the latest armbian and try to build the image. However I am not sure it would work. 

In generally, if you found worked bookworm image, just stay with it. Use apt-update to update OS specific packages and stay with 6.1 (whatever version works for you) kernel. I am not sure you would really benefit from 6.1 -> 6.9 upgrade.

redrathnure, 2025-01-22:
Are you sure you have exactly `MKS PI-TS35` display? How it connected with your board? Could you share photos?

karabas2011, 2025-01-23:
[image]

redrathnure, 2025-01-24:
> What does it mean? 

It means on that period 6.6. Kernel has a problem with SPI/DMA, which leaded to half worked touchscreen. Sometimes it was reacted, sometimes not. And this means on that period of time all 6.1. patches were existed as well as Armbian support for it, so I managed to downgrade kernel version.

> Who decided to downgrade?

Sounds weird, assuming you are asking about author of custom images. 

BTW each release has a related git tag, each git tag has related commit history. So you may easily found related changes as well as commit comments for them.

karabas2011, 2025-02-06:
Sorry for wrong place to answer
But howto make custom boot picture?
I tryed mkbootsplash tool but it requires linux kernel headers to install.  i tryed to install headers via armbian-config but it seems do nothing...

redrathnure, 2025-02-06:
custom image should have it preinstalled. BTW boot picture should be just simple file somewere on the /boot partition. So you may just replace it and enable bootsplach in boot configuration

karabas2011, 2025-02-07:
I believe /boot/boot.bmp is for hdmi display. In my case it does nothing. I see  logo on my pi-ts35 that is differ from /boot one. 
All I can do is to turn it off/on.
Does is mean that I need to recompile the kernel?

redrathnure, 2025-02-07:
Kernel and boot loader are different modules

karabas2011, 2025-02-07:
Solved simply by modifying one of plymouth theme pics and then:
`plymouth-set-default-theme your_mod_theme ; update-initramfs -u`

closed 2025-02-27 by karabas2011