redrathnure/armbian-mkspi #39: [Bug]: The v1.0.0-25.2.0 gives me a blank screen on TS35

FabioM3, 2024-12-21:
### What happened?

I tried to update to 1.0.0-25.2.0-trunk - bookwork and jammy, but they give just a blank screen on TS35.
The last bullseye works normal.
I have a skipr v1.1.

And, there's no option to toggle GPIO in the settings? 

### How to reproduce?

intall latest version to emmc

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

redrathnure, 2024-12-21:
> I tried to update to

what does "update" mean? Was it a fresh new installation or you just update kernel packages?
Any /dev/fb* devices? Do you see any unusual or error dmesg  logs?

>  there's no option to toggle GPIO

Yeah, Armbian reworks armbian-config tool. There are some "surprises" here. It's out of my control, I would recommend to ask the question or report a bug in official Armbian forum.

FabioM3, 2024-12-21:
Is a fresh installation.

The GPIO thing, I think the GPIOs are all on by default.

redrathnure, 2024-12-21:
It's strange... I just updated my printer (SKIPR) to Noble current (6.6.65) and tested current and edge images on test stand (MKSPI and SKIPR) - everything works fine. 

In worst case you may stay with pre 1.0.0 image, there are no big functional changes.

FabioM3, 2024-12-21:
I tested bookworm an Jammy current 🤷‍♂️
But is ok, I will stay with bullseye for a few years.
Thanks for this incredible work!

closed 2024-12-21 by FabioM3