redrathnure/armbian-mkspi #41: Getting Help to make this work [SKIPR]

Hassla, 2025-01-07:
### Which feature would you like to have?

Hello,
apparently I am too stupid for this.
Since I can't get it to work properly.
I don't even know if this is the proper place to complain about my issue.

- Using 22.05 in VBox
- Updating it after installing
- Open Terminal
`apt-get -y install git`
`git clone https://github.com/redrathnure/armbian-mkspi.git`
`cd armbian-mkspi`
`./compile.sh  BOARD=mkspi BRANCH=current RELEASE=noble    BSPFREEZE=yes BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img INSTALL_HEADERS=yes BUILD_KSRC=yes INSTALL_KSRC=yes`

- Via FileBrowser I copy the image to my W10 PC and use Balena Etcher to write it to the EMMC.
- Install it in my Printer
- Choose root password, normal user + password
- bash
- language and location

- Access via Putty
- update and installing everything except Mainsail via Kiuah
- reboot
- compiling the Robin Part of the board with [these settings](https://klipper.discourse.group/t/mks-skipr-can-bus/5377)
- Copy the klipper.bin via WinSCP to the SDCard, rename it to "mks_skipr.bin"
- Put it in the board and reboot

The "mks_skipr.bin" was not deleted or changed, so no upload of the new firmware to the robin part.

After many tries, I tried to compile another VBox, was useing this [video](https://www.youtube.com/watch?v=Fg966ivZlrc) as reference.
- 24.04
- Terminal
`apt-get -y install git
git clone --depth=1 --branch=main https://github.com/armbian/build
cd build`
`./compile.sh  BOARD=mkspi BRANCH=current RELEASE=noble    BSPFREEZE=yes BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img INSTALL_HEADERS=yes BUILD_KSRC=yes INSTALL_KSRC=yes`

With this I was able to flash a new firmware onto the Robin and the CAN-Toolhead but now I get this problem:
"Internal error during connect: module 'can' has no attribute 'Message'"

Thanks for your help in advance, this is the first time I have compiled an armbian build myself.
Or does anyone have a Image i can have, which I can directly write to my emmc?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2025-01-07:
Hi @Hassla ,

first of all this is cli which I use to prepare release images `./compile.sh  BOARD=mkspi BRANCH=edge    RELEASE=noble    BSPFREEZE=yes BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img INSTALL_HEADERS=yes BUILD_KSRC=yes INSTALL_KSRC=yes` It looks similar to your variant. So if you do not have any personal customization/patches/etc you may use [release images](https://github.com/redrathnure/armbian-mkspi/releases) and save a "few minutes" of waiting. 

In any case your "Using 22.05 in VBox" process looks totally fine. And I do belie that Armbian on MKSPI/SKIPR does **not** affect  "MCU/Robin" part. `Copy the klipper.bin via WinSCP to the SDCard, rename it to "mks_skipr.bin"` - this naming is correct too. Tricky part may be to proper trigger flashing process. Usually board detects microsd cart inserting event and try to flash it, however sometimes I've needed to use hardware "reset" button on the **"MCU/Robin"** part (not MKSPI/SKIPR main one). If you have some issues on this step (or e.g. use SD card extenders) you may simply power down plate (not reset but switch off power), insert microSD and then power it on. It takes a minute more, but is less error prove process:)

And about `"Internal error during connect: module 'can' has no attribute 'Message'"` It looks like Python related message (means Klipper related). I had something sliglty different on the "relatively fresh" Klipper version. For me fix was:
```
# Temp Klipper fix for:
# ModuleNotFoundError: No module named 'pkg_resources'

cd ~/klippy-env/bin
./pip install --upgrade python-can==4.3
```
So I would recommend to double check Klipper bugs and/or their forum. In worst case you may try to downgrade the Klipper for -2-3 weeks version. I am not sure how to do this via `Kiuah` but (in theory) it should be possible to do manually.

Hassla, 2025-01-08:
Damn, that was fast. Thanks a lot.
Now I feel even more stupid, since I didn't saw the release section. Sometimes github is confusing to me.
I am going to try to use one of the release versions. Yes the variant was directly copied from the github page.
The usage of the reset button on the Robin is a good idea, I never had a problem with the flashing of the Robin till now, so I just forgot about the reset button.

I've used the temporary fix via. a python upgrade but it didn't help. 

I come back to you after I tried one of the releases.
Thank you!

Hassla, 2025-01-08:
Hello,
I've tried it again, this time with one of the release versions. "Noble Current"
The installation on the Pi worked flawlessly, this time the flashing of the firmware on the Robin also worked perfectly.
But I cant get a connection via CANBus between the Pi and the Robin
 ```
mcu 'mcu': Starting CAN connect
Created a socket
mcu 'mcu': Unable to open CAN port: Failed to transmit: Network is down [Error Code 100]
webhooks client 281473085946576: New connection
webhooks client 281473085946576: Client info {'program': 'Moonraker', 'version': 'v0.9.3-4-ga4604e3'}
 ```

Do you know where it went wrong?

redrathnure, 2025-01-08:
Kinda dummy answer: "Network is down" :)

There are may be plenty of causes for this. I would double check following points:
- Physical layer:
    - "Pi" <--> "MCU" connected using USB
    - "MCU" <--> "TH" connected using proper can cable (cable and sockets are in good shape, there are power on TH side and data lines are connected correctly). Better if you have already worked configuration to check TH and cable.
    - TH in CAN mode and with proper terminator connected on the both sides.
    - MCU and TH are not in DFU mode. Means flashing was successful and devices were reboot after it
 - Software part:
     - "Pi",  "MCU" and TH were flashed with the same version of Klipper
     - "Pi",  "MCU" and TH have the same baudrate config
     - MCU was flashed with USB to CAN Bridge mode (not UART)
     - CAN interface are correctly configured in Armbian (including baudrate), can interface is able to up from command line and shows correct status
     - can scan tool see MCU device

Only after this precondition it has sense to start Klipper on Pi part. And at least at the first time or  a few prints it has sense to monitor amount of error packages. It must not increase during normal work.

Hassla, 2025-01-09:
So, I've apparently got it to work.
Pretty timeconsuming, if you have no idea, what you are doing.

After flashing the Pi and the MCU, I got the MCU error.
[# Temp Klipper fix for:
# ModuleNotFoundError: No module named 'pkg_resources'

cd ~/klippy-env/bin
./pip install --upgrade python-can==4.3](url)

The cannetwork was not operational.
Fixed with [this](https://canbus.esoterical.online/Getting_Started.html) 

After this, I could finally flash the TH via CAN with Katapult.
Then the mcu "rpi" was not found, this was fixed with [this](https://www.klipper3d.org/RPi_microcontroller.html)

Now I can finally use my Printer again. Thank you.
There are still problems, with the camera having like 1 FPS and the rotating of the screen.

Did I took a wrong turn somewhere or is this normal for this kind of image?
- Flashing the Pi via make flash
- Make the CAN network functional

Nevertheless, thank you for your help

redrathnure, 2025-01-09:
Congratulations and welcome to the cub for those who love and hate CAN bus :)

> ModuleNotFoundError: No module named 'pkg_resources'

It's issue with the newest Klipper. IMO it's a bug, I have not seen it a few months before

> The cannetwork was not operational.
> Fixed with [this](https://canbus.esoterical.online/Getting_Started.html)

Came from relatively fresh changes in Debian distros. New simple configuration from official Klipper side does not works anymore. I put [a few words on the main readme.](https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus)

> Then the mcu "rpi" was not found, this was fixed with [this](https://www.klipper3d.org/RPi_microcontroller.html)

IMO it's pretty standard pitfall (super confusing first time you see it) :) A briefly described in [the readme](https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#adxl345spi-usage) and (I believe) in official Klipper instructions.

> There are still problems, with the camera having like 1 FPS and the rotating of the screen.

Oh, unfortunately I cannot help here. I have not configured camera yet, neither on tests stand nor in my printers.

In generally I would say it's pretty normal learning path for so complex system :)

redrathnure, 2025-01-09:
BTW, feel free to close the ticket if the issue(s) was solved.

Hassla, 2025-01-09:
Yes, the CANBus was already a problem 2 years prior as I've first installed klipper on my SKIPR.

redrathnure, 2025-01-09:
Yeah, unfortunately few dozens of long cables give even more problem than CAN bus :/ Especially on the relatively big printers