redrathnure/armbian-mkspi #44: [Feature Request/Question]: no "freeze" in armbian-config?

bsafh, 2025-01-18:
### Which feature would you like to have?

Yesterday I downloaded and installed the latest version [](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F1.0.1-25.2.0-trunk/Armbian-unofficial_25.02.0-trunk_Mkspi_bookworm_current_6.12.8.img.xz) on a MKS SKIPR Mini 1.0 for a TronXY X5SA Pro Klipper Upgrade Kit.

All went well. but when I start armbian-config, there is no "freeze" option anywhere in the menues. Nothing of that kind.

Well, before I remembered that this is needed, I did run an
```
apt update
apt upgrade
```
without freeze. There must have been some Kernel relaxed upgrades as a new kernel boot image and initramfs has been created and installed.

To my surprise, the system booted just fine with these changes.??

Is the kernel freeze no longer needed?

 

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2025-01-18:
Yes, looks like there are no more option ins armbian-config. however I still build images with related flag, so executing `apt upgrade/dist-upgrade` should be safe.

bsafh, 2025-01-18:
let's hope for the best ...

this looks like there have been updates and they have been installed ... 

```

Start-Date: 2025-01-04  19:16:42
Commandline: apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/l
inux-u-boot-mkspi-current_25.02.0-trunk_arm64__2022.07-Se092-Pe0f5-H2f98-Vb99b-Bdacf-R448a.deb
Install: linux-u-boot-mkspi-current:arm64 (25.02.0-trunk)
End-Date: 2025-01-04  19:16:44
Start-Date: 2025-01-04  19:16:52
Commandline: apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/l
inux-image-current-rockchip64_25.02.0-trunk_arm64__6.12.8-S77f8-D6d0e-Pa8e3-Ca562Heb51-HK01ba-Vc222-B7bd4-R448a.deb
Install: linux-image-current-rockchip64:arm64 (25.02.0-trunk)
End-Date: 2025-01-04  19:17:00
Start-Date: 2025-01-04  19:17:06
Commandline: apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/l
inux-dtb-current-rockchip64_25.02.0-trunk_arm64__6.12.8-S77f8-D6d0e-Pa8e3-Ca562Heb51-HK01ba-Vc222-B7bd4-R448a.deb
Install: linux-dtb-current-rockchip64:arm64 (25.02.0-trunk)
End-Date: 2025-01-04  19:17:07
Start-Date: 2025-01-04  19:17:14
Commandline: apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/l
inux-headers-current-rockchip64_25.02.0-trunk_arm64__6.12.8-S77f8-D6d0e-Pa8e3-Ca562Heb51-HK01ba-Vc222-B7bd4-R448a.deb
Install: linux-headers-current-rockchip64:arm64 (25.02.0-trunk), libelf-dev:arm64 (0.188-2.1, automatic), zlib1g-dev:arm64 (1:1.2.13.dfsg-1,
 automatic), libssl-dev:arm64 (3.0.15-1~deb12u1, automatic)
End-Date: 2025-01-04  19:21:26
Start-Date: 2025-01-04  19:21:37
Commandline: apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/a
rmbian-firmware_25.02.0-trunk_all__1-SAdc2d-B6c7f-R448a.deb
Install: armbian-firmware:arm64 (25.02.0-trunk)
End-Date: 2025-01-04  19:21:48
```

Hope that doesn't break things now ...

redrathnure, 2025-01-18:
I managed to merge almost all customization to main Armbian project. Means kernels in this and Armbian repositories should be the same. 
So updates should be safe, at least till ether you have no any local customization (e.g. changed DTS files) or we have some un-merged patches here.

bsafh, 2025-01-18:
MakerBase have these two files in 
`/root/ `
of their Buster Image
```
-rw-r--r-- 1 root   root     64K Aug  1  2023 rk3328-roc-cc.dtb
-rw-r--r-- 1 mks    mks      75K Jul 24  2023 rk3328-roc-cc.dts
```

No ideas why though ...

redrathnure, 2025-01-18:
leftovers from installation process? dts is human readable/unpacked Device Tree Source descriptor. dtb is packed version, at least should be. Also dtb should be the same as in /boot dir, at least if my feeling is right. 

So just guessing, how MKS engineers make adaption for OEM image: 
1. Run default OEM image, probably from MKSPI one
2. extract /boot/*.dtb to the home/dts. 
3. Patch dts, pack and replace original /boot/*.dtb
4. Install all custom services, software etc.

Please keep in mind it's just my personal guess, I have no idea who and how prepares these images and why they left these files in result image.

bsafh, 2025-01-19:
There are two uart-* binaries there as well ....

armbian-config seems to have the "kernel-updates on hold" again, it's just deeper in the menues and not called "freeze" any more.

[image]

redrathnure, 2025-01-21:
OK, need to adjust readme.