redrathnure/armbian-mkspi #45: ADS7846 Touchscreen with MKS SKIPR Mini v1.0 / MKSCLIENT Software by Makerbase

bsafh, 2025-01-18:
### Which feature would you like to have?

I'm trying to use armbian-mkspi on a
MKS SKIPR Mini v1.0 on a TronXY printer with the Tronxy Klipper Upgrade Kit (which is this board essentially)

The Tronxy/Makerbase Upgrade kit uses a special display ...

`hwinfo` says:

```
Model: "ADS7846 Touchscreen"
  Device: 0x1ea6 "ADS7846 Touchscreen"
  Compatible to: int 0x0210 0x0000
```
```
Interrupt
58:          0          0          0          0  rockchip_gpio_irq  22 Edge      ads7846
```
```

         ads7846: /devices/platform/ff190000.spi/spi_master/spi0/spi0.1
         ads7846: module = ads7846

```

```
input: name = mouse0, path = /devices/platform/ff190000.spi/spi_master/spi0/spi0.1/input/input0/mouse0
    dev = 13:32
    input device: bus = spi, bus_id = spi0.1 driver = ads7846
      path = /devices/platform/ff190000.spi/spi_master/spi0/spi0.1
  input: name = input0, path = /devices/platform/ff190000.spi/spi_master/spi0/spi0.1/input/input0
    no dev - ignored
  input: name = event0, path = /devices/platform/ff190000.spi/spi_master/spi0/spi0.1/input/input0/event0
    dev = 13:64
    input device: bus = spi, bus_id = spi0.1 driver = ads7846
      path = /devices/platform/ff190000.spi/spi_master/spi0/spi0.1

```

```
>> input.2: input
----- /proc/bus/input/devices -----
  I: Bus=001c Vendor=0000 Product=1ea6 Version=0000
  N: Name="ADS7846 Touchscreen"
  P: Phys=spi0.1/input0
  S: Sysfs=/devices/platform/ff190000.spi/spi_master/spi0/spi0.1/input/input0
  U: Uniq=
  H: Handlers=mouse0 event0 
  B: PROP=0
  B: EV=b
  B: KEY=400 0 0 0 0 0
  B: ABS=1000003

```

```
bus = 28, name = ADS7846 Touchscreen
  handlers = mouse0 event0
  key = 000000000000040000000000000000000000000000000000000000000000000000000000000000000000000000000000
  abs = 0000000001000003
  mouse buttons = 0
  mouse wheels = 0
  is_mouse = 1
  is_joystick = 0

```
```

P: /devices/platform/ff190000.spi/spi_master/spi0/spi0.1
  M: spi0.1
  R: 1
  U: spi
  V: ads7846
  E: DEVPATH=/devices/platform/ff190000.spi/spi_master/spi0/spi0.1
  E: SUBSYSTEM=spi
  E: DRIVER=ads7846
  E: OF_NAME=spi_for_touch
  E: OF_FULLNAME=/spi@ff190000/spi_for_touch@1
  E: OF_COMPATIBLE_0=ti,tsc2046
  E: OF_COMPATIBLE_N=1
  E: MODALIAS=spi:tsc2046

```

```
P: /devices/platform/ff190000.spi/spi_master/spi0/spi0.1/input/input0
  M: input0
  R: 0
  U: input
  E: DEVPATH=/devices/platform/ff190000.spi/spi_master/spi0/spi0.1/input/input0
  E: SUBSYSTEM=input
  E: PRODUCT=1c/0/1ea6/0
  E: NAME="ADS7846 Touchscreen"
  E: PHYS="spi0.1/input0"
  E: PROP=0
  E: EV=b
  E: KEY=400 0 0 0 0 0
  E: ABS=1000003
  E: MODALIAS=input:b001Cv0000p1EA6e0000-e0,1,3,k14A,ra0,1,18,mlsfw
  E: USEC_INITIALIZED=9475895
  E: ID_INPUT=1
  E: ID_INPUT_TOUCHSCREEN=1
  E: ID_PATH=platform-ff190000.spi-cs-1
  E: ID_PATH_TAG=platform-ff190000_spi-cs-1
  E: ID_FOR_SEAT=input-platform-ff190000_spi-cs-1
  E: TAGS=:seat:
  E: CURRENT_TAGS=:seat:

```

They (Makerbase) seem to use this with a self written software named "mksclient" - the binary is on their supplied Klipper Armbian image, but the source code is not.

It lives in
```
file /home/mks/Desktop/myfile/ws/build/mksclient 
/home/mks/Desktop/myfile/ws/build/mksclient: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, BuildID[sha1]=598e6006a109ab1b43af0e2f8fd62d5cc5721c05, not stripped
root@x5sa:/lib# 
```

and uses these libraries
```

x5sa:~:# ldd /home/mks/Desktop/myfile/ws/build/mksclient 
        linux-vdso.so.1 (0x0000ffff90700000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff9068d000)
        libboost_system.so.1.67.0 => /lib/aarch64-linux-gnu/libboost_system.so.1.67.0 (0x0000ffff90678000)
        libcurl.so.4 => /lib/aarch64-linux-gnu/libcurl.so.4 (0x0000ffff905e1000)
        libpython3.7m.so.1.0 => /lib/aarch64-linux-gnu/libpython3.7m.so.1.0 (0x0000ffff900de000)
        libwpa_client.so => /lib/libwpa_client.so (0x0000ffff900c3000)
        libsqlite3.so.0 => /lib/aarch64-linux-gnu/libsqlite3.so.0 (0x0000ffff8ff96000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff8fe0b000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff8fd4e000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff8fd2a000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff8fbb9000)
        /lib/ld-linux-aarch64.so.1 (0x0000ffff906d2000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000ffff8fba1000)
        libnghttp2.so.14 => /lib/aarch64-linux-gnu/libnghttp2.so.14 (0x0000ffff8fb6a000)
        libidn2.so.0 => /lib/aarch64-linux-gnu/libidn2.so.0 (0x0000ffff8fb3d000)
        librtmp.so.1 => /lib/aarch64-linux-gnu/librtmp.so.1 (0x0000ffff8fb12000)
        libssh2.so.1 => /lib/aarch64-linux-gnu/libssh2.so.1 (0x0000ffff8fad6000)
        libpsl.so.5 => /lib/aarch64-linux-gnu/libpsl.so.5 (0x0000ffff8fab5000)
        libssl.so.1.1 => /lib/aarch64-linux-gnu/libssl.so.1.1 (0x0000ffff8fa1a000)
        libcrypto.so.1.1 => /lib/aarch64-linux-gnu/libcrypto.so.1.1 (0x0000ffff8f771000)
        libgssapi_krb5.so.2 => /lib/aarch64-linux-gnu/libgssapi_krb5.so.2 (0x0000ffff8f718000)
        libkrb5.so.3 => /lib/aarch64-linux-gnu/libkrb5.so.3 (0x0000ffff8f62d000)
        libk5crypto.so.3 => /lib/aarch64-linux-gnu/libk5crypto.so.3 (0x0000ffff8f5eb000)
        libcom_err.so.2 => /lib/aarch64-linux-gnu/libcom_err.so.2 (0x0000ffff8f5d7000)
        libldap_r-2.4.so.2 => /lib/aarch64-linux-gnu/libldap_r-2.4.so.2 (0x0000ffff8f575000)
        liblber-2.4.so.2 => /lib/aarch64-linux-gnu/liblber-2.4.so.2 (0x0000ffff8f555000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000ffff8f52b000)
        libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000ffff8f4eb000)
        libcrypt.so.1 => /lib/aarch64-linux-gnu/libcrypt.so.1 (0x0000ffff8f4a4000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffff8f490000)
        libutil.so.1 => /lib/aarch64-linux-gnu/libutil.so.1 (0x0000ffff8f47c000)
        libunistring.so.2 => /lib/aarch64-linux-gnu/libunistring.so.2 (0x0000ffff8f2f6000)
        libgnutls.so.30 => /lib/aarch64-linux-gnu/libgnutls.so.30 (0x0000ffff8f11e000)
        libhogweed.so.4 => /lib/aarch64-linux-gnu/libhogweed.so.4 (0x0000ffff8f0d7000)
        libnettle.so.6 => /lib/aarch64-linux-gnu/libnettle.so.6 (0x0000ffff8f090000)
        libgmp.so.10 => /lib/aarch64-linux-gnu/libgmp.so.10 (0x0000ffff8f008000)
        libgcrypt.so.20 => /lib/aarch64-linux-gnu/libgcrypt.so.20 (0x0000ffff8ef3b000)
        libkrb5support.so.0 => /lib/aarch64-linux-gnu/libkrb5support.so.0 (0x0000ffff8ef1c000)
        libkeyutils.so.1 => /lib/aarch64-linux-gnu/libkeyutils.so.1 (0x0000ffff8ef07000)
        libresolv.so.2 => /lib/aarch64-linux-gnu/libresolv.so.2 (0x0000ffff8eee1000)
        libsasl2.so.2 => /lib/aarch64-linux-gnu/libsasl2.so.2 (0x0000ffff8eeb6000)
        libp11-kit.so.0 => /lib/aarch64-linux-gnu/libp11-kit.so.0 (0x0000ffff8ed70000)
        libtasn1.so.6 => /lib/aarch64-linux-gnu/libtasn1.so.6 (0x0000ffff8ed4b000)
        libgpg-error.so.0 => /lib/aarch64-linux-gnu/libgpg-error.so.0 (0x0000ffff8ed1b000)
        libffi.so.6 => /lib/aarch64-linux-gnu/libffi.so.6 (0x0000ffff8ed03000)
x5sa:~:# 

```
Armbian Bookworm does not provide all of those libraries

`libboost_system.so.1.67.0 => /lib/aarch64-linux-gnu/libboost_system.so.1.67.0 (0x0000ffff90678000)`

this seems to be in `libboost-all-dev` in version 1.74.0, so I might get away with just creating a symlink from the new version with the filename of the old version.

same with `/lib/aarch64-linux-gnu/libpython3.7m.so.1.0 (0x` which hopefully ca be cheated with ` libpython3.11....` 

next is `/lib/libwpa_client.so' - I can't find that in any package for Bookworm, only `/usr/lib/aarch64-linux-gnu/libwpa_client.a` , not `.so` 

any idea?
copying the old `libwpa_client.so' from the Buster Armbian on the SKIPR Mini Image does not work, it results in
```
root@x5sa:/lib# /home/mks/Desktop/myfile/ws/build/mksclient 
/home/mks/Desktop/myfile/ws/build/mksclient: symbol lookup error: /home/mks/Desktop/myfile/ws/build/mksclient: undefined symbol: _ZN5boost6system15system_categoryEv
```

and of course you can't just copy the .a to a .so ... that can't load dynamically ...

so, is there any version of libwpa_client.so for Armbian Bookworm?

Or: Does anyone perhaps can get a hold of the source code of this thing ... ?

or any other idea on how to proceed ?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2025-01-18:
Oh, this is pretty complex topic IMO.

Let's start from easies and maybe most unrealistic options:
what about contact vendor or/and makerbase support to get sources or something like this? At the end thay. should be intrested in support of their product. 

Then, I am not sure about aproach of copying binary files with symlinking libs for different versions. IMO it's unrealistic.

BTW, your cli output is related to the touchscreen but not for display device. For output device I would expect something vga or fb/frame buffer related.

redrathnure, 2025-01-18:
Hm, pretty interesting. Please drop an update here, at least if display and touchscreen do/do not work. Perhaps someone instrested in this hardware too..

bsafh, 2025-01-18:
> BTW, your cli output is related to the touchscreen but not for display device. For output device I would expect something vga or fb/frame buffer related.

you're totally right, I just mixed it up. There is only a touchscreen on the printer, no separate display.

bsafh, 2025-01-18:
the service starts (without actual touchscreen attached):

```
root@x5sa:~# systemctl status makerbase-client.service   
● makerbase-client.service - MakerbaseClient
     Loaded: loaded (/lib/systemd/system/makerbase-client.service; enabled; preset: enabled)
     Active: active (running) since Sat 2025-01-18 19:17:42 CET; 7min ago
   Main PID: 1175 (bash)
      Tasks: 5 (limit: 974)
     Memory: 21.8M
        CPU: 365ms
     CGroup: /system.slice/makerbase-client.service
             ├─1175 bash /home/mks/Desktop/myfile/ws/start.sh
             └─1204 /root/mksclient-static

Jan 18 19:25:17 x5sa bash[1204]: [2025-01-18 19:25:17] [fail] WebSocket Connection Unknown - "" /websocket 0 websocketpp.transport:2 Underlying Transport Error
Jan 18 19:25:17 x5sa bash[1204]: [2025-01-18 19:25:17] [info] asio async_shutdown error: system:9 (Bad file descriptor)
Jan 18 19:25:17 x5sa bash[1204]: WebSocket连接失败Jan 18 19:25:17 x5sa bash[1204]: 尝试重新连接...
Jan 18 19:25:22 x5sa bash[1204]: [2025-01-18 19:25:22] [info] asio async_resolve error: system:16 (Device or resource busy)
Jan 18 19:25:22 x5sa bash[1204]: [2025-01-18 19:25:22] [info] Error getting remote endpoint: system:9 (Bad file descriptor)
Jan 18 19:25:22 x5sa bash[1204]: [2025-01-18 19:25:22] [fail] WebSocket Connection Unknown - "" /websocket 0 websocketpp.transport:2 Underlying Transport Error
Jan 18 19:25:22 x5sa bash[1204]: [2025-01-18 19:25:22] [info] asio async_shutdown error: system:9 (Bad file descriptor)
Jan 18 19:25:22 x5sa bash[1204]: WebSocket连接失败Jan 18 19:25:22 x5sa bash[1204]: 尝试重新连接...
```

The original version on the MKS Buster image shows this:

```
x5sa:multi-user.target.wants:# systemctl status makerbase-byid.service   
● makerbase-byid.service - get usb id
   Loaded: loaded (/lib/systemd/system/makerbase-byid.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2025-01-13 17:40:10 CET; 5 days ago
 Main PID: 1425 (code=exited, status=0/SUCCESS)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
x5sa:multi-user.target.wants:# systemctl status makerbase-client.service
● makerbase-client.service - MakerbaseClient
   Loaded: loaded (/lib/systemd/system/makerbase-client.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2025-01-13 17:40:10 CET; 5 days ago
 Main PID: 1438 (bash)
    Tasks: 8 (limit: 998)
   Memory: 4.1M
   CGroup: /system.slice/makerbase-client.service
           ├─1438 /usr/bin/bash /home/mks/Desktop/myfile/ws/start.sh
           └─1459 /home/mks/Desktop/myfile/ws/build/mksclient

Jan 18 19:25:11 x5sa bash[1438]: current_page_id = 41
Jan 18 19:25:11 x5sa bash[1438]: [Aug 26 2024][17:42:44] /home/l123456/share/responsity/serial_screen_project/chuang_xin_yuan/klp-chuangxingyuan/src/refreshui.cpp: 350
Jan 18 19:25:11 x5sa bash[1438]: current_print_state = standby idle_timeout_state =  webhooks_state =
Jan 18 19:25:11 x5sa bash[1438]: [Aug 26 2024][17:42:44] /home/l123456/share/responsity/serial_screen_project/chuang_xin_yuan/klp-chuangxingyuan/src/refreshui.cpp: 349
Jan 18 19:25:11 x5sa bash[1438]: current_page_id = 41
Jan 18 19:25:11 x5sa bash[1438]: [Aug 26 2024][17:42:44] /home/l123456/share/responsity/serial_screen_project/chuang_xin_yuan/klp-chuangxingyuan/src/refreshui.cpp: 350
Jan 18 19:25:11 x5sa bash[1438]: current_print_state = standby idle_timeout_state =  webhooks_state =
Jan 18 19:25:11 x5sa bash[1438]: [Aug 26 2024][17:42:44] /home/l123456/share/responsity/serial_screen_project/chuang_xin_yuan/klp-chuangxingyuan/src/refreshui.cpp: 349
Jan 18 19:25:11 x5sa bash[1438]: current_page_id = 41
Jan 18 19:25:11 x5sa bash[1438]: [Aug 26 2024][17:42:44] /home/l123456/share/responsity/serial_screen_project/chuang_xin_yuan/klp-chuangxingyuan/src/refreshui.cpp: 350
x5sa:multi-user.target.wants:# 

```

redrathnure, 2025-01-18:
> There is only a touchscreen on the printer, no separate display.

For me it does not have sense. You should see some image to be displayed, especially in case of Klipper. Otherwise what the point in touchscreen...

In generally it remind me [this ](https://github.com/redrathnure/armbian-mkspi/issues/42) and [that ](https://github.com/redrathnure/armbian-mkspi/issues/37) discussion. Perhaps you may ask (or get some ideas) how they make screen and wifi work.

And just for the infor:
MKSPI and SKIPR use two serial buses, one to connect display and second for touchscreen. There are 2 declarations in DTS file, two separated kernel modules but one single "cable" connection screen module and main PCB. Ideally would be the good idea firstly understand hardware (which bus and controllers are used for display and touch) and then try to configure DTS file for it. But this works only if you have access to the schematic for the board (e.g. if customer support sent it to you). Otherwise you may try to compare unpacked DTS files from OEM and costume images (basically the same story as in WIFI related issue)

karabas2011, 2025-01-21:
There is no mksclient in Kingroon's buster. ( modified mkspi )
I guess they hardcoded it or something like this.
I see dmesg lines that are absent in DTB

bsafh, 2025-01-22:
> Hm, pretty interesting. Please drop an update here, at least if display and touchscreen do/do not work. Perhaps someone instrested in this hardware too..

well, it was just a slight chance.

the statically linked mksclient binary does not work on the armbian image

```
klipper@x5sa:~/printer_data/config$ systemctl status makerbase-client
● makerbase-client.service - MakerbaseClient
     Loaded: loaded (/lib/systemd/system/makerbase-client.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-01-22 17:59:15 CET; 33min ago
   Main PID: 1116 (bash)
      Tasks: 5 (limit: 974)
     Memory: 13.0M
        CPU: 857ms
     CGroup: /system.slice/makerbase-client.service
             ├─1116 bash /home/mks/Desktop/myfile/ws/start.sh
             └─1144 /root/mksclient-static

Jan 22 18:32:48 x5sa bash[1144]: [2025-01-22 18:32:48] [fail] WebSocket Connection Unknown - "" /websocket 0 websocketpp.transport:2 Underlying Transport Error
Jan 22 18:32:48 x5sa bash[1144]: [2025-01-22 18:32:48] [info] asio async_shutdown error: system:9 (Bad file descriptor)
Jan 22 18:32:48 x5sa bash[1144]: WebSocket连接失败
Jan 22 18:32:48 x5sa bash[1144]: 尝试重新连接...
Jan 22 18:32:53 x5sa bash[1144]: [2025-01-22 18:32:53] [info] asio async_resolve error: system:16 (Device or resource busy)
Jan 22 18:32:53 x5sa bash[1144]: [2025-01-22 18:32:53] [info] Error getting remote endpoint: system:9 (Bad file descriptor)
Jan 22 18:32:53 x5sa bash[1144]: [2025-01-22 18:32:53] [fail] WebSocket Connection Unknown - "" /websocket 0 websocketpp.transport:2 Underlying Transport Error
Jan 22 18:32:53 x5sa bash[1144]: [2025-01-22 18:32:53] [info] asio async_shutdown error: system:9 (Bad file descriptor)
Jan 22 18:32:53 x5sa bash[1144]: WebSocket连接失败
Jan 22 18:32:53 x5sa bash[1144]: 尝试重新连接...
klipper@x5sa:~/printer_data/config$
```