redrathnure/armbian-mkspi #47: [Question] Upgrade to latest trunk and kernel from

Phil1988, 2025-01-25:
### Which feature would you like to have?

Hello @redrathnure,

Thanks again for your great work! 
I'm the guy who [opened the door to the QIDI mainboards](https://github.com/redrathnure/armbian-mkspi/issues/21#issuecomment-1926636610) with your fantastic work here!

There are quite a few users on the QIDI boards X-4, X-6, and X-7 with no issues 👍. 
In case you missed it, here is a thank you in my [video](https://youtu.be/_ZmSTJBWUus?feature=shared&t=2726) and the [FreeDi project](https://github.com/Phil1988/FreeDi).

We are currently using your **0.3.6-24.2.1** release, but I saw your recent releases and have a few questions:

1. I'm confused about the versions/releases. For example, your release table says **Current 6.6** (and Edge 6.12). I guess it's the kernel. 
  The attached images for bookworm being **Current 6.12.8** (and edge 6.13.0). Is it a mistake in the table?

2. Is it possible to upgrade from your 0.3.6-24.2.1 to 1.0.1-25.2.0-trunk? 
  How would you approach this or, better yet, recommend doing it?
  You are the professional here, and I would like to know if the running system could be upgraded without having to rebuild or flash the OS.
  Pperhaps by using something like this?

```
sudo apt update
sudo apt upgrade
sudo dpkg -i <latest_kernel>.deb
```
   

3. Jumping from 0.3.6 to 1.0.1 seems like a huge leap for me. 
  I would like to know your opinion: would you recommend upgrading? 
  What benefits/improvements do you see in 25.2.0-trunk with the latest kernel?

Thanks in advance for your help and good work.
Your OS base 

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2025-01-25:
Hi @Phil1988, appreciate your feedback and thank you for contribution.

**TLTR:**
1. If you already have fully working custom image I would not recommend update without concrete needs. Exception would be a wish to jump to the other distro or updating related manuals/instructions.
2. `apt upgrade/dist-upgrade` do not touch Armbian tools and kernel related components. In theory you may do this manually (e.g. via `dpkg -i` or afficial armbian packages), however I have newer tested it.
3. If you decided to go ahead with full update, I would recommend flashing a new image and reconfigure everything from scratch. And please note a) backup is your best friend b) be ready to spend some time to update your instructions (some files was changed and no more `/etc/interfaces/*` files).
4. and yes, release notes may have a typos.

**Extended version:**

There are a few main update strategies: cutting edge and lazy updates

Cutting edge: stay always with the latest versions. While it sounds good, it may be not the best strategy. We are dealing with big amount of custom and non well tested components so installing the latest versions of Armbain and Klipper is always bit risky. I would stay away from it unless a) your have internet facing system and you have to care about CVEs and b) you just having fun out of updating and troubleshooting process.
 
Lazy updates: update only when you really need it, e.g. if you got a concrete problem and you are 100% sure that a new version will solve it.

In my case (SKIPR + CANBus on a CoreXY machine) I have not noticed any performance of functional benefits during upgrading from 0.3.x to latest 1.0.x images. But a few peoples reported some strange kernel or/and Debian related issues when jumping to the latest versions ([1](https://github.com/redrathnure/armbian-mkspi/issues/37), [2](https://github.com/redrathnure/armbian-mkspi/issues/35), [3](https://github.com/redrathnure/armbian-mkspi/issues/34)). So for myself I perform partial update regularly  (`apt update` + klipper update) and full distro update in very rear cases (e.g. when I want to switch to the new Ubuntu release).

If you decide to upgrade, there are some information which may be helpful:

1. Backup first. It does not really matter are you use just "copy to USB stick" or backup with `resticprofile` approach, but in most cases you would need access to the old `/etc/*` and `/home/*` files. If you have 2 microSD/eMMC storage it's even better.

2. A `custom` version notation consist of 2 parts: e.g. in `0.3.5-24.2.0-trunk`, `0.3.5` points to internal version and `24.2.0-trunk` is Armbian version, which was used as base for the build. 
A few remarks here:
 - 1.0.0 is a version where almost all changes were contributed to main Armbian repo. Good side is now we have a deb packages on the Armbian mirrors. As a side effect I had to rename DTS files (`rk3328-roc-cc` to `rk3328-mkspi`) and custom overlays (`mkspi-uart1` to `mkspi-disable-lcd-spi`). Please check [root readme](https://github.com/redrathnure/armbian-mkspi) if you have DTS customizations or use custom overlays.
 - in a few latest versions Debian based distro was moved to NetworkManager, this may affect your network configuration steps, especially in case of CANbus. Please see [readme](https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus) for more details.
 - lot of custom images are based on trunk Armbian, means cutting edge and are not always stable changes. E.g. `1.0.1-25.2.0-trunk` was moved to Kernel 6.12. We have [an issue](https://github.com/redrathnure/armbian-mkspi/issues/37) which may be related to the latest kernel changes. Feel free to step back a few previous releases to find the best working image for you.

3. In generally an image consist of 3 main parts + 1 with your custom apps:
 - kernel + kernel header + u-boot files (e.g. something what you see on the /boot partition) 
 - Armbian tools, e.g. armbian-config
 - Debian/Ubuntu packages
 - your custom stuff like Klipper, KlipperScreen ete

The updates for the first two parts are blocked by default, the last two may be easy to update via apt and kiauh/WEB UI. And this should give you a good stability level by default. 
In theory you may update even the first 2 parts:
 - for pre v1.0.0 images you have to install related packages manually (another good question is where to get these packages :) )
 - for v1.0.0 and later images you may use official Armbian package mirrors. Means enable/unfreeze kernel updates in `armbian-config` and execute `apt upgrade`. It will install latest Armbian packages (but not a custom from my repo), which should works. However I have never tested it.
 - and in theory you may enable/unfreeze kernel updates even for pre v1.0.0 images, however it will install new rk3328-**mkspi** files and overlays with sligtly different names.

None of these options were tested and none of these options may be recommended from my side :) I prefer clean install if you really need it.

4. Last but not least. Although the difference between `0.3.6-24.2.1` and `1.0.1-25.2.0-trunk` looks big, I don't think you'll get many new features. You will stay with 6.x kernel family (probably jump from 6.6 to 6.12). Usually we do not need too much from Armbian and Debian packages, and latest versions may be easily gotten via apt. From custom patches you would get a few really minor changes, mostly around external HDMI and `irq xx: nobody cared` message. 
IMO the main reasons to do update would be ether jumping to the new distro or wish to update all related instructions.

Phil1988, 2025-01-25:
Thanks @redrathnure for your answer!

You are welcome - feedback and proper credits are always nice to receive as a developer, and you did amazing work and truly deserve it.

For me, you can always skip the TLDR part, as I read everything 😅

So, in a nutshell, there isn't really a workflow for updating existing systems.  
That's what I feared, and I didn't see another way either, so I double-checked with you just to be sure.  
Backing up and setting up again isn'’t an issue for me, but for FreeDi users, it might cause some frustration.  
Not all users are very tech-savvy, and it could be error-prone, time-consuming for them, and likely lead to new issues (wifi/network manager/DTS/dtbo).  
The main advantage that would really matter would be the "deb packages," and the other points would only be "nicer numbers" like the latest Armbian version and kernel.

If I misunderstood or misinterpreted anything, feel free to correct me!

Thanks again for your great work, which I truly appreciate!

redrathnure, 2025-01-30:
Yes, from practical side it's more about "nicer" numbers.

> Not all users are very tech-savvy, and it could be error-prone, time-consuming for them, and likely lead to new issues (wifi/network manager/DTS/dtbo).

I am not sure which degree of "tech-savvy"es you are talking about. However I would assume that people are able to open cover, flash emmc and do initial klipper/moonraker/screen setup (because they have already landed to custom image somehow). And if you are asking about upgrade without full re-flashing of emmc drive, in theory it should be possible reinstall or upgrade kernel and armbian packages, however it looks more difficult and error prone procedure than backup-flash-restore cycle.

Phil1988, 2025-01-30:
thanks man, all questions being covered.

regarding the people: you "assume" too much 😆

closed 2025-01-30 by Phil1988