redrathnure/armbian-mkspi #48: [Bug]: MKS X-7 V1 Board Not Booting with Latest Release

cgarwood82, 2025-01-29:
### What happened?

Wasn't sure what changed between releases, but the latest release doesn't boot up on the MKS X-7 V1 board whereas the previous release did. Wasn't sure if build targets changed or what, but just wanting to chime in and sound off on my issue. I tried each version in this release to no avail. Last successful image was Armbian-unofficial_25.02.0-trunk_Mkspi_bookworm_current_6.6.65.

Thanks in advance. 

### How to reproduce?

Flash image with Balena and plug into device, power on. Monitor SOC board lights for activity, that they stay static whereas when it boots, they start flickering. 

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

redrathnure, 2025-01-30:
@cgarwood82 I do not have this board so may only guess what would be a root cause. Images are tested only with MKSPI and SKIPR boards, the adaptions for other MKS deviations are up to community. 

About ideas:
1. (sorry for repeating this) please double check your media. MicroSD/eMMC storage sometimes fail, so trying another one of just try to flash the same image but second time could be a good attempt for trableshooting.

2. (more realistic cause). Diff between `1.0.0-25.2.0-trunk ` and `1.0.1-25.2.0-trunk` has no MKS specific changes. However there are few major changes from Armbian side. One of them is kernel bump. Current: `6.6` -> `6.12` and Edge: `6.12` -> `6.13`. Perhaps your board does not works well with newest 6.12 kernel. And it possible that kernel config was changes (some modules were disabled)

Please keep in mind that latest release do not introduce any significant changes. So if you already have stable system based on `0.4.x` or `1.0.x` image I do not see any practical reasons to switching to the latest release

If you decide to do some troubleshooting I would recommend following steps:
- try prev release but Edge variant. This will clarify question about 6.12 kernel support
- try to find boot debug console, on mkspi it's USB TypeC output. Main idea is to see all kernel message even before screen start working. Sometimes partitions layout may be incorrect.  
- you may try different distros, just to prove that it is not e.g. Debian specific issue. In theory Ubuntu and Debian should be almost identical, but in practice it isn't always the case.
- please try to disconnect all USB devices, external displays and other "additional" devices before first boot. It may address a cases with ether unsupported USB device (which may block booting) or with power supply related issues. 
- and I would recommend to refresh a image from the scratch after each non success attempt to boot. The first boot may take significant time before first screen message (a minute or two) + there are phase when armbian extend partitions and doing other preparations. If board was power off or rebooted during this phases, then data on microsd/emmc may be corrupted. So one attempt is one flashing cycle :)

corndog2000, 2025-03-20:
Also experiencing the same issues on my MKS X-7 V1 board in my QIDI PLUS 4. Unable to boot with any images using kernel version 6.12.x or newer. Do you have a list of the modules that were disabled when bumping kernel versions?

redrathnure, 2025-03-22:
We may compare[ kernel config](https://github.com/redrathnure/armbian-mkspi/blob/main/config/kernel/linux-rockchip64-current.config) from current and old_working branches. It may show lot of changes, not sure how informative it would be. 
Or I may compare `lsmod` output for old and new images for mkspi board. It will show diff only for this board, howeverlooks like MKSPI is not very similar to your one.

Questions: 
1. any ideas what are we looking for? Filesystem, image layout, some other modules? 
2. Do you have access to kernel debug console? E/g/ MKSPI has USB Type C output which dumps all lobs including very early stages like U-Boot. If you could share U-Boot and Kernel logs it would significantly simplify the troubleshooting.

@corndog2000, @cgarwood82

redrathnure, 2025-03-22:
Diff of Kernel configs are pretty big, please check them locally:
- Current version, kernel 6.12: https://github.com/redrathnure/armbian-mkspi/blob/main/config/kernel/linux-rockchip64-current.config
- 1.0.0-25.2.0-trunk version, kernel 6.6: https://github.com/redrathnure/armbian-mkspi/blob/mkspi/1.0.0-25.2.0-trunk/config/kernel/linux-rockchip64-current.config

cgarwood82, 2025-03-29:
I've been chomping at the bit to hook up a usb to serial adapter. I'll try and put some cycles into this this weekend. It's weird because someone else with a plus 4 is able to use these images just fine.

spolack, 2025-04-11:
Upstream armbian works for me on Plus 4. 
>  v25.5 rolling for mkspi running Armbian Linux 6.12.22-current-rockchip64

But i noticed the system wouldnt boot, if there are usb peripherals like a stmlink programmer connected.

redrathnure, 2025-05-13:
Hm.. about WIFI: [there are some report of aic8800 problems on the latest kernels](https://forum.armbian.com/topic/50332-aic8800-wifi-sdio-module-not-working-with-kernel-612/). They refer to some PR with fix for the problem (mid of May). Probably I should prepare a fresh images for testing :)

>  it wont boot .... with an st-link programmer connected.

Just curious why do you need st-link during normal boot?

cgarwood82, 2025-05-14:
If I was a gambling man, it's because the tool head board on the qidi plus 4 requires an stlink programmer for the toolhead flashing process and they are doing all of the testing.

Phil1988, 2025-05-14:
@redrathnure as what @cgarwood82 said its not required by any means but while tinkering/testing and updating the Plus4 toolhead it sometimes stayed connected to the USB port :)

About the wifi module:
I got the same warnings/errors like north1 did in the link you posted.

I would appreciate if you can include the fix for your image and can you estimate roughly when this will happen? :)

redrathnure, 2025-05-18:
(may be relevant to this issue) there are few reports about issues with a kernel BTF enabled. If I got it right issue may be caused a slightly bigger boot and kernel images and U-Boot fails on the boards with small amount of RAM.

Just in case I am preparing latest few releases with a `KERNEL_BTF=no` flag. It's related to `1.0.3-25.2.1`, `1.1.0-25.5.1` and `1.1.1-25.08.0-trunk` releases. Warning: I do not have time for proper testing of all these releases, please use new images with caution, probably `current` is more stable then `edge` ones.

Phil1988, 2025-08-02:
@redrathnure 
Are there any plans for a new release with the new 6.13+ kernel that (hopefully) fixes the AIC8800 [driver bug](https://github.com/radxa-pkg/aic8800/pull/27) that have been present on 6.12 ?

redrathnure, 2025-08-02:
> Are there any plans for a new release with the new 6.13+ kernel

Let me check upstream project. Probably today/tomorrow will publish new images.

redrathnure, 2025-08-02:
The current still on 6.12. However the latest 2 edges use 6.14.7 and 6.16.0 kernels.

Phil1988, 2025-08-03:
I will give it a go - thanks!

cgarwood82, 2026-10-02:
@redrathnure Probably okay to close this bug. I don't have the hardware anymore and not sure it's worth keeping around. Feel free to re-open if needed.

closed 2026-10-02 by cgarwood82