redrathnure/armbian-mkspi #51: [Thoughs needed] Random system freezes on Armbian 25.02.0 und 6.6.65 kernel

Phil1988, 2025-08-03:
### Which feature would you like to have?

Hey mate,
Phil here the developer of [FreeDi](https://github.com/Phil1988/FreeDi).

I wanted to reach out because quite a few users (about 30 so far) have reported experiencing random system freezes using the current Armbian image from your repo together with FreeDi:
https://github.com/Phil1988/FreeDi/issues/290
https://github.com/Phil1988/FreeDi/issues/288

When this happens, the whole system becomes unresponsive (no SSH, no local input, etc.) and it can only be recovered by a power cycle. So far, I haven’t found a clear cause or a way to reliably reproduce it.

Some context that might be relevant:

My previous FreeDi releases were based on Armbian 24.2.0 (Bookworm), kernel 6.6.17, and klipper v0.12.0-289. This setup was pretty stable overall, though it was possible to freeze the board by updating the screen firmware. (The screen is a Nextion-like TJC serial display, firmware updates involve sending ~16MB via serial in ~3,500 blocks of 4,096 bytes, which occasionally triggered a freeze. [Example](https://github.com/Phil1988/FreeDi/issues/138#issuecomment-2633005270))

Since moving to version 0.13.x (Armbian 25.02.0, kernel 6.6.65), users now report freezes mid-print, not just during screen firmware updates. I haven't been able to establish a clear pattern or way to provoke the freeze.

Resource usage (CPU/RAM) is low enough that I don’t believe this is due to FreeDi itself. 

I've also discussed this with the Klipper devs to rule out issues on their end:
https://discord.com/channels/431557959978450984/431559228050636820/1401329004899991652
https://discord.com/channels/431557959978450984/431559228050636820/1401331037887398001

Given that a similar freeze happened in the past when the onboard (/dev/ttyS0) mainboard MCU was busy, I wonder if there could be an OS/kernel-related issue involved.

Do you have any ideas what could be causing this, or suggestions on how we could further diagnose it? I really appreciate your work and any help or insight you can provide. I’m happy to run any tests or share more logs if needed.

Thanks so much in advance!

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

redrathnure, 2025-08-03:
Yesterday I got a freez on SKIPR, could you try to reproduce it on your board?

What I did: a big traffic via Python/Moonraker for a 2-5 minutes. E.g. in my case it was attempt to download 1G timelaps file via WEB interface. Aster a few minutes system stop response for external connections (WEB, SSH) and report high CPU utilization on KlipperScreen. After a few minutes after freezing, KliperScreen came back to normal responsive mode, however WEB interface/Moonraker did not.

redrathnure, 2025-08-03:
OK, just a guesses, which definitely need to be proofed.

I may think about 2 causes: a) CPU/power/thermal issues or b) something with IO subsystem

a) CPU/power/thermal related issues

For some reason MKS used non recommended configuration for power chip and related CPU frequencies. In the latest few images I adjust this config (in DTS) which potentially may lead for under power or/and over heating of MCU. On MKS boards I was noticed about slightly lower voltage on the USB power lines, which may cause an stability issues for stuff like WIFI dongle or WEBcams. Monitoring of temperature (e.g. external thermo couple) and/or MCU freq (e.g. kernel sensors) may give more information. However this kind of issue should correlate with ether high CPU load or high power consumption of connected peripheral devices (e.g. USB dongles)

b) something with IO subsystem

It may happened that in the latest kernels or latest custom patches something IO related was changed. In my case I have quite a few devices connected (WIFI, CANbus board, EMMC, WEB cam and HDD via USB). So may be kernel just unable to handle susch intensive IO requests. However it may be hard to debug because I may loss a part of logs and network connection. Probably collection kernel logs directly may help in this situation (e.g. MKS/SKIPR board has USB Type C port which used to dump kernel log).

In any case it's pretty hard to guess around causes with so many variables. I would recommend to try:

1. try to found condition of fail. High load, IO overloading (intensive packer flow through CAN/UART/USB), may be correlation between some new hardware or new printing settings (e.g. increasing printing speed)
2. try to rollback to the latest stable Armbian OR latest stable Klipper. Main point is to understand what cause the issue (kernel, Armbian config for Klipper itself)
3. look to kernel logs. Probably dmesg would help here, however if you have "direct" access to kernel debugging console it would be better.

PS: these are just guesses from a gay who do not ahve access to your hardware and have very limited experience in this area :)

redrathnure, 2025-08-03:
... and double check the right overlay name. It should be `overlays=mkspi-uart1` for images prior 1.0.0 and  `overlays=mkspi-disable-lcd-spi` for the images 1.0.0 and later on.

Phil1988, 2025-08-05:
I am sure I have `overlays=mkspi-uart1` set on the 1.0.0-25.2.0-trunk image.
It is working for many as the screen is working properly (a TJC screen that uses serial to communicate) - but I can not say if that is causing issues in some conditions.

If `overlays=mkspi-disable-lcd-spi` would be the correct way, I would assume, that `overlays=mkspi-uart1` shouldnt work at all.

For the issue:
I dont think its a CPU or thermal issue - maybe a power issue in some rare cases but for me this seems to be unlikely.

As the system freezes happens occationally when uploading the screen firmware via serial, it could be very well something with the IO subsystem.

My question is, how we can track and debug this.
I am asking affected users to collect data here:
https://github.com/Phil1988/FreeDi/discussions/303

As I didnt had a freeze yet, I can not share details about it.

Btw: I did up/download big files via moonraker, but could not provoke a system freeze on my end.

redrathnure, 2025-08-05:
About overlay names:

I rename it when we went to official Armbian repo. E.g. see [files for old name](https://github.com/search?q=repo%3Aredrathnure%2Farmbian-mkspi%20mkspi-uart1&type=code) and for [the new one](https://github.com/search?q=repo%3Aredrathnure%2Farmbian-mkspi+mkspi-disable-lcd-spi&type=code). Perhaps uboot just ignore absent/misnamed overlays, probably with some warnings. 

> My question is, how we can track and debug this.

I have hard times to answer questions like this. Probably observing kernel logs (e.g. see [this question](https://serverfault.com/questions/305311/how-to-view-linux-kernel-logs-live)). However the very important precondition is to be able to reproduce the bug.

My experience 1:
Open SSH + `journalctl --system -f` and try to reproduce bug (the big file downloading in my case). This is how I found issues with my WLAN adapter and corrupted FS :)

My experience 2:
Some (cheap?) plates are very sensitive for power quality. Under voltage and/or huge noise (Vpp >10% of Vout) may introduce system instability (hanging out, resets or just broken FS). A basic multimeter may help to realize what going on with board power line under heavy load (e.g. enable all connected devices + all heaters). It is not correlates too match with Kernel updates but it may answer a question why the same boards work differently in different machines.

Phil1988, 2025-08-07:
2 days straight testing on 2 printers from my side - with no lock getting this system freeze once 😅

One user got it, while monitoring some logs
```
sudo dmesg --follow           # Live view of kernel messages (hardware, drivers)
sudo tail -F /var/log/syslog  # Real-time monitoring of system and service logs
sudo tail -F /var/log/kern.log  # Real-time monitoring of kernel-specific log file
```

I cant find anything suspicious - maybe you see more then I do?
https://github.com/Phil1988/FreeDi/issues/288#issuecomment-3163889828

redrathnure, 2025-08-07:
@Phil1988 not sure how to interpret these logs. Let's start from bit of lyric. Please do not take it as blame game or any other kind of weird communication :)

First of all, if problem can be reproduced by 7 or 10 users, it's about bug or design flaw. If problem occurs only for 1 or 2 or 10 users, most likely it's about a personal local issue. From my experience it may be ether causes some manufacture defect or no ability to observe/detect some technical issues. 

I have no any tech insides from MKS manufacturer, however I see that "MKS PI like" boards just flooded market of low costs machines. A number of devices from different brands are packed with mainboards with the same/close to the same schematic. IMO all these hardware were manufactured on the same factory and biggest part of the mainboard should be the same (CPU, RAM, main power and communication related components). And most important all these hardware must have very good repetitables.

This is why I hardly believe that we have a Kernel, Armbian or Klipper related bug here, otherwise we would have much more reports and it would be "more obvious" for our own machines and testing stands.

My personal top of causes for "magic random outstages":

1. Power supply AND/OR power lines. Some low-end technic from Ch. is usually packed with extremely crappy components in this area. Starting from "oversimplified filters" and ending by "just" very thin and "lack of cuprum" wires. 

1.1. Some of PSU are very sensitive for noise and other EMI related factors. Sometimes it means a motor or more-less power device near by may produce pretty long spike of "unstable" power which may be related to random issues such data corruption

1.2. Thin power cables OR not repayable contacts may lead to random unsuitability, e.g. data lost, especially for external storage or audible noticed difference in cooling fan noise when e.g. heaters are turned to full power.

I personally had a lot of very random "fun" with a Trident kit, especially during CANbus migration... which was gone after replacing all power lines by proper cables + adding an input AC filter.

Unfortunately, 99% of owners of cheap machines and printer sets do not pay due attention to this and look for a problem in the firmware or the phase of the moon...

Kinda 1.3. MKS use very weird CPU power profile AND RK805 configuration (RK805 is a controller to generate power for all board components). I am not 100% sure it's related to exactly your versions upgrade, however I changed CPU power/freq. profiles as well as RK805 configuration and SPI freq for TS35 a few times. If your machine has power related an issues it may explain an instability of some releases. e.g. higher CPU frequency leads to higher power consumption which leads to stability issues. However it does not really explain why only some of users have reproducible problems. In any case you may check releases changelog by "rk805" or "SPI freq"  keywords.

2. Storage layer, which includes extremely cheap SD, EMMC or SSD components. Sorry to use the word "cheap" again, but we all know that's usually the opposite of "reliable". Writing it here just because I pretty often observing a random issues up to "non bootable" system or kernel panic on very early booting phase during releases testing... which were fixed just by re-formatting or re-writing images... yes, the same image to the same SD/EMMC card. And yes, in 9/10 cases my "mystery" problem just gone. And a few days ago I was too close to re-flashmy main printer from scratch just because a few corrupted files lead to fully irresponsible system (no WEB, no SSH,  KlipperScreen is unresponsive). 

So my advice here is to backup printer_data and re-flash system from scratch. Probably with another one EMMC card. Yes, it means disassemble the printer and spend a hour for configuration, but in some cases it may be the faster and probably cheaper solution. And IMO ordering another one EMMC card for a price of a filament roll may be a reasonable investment, especially for people who cares about updates:)

3. Sometimes OS distro may introduce significantly changed in scope of the same release. E.g. once Debian decided to switch to NetworkManager which bring a lot of surprises in CANbus area.

About logs, I see a few confusing points:

1. Flood of "need cfm ethertype:" messages. I am not sure this is a problem, just a strange observation. And looks like a lot of messages were produced after 2025-08-07T10:16:39.704118+00:00. 

2. Syslog has a strange log events sequence:
```
> Looks like normal Klipper logs

2025-08-07T10:15:00.561982+00:00 mkspi sh[1370]: #033[38;5;214m[Thread-6 (dynamic_lcd_update)] Response type in lcd: unknown - Message: B:61.4 /60.0 T0:229.8 /230.0#033[0m
2025-08-07T10:15:12.606352+00:00 mkspi sh[1370]: #033[38;5;214m[Thread-6 (dynamic_lcd_update)] Response type in lcd: info - Message: pressure_advance: 0.040000
2025-08-07T10:15:12.606897+00:00 mkspi sh[1370]: // pressure_advance_smooth_time: 0.030000#033[0m

> DHCP lease expires? which is totally fine...

2025-08-07T10:46:35.407207+00:00 mkspi NetworkManager[571]: <info>  [1754563595.4061] dhcp4 (wlxc83a35d1c61c): state changed new lease, address=192.168.66.127

> But this part is strange. Assuming Python means Klipper or Moonraker, they keep connection to localhost and some other address... why? WEBCam?

[... cut ...]
3. And crowsnest has latest message with `[08/04/25 08:46:45]` timestamp. Moonraker with `2025-08-07 12:18:05,554`. Does it mean that printer uptime is more then 3 full days and only on 4-th it decides to crash? It's kinda strange... IMO should be some triggers for the outage. Any observation around changed conditions? Print in "other bed parts" (e.g. on edge corners)? Or may be some cronjobs in background (which may grab all open FS limits)? Or may be some unusual sounds, noises or tone of fans? Or may be something else around power, temp or network conditions?

jmpocheau, 2025-08-09:
First, let me thanks a lot all who try and help resolving this issue.
Please find here the answers to your questions.
Feel free to ask me whatever you need, I will do my best to answer, but since I am currently on the road, this may take some time before I can reply. I'll be back on monday.

> You are right, I pinged [@jmpocheau](https://github.com/jmpocheau) to hopefully get some more information about that.
> 
> Questions would be: What is behind 192.168.66.127 (likely the printer itself) and 192.168.64.133 Was the printer running for about 4 days straight?

Yes, absolutely. I did a couple of prints before the problem occured. I first started the ssh sessions as required, and set up the log level to debug before launching the commands to watch the logs. I let everything working until the failure occured. So 4 days is really possible.
 
> 
> The `EVENT LOOP BLOCKED: 11.48 seconds` is possibly a main problem. It is followed by `Websocket Closed: ID: 281473682570704 Close Code: None, Close Reason: None, Pong Time Elapsed: 20.84`.

I also suspected a mains issue. In fact the mains network is not good where I live, and it can stop sometimes for several seconds. Since I host a few servers, this was an issue for me. So I set up an UPS for my servers, and when I suspected a mains issue for my printer failures, I plugged my printer to the UPS.
Now I am pretty confident that my UPS works well, because I cannot see any problem with the other devices plugged to it, and because I know that I had one mains issues (then the UPS beeps), and the printer didn't failed.

Thanks again.

jmpocheau, 2025-08-09:
> Thanks a lot @redrathnure for taking your time to think about possible
> reasons and to check the data we have so far.
> I am totally fine with an open communication and its not about blaming
> anyone as I am sure every developer involved did the best to provide
> working stuff :)
>
> I totally agree that this problem is a hard one.
>
> 1. PSU
> In your first section you discuss about the PSU and related things
> (filter, wires etc.):
> I am pretty sure that this is not the source of this issue.
> The reason for my opinion is that I had not a single user on the
> previous release with this error and some of them updated and have it
> now.
> I would expect that some of these "edge cases" would have happened at
> least once previously (also in regard that the first release was
> published ~1 year ago and there are hundreds of users).
>
> Currently I know about 25 users with this error.
> Not much but even the ones that have it can sometimes print days with
> no issue.
> So maybe the other "dont affected" users dont print as much (and
> currently I am also not printing much because of private live and my
> spare time almost fully goes into developing rather then printing on
> [... 64 lines cut ...]
>
>
> I hope together we can track and get rid of this bug.
> Thanks again for your nice work here at the framework and also for
> helping with this error 🙏😊

What is behind 192.168.66.127 (likely the printer itself) and 192.168.64.133

⁣192.168.66.127 is the WiFi network interface of the printer.
The ethernet interface is 192.168.66.130 but not used then.

192.168.64.133 is the address of the PC runing mainsail.

For your information, I use Orca.

Best regards

Phil1988, 2025-08-12:
@redrathnure 
As a short and promising update:
The 2 images I prepared 5 days ago looks both really promising.
So far none of the users have had a single freeze, while they had many on Armbian 25.02.0 with kernel 6.6.65.

I am not sure what else was changed beside the OS and the kernel (like cpu parameters, IO things etc. - you are the king here), but I tend to say that the freezes are gone now and are linked to that specific release.

I will leave this open for a few days and close it, when the ongoing tests of the users are still flawless.

Thanks again for your time and this project 👍